tbnobody/OpenDTU #1363: [Request] modify information display

m4sis, 2023-09-20:
### Is your feature request related to a problem? Please describe.

I have a white/yellow screen and the 'daily total' overlaps the two colors. which is not readable, I attach the link to better understand in pictures
[https://ibb.co/jHW0MK9](url)

### Describe the solution you'd like

I would like to simply delete the daily total line, or otherwise know how to modify the font size and layout to adapt to my screen

NintendoTito, 2023-09-24:
That would be very helpful! A friend of mine has the same problem.

stefan123t, 2024-08-15:
@m4sis added your picture inline