tbnobody/OpenDTU #1364: Wlan AP ständig on

apreller64, 2023-09-20:
### What happened?

Bei Verwendung von Wired Ethernet (Olimex ESP32-POE) bleibt das WLAN im AP Modus ständig aktiv. Die SSID "OpenDTU-****** wird gesendet und unter Info -> Netzwerk steht bei "WLAN-Informationen Status aktiv". 

### To Reproduce Bug

Keine Wlan Konfiguration,  Anschluss per LAN

### Expected Behavior

Ich hatte erwartet das die OpenDTU den AP Modus nach x Minuten deaktiviert. Das ist aus Sicherheitsgründen wünschenswert. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V23.9.18

tbnobody, 2023-09-20:
What do you see on the serial console? There should be something like 
`Admin AP remaining seconds:`
which counts the seconds until the AP gets disabled... what happens if your count down reaches its target?

apreller64, 2023-09-21:
The DTU is installed in the roof box, the serial port is not routed to the outside. I will expand the DTU this weekend and report back.

tbnobody, 2023-09-21:
Was ist denn hier deine Einstellung?
[image]

apreller64, 2023-09-22:
[image]

tbnobody, 2024-01-29:
Due to the fact that there was no further response, I assume we can close this issue. Feel free to reopen it if it persists.

closed 2024-01-29 by tbnobody