tbnobody/OpenDTU #1366: MQTT Value jumping randomly in Home Assistant

koch86, 2023-09-20:
### What happened?

Using OpenDTU version 23.8.28
Almost every day I need to do corrections in the Home Assistant statistics, since there is one or two invalid values.
This looks like this:

[image]

### To Reproduce Bug

Happens randomly but almost every day at different times

### Expected Behavior

No jumps in the value

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.8.28

tbnobody, 2023-09-21:
To determine if this value is coming directly from the inverter or is somehow wrong interpreted by OpenDTU it is unfortunately required to have a serial log of the time period when the error occurs.

koch86, 2023-09-21:
Ok, will try to record one of the moments.
Might take a while since the issue happens sporadically at any time of the day

Is there any better way to obtain such a log than having a PC attached to the ESP that runs a serial monitor?

tbnobody, 2023-09-21:
> Is there any better way to obtain such a log than having a PC attached to the ESP that runs a serial monitor?

Unfortunately not... That's the most reliable way (to also capture exceptions from the core etc.)

jstammi, 2023-09-21:
There is also wscat, that you can attach to the webapp monitor

koch86, 2023-09-22:
Thanks for your assistance! Decided to switch to other power measurement in Home Assistant.

closed 2023-09-22 by koch86