tbnobody/OpenDTU #1369: [Request] adjustable reactive power

DarkSummit, 2023-09-22:
### Is your feature request related to a problem? Please describe.

According to my local grid operator I have to make an adjustment of the reactive power.
The default value for the Hoymiles is 1.0, I have to set 0.9 here.

### Describe the solution you'd like

Please add the option to set the reactive power.
How it is possible with the DTU Pro

tbnobody, 2023-09-22:
duplicate of #900
If you really have to change the profile you should contact your reseller. It's not event sure that the grid profile save method can somehow be reverse engineered or implemented.

closed 2023-09-22 by tbnobody

Fyr42, 2024-02-27:
I found this document.  Just before the "4 - Conclusion" the picture mention the need of DTU Pro or Pro-S to access to the profile and modify it. The web interface is the vendor one "O&M" 

https://www.hoymiles.com/wp-content/uploads/2023/01/Technical-Note-Output-Power-Parameter-Interpretation-and-Control-Instructions-for-Hoymiles-Microinverter_REV1.1.pdf