tbnobody/OpenDTU #1372: Connection to inverter lost after attaching display

sundance-kid, 2023-09-23:
### What happened?

I am running OpenDTU for several weeks now (current version is v23.9.18, but issue was also present with 23.8.8)
Board is an NMCU-ESP32 (BerryBase)
My configuration is LED0=25, LED1=26, NRF24: CLK=18, CS=4, EN=4, IRQ=16, MISO=19, MOSI=23.

Everything worked just find until I connected an 1.3" display (SH1106): CLK=22, DATA=21, type=3

When I connect the display, it shows all the inverter's information correctly, but at the same time, the RF connection to the inverter is lost (RX Period End - All missing - Nothing received, resend whole request) and soon after that the inverter's LED goes off and the display reads "Offline". After detaching the display, everything is back to normal. Tried another display (same type) with the same results.

Any idea?

### To Reproduce Bug

Connect the display

### Expected Behavior

Working display and online inverter

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

d419fd4

mroenne2022, 2023-09-23:
CS=4 and EN=4    Is this correct?

sundance-kid, 2023-09-23:
Sorry, typo:

CS=5, EN=4

 

This is my config file:

    {

        "name": "NRF24L01+, LEDs (GPIO25/26), OLED-Display SH1106",

        "nrf24": {

            "miso": 19,

            "mosi": 23,

            "clk": 18,

            "irq": 16,

            "en": 4,

            "cs": 5

        },

        "display": {

            "type": 3,

            "data": 21,

            "clk": 22

        },

        "led": {

            "led0": 25,

            "led1": 26

        }

    },

sundance-kid, 2023-09-28:
Thanks a lot for this, you made my day. I already had a 10 uF cap directly attached to the NRF24L01+ transceiver, and after attaching another one + 10nF ceramic next to the OLED module, everything works fine now!

broth-itk, 2023-09-28:
Glad to hear your news, thanks for sharing!
Please consider closing the issue as the root cause is not software related.
Cheers,
Bernhard

closed 2023-09-29 by sundance-kid