tbnobody/OpenDTU #1375: [Request] Global MPPT für HM-Series

k0017, 2023-09-24:
### Is your feature request related to a problem? Please describe.

HM Micro-Invertes don´t track for a global maximum point. This is very import to activate the bypass diodes when the panels are shadowed. The panels have three bypass diodes. When a panel is shadowed, the voltage should decrease which actives the bypass diodes and the shodowed vertical segment will bypassed. This is not implemented in the firmware of the hm inverters. There should be a scan between the low and high end of the voltage the inverter regularly eg. every minute. The scan should be very fast, because when this happens, we don't have a maximum power point.

We now have millions of balcony solar systems, which did not work well and loose almost all of their power due to the lack of the global scan for a maximum powerpoint. The panels may be damanged due to hotspots, which occur when panels are shadowed and the bypass diodes will not be activated by the inverter. It is a scandal.

### Describe the solution you'd like

Import the firmware, analyse it, maybe, there is already a subprogram, which isn't activated or it has to be added. Best would be to have a switch to activate it in openDTU and a parameter for the time, between the global scans.

### Describe alternatives you've considered

there is none. The manufacturer won't do it

k0017, 2023-09-24:
looses almost all of their power, when they are in a shadowed condition.

jstammi, 2023-09-26:
a) I doubt this is something that openDTU will ever implement - this is not about a inverter's firmware. At a maximum the inverter's firmware update is something that I would consider related

b) despite this: my panels actually get shadowed by roof of neighbouring house to the south. I just took a picture and checked the most recent power history for the 2 panels (400W, with 2 diodes each afair, connected to HM-600). And to my understanding I suppose that diodes are doing their work as I would expect. Else the blue line had to stay close to the magenta one with both panels still partially being hidden from the sun ... ?
[image]
[image]

For reference the history for a recent all sunny day in the beginning of september:
[image]

k0017, 2023-09-26:
Most users have three diodes, left, middle end right vertical segment. Those diodes have to be activated by the inverter, but that don't do the Hoymiles. 

And the newer panels are divided in two halves as well. So when the upper or lower segment is shadowed, the other half can produce. This is done without the help of the inverter.
Those newer panels have the tree vertical segments as well. That doesn't change it is nescessary the inverter finds the global MPP.

https://www-photovoltaikforum-com.translate.goog/core/article/47-der-vergleich-halbzellenmodule-und-vollzellenmodule/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp

Maybe you have three diodes as well, you can see the at the backside.
Thanks for your attention!

jstammi, 2023-09-26:
Independent if my panels have 2 or 3 - with those not being activated, to my understanding, they ARE activated. Indicated by the stepwise increase and lowering with the shadow passing the panels. With the diodes not getting activated, both would go down to min directly, and increase power again to max without an intermediate step between. As then the solar cell with the lowest current would dominate for the complete panel.

Here again the power history, now additionally accomplished by the voltages. With power reduced because of increasing shadow, and voltages varying - I would asume due to MPPT?:
[image]

So where do you see from in the power histories provided, that those diodes are not activated?

Or where do you have the information from? I did not find complains on this by quick search?

k0017, 2023-09-26:
you can see, that the diodes are not activated for the vertical diodes. When a vertical segment is shaded, then the diode will only bypass the shadowed segment, wenn the voltage is reduced. Then the segment delivers no power due to the activated bypass diode. If there would be a vertical segment bypassed, it would reduce the voltage about 1/3 of the voltage. You can see, that the voltage doesn't drop on your case.

k0017, 2023-09-26:
The Problem is huge, but people aren't aware of it. But you can try to shade a very small part of a vertical segment (upper and lower both at the same time) and you panel will loose almost all of its power. This does not have to be if the inverter scans the Voltages from the lower to the upper end and search a global maximum power point (which is more on the left --> lower voltage).

k0017, 2023-09-27:
https://www-photovoltaikforum-com.translate.goog/core/article/13-wie-wirkt-sich-verschattung-auf-pv-module-aus/?_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp#google_vignette

avandorp, 2023-10-16:
My HM-600 does seem to trigger the diodes in my panel. You mileage may vary.
[image]

k0017, 2023-10-16:
If one bypass diode would be activated, 1/3 of the voltage would have been dropped, which isnt´t the case. I wrote to Hoymiles. They confirmed, that they don't search for a global MPP and therefore don't activate the bypass diodes.

avandorp, 2023-10-16:
Then they should call their MPTT LMPTTs. :( Local maximum power point trackers... I have assumed that scanning for the global maximum is state of the art since years.

k0017, 2023-10-16:
Meanwhile I am convinced, that none of the micro inverter can do the sweep of the voltage curve, but this is important.

This should be simple to implement for the manufacturer, almost all string converters do have this feature.

k0017, 2023-10-16:
Oh, there is one microinverter, that is able to kick in the bypass diodes. It is an enphase microinverter. 
The youtube video states, that Hoymiles never activates the bypass diodes.

https://youtu.be/yhicm5NKTlU?si=jdps9qAZb6KrOenl

jstammi, 2023-10-18:
As I did not want to believe that it is true, that the diodes become used (believed it to be indepedent from the inverter in fact), I yesterday gave it a try with sun constantly shining.
And, IMHO, AFAIS, the results show ... diodes do not get triggered :-(. As - at least to - it becomes obvious by "B" in the illustrated measuring of my panel's resulting power for different coverings (using an opaque towel)
[image]

This is NOT the result that I wanted to see :-(

k0017, 2023-10-18:
In Fact, nobody is aware of it, but in my opinion, it is a scandal, because all balcony power plants have micro inverters. they loose exraordinary power in part shading and they have the risk of fearlier degradation and even fire.

And it is not so hard to implement. Maybe, it is impossible for two shaded segments, because the inverter can't go down to 10 Volt (min is 16). But 20 Volt with one shaded segment muss work. And  I am not sure, If with two segments shaded, we get bypass diodes kick in at 16 volt. It would be a dream to have a solution. 

I have seen a lot of work on e-scooter firmware reengeering with very much success. It is clear, there is a risk of damage, but they did it.
There should be a scene for inverters, which could be tweaked. There is also a risk, but maybe someone tries

k0017, 2023-10-18:
BTW: the results you have are better than results, you would hav gotten with a standard panel, wich isn't devided in upper and lower part. In Fact, half-cell panels have little advantages in part shading, even if the micro inverter can't find a global mpp. If the upper or lower half of the vertical panel is shaded, the panel will reduce the shaded half completely, but is able to deliver the power of the other half without interaction with the micro inverter.

k0017, 2023-10-18:
In fact, micro inverters get more popular even in bigger solar plants instead of using one string inverter due to the control of each panel. But because of the lack of global MPPT, there is higher risk of burn in such plants on the roof, where there is much wood. String inverters do have a global MPPT!!

stefan123t, 2023-10-18:
@jstammi we have been discussing the topic in Discord too these days:
The diodes will be irrelevant if you connect two identical panels in parallel (using a Y connector) and shade only one of the two panels.
The explanation / hypothesis was that the other panel will deliver the necessary voltage to keep the diodes satisfied (non-conducting).
Hence the current from that second shaded panel will still be flowing but reduced.

Is that something you could try out with your test setup ?

k0017, 2023-10-18:
> @jstammi we have been discussing the topic in [Discord](https://discord.com/channels/984173303147155506/998975048230641736/1163766022596214825) too these days: The diodes will be irrelevant if you connect two identical panels in parallel (using a Y connector) and shade only one of the two panels.

This situation has nothing to do with what is discussed here:

We here are discussing one partly shaded module --> one or two segments are shaded and we try to deliver maximum power despite of partly shading from this one module.

What stefan123t describes seems to be  a completely shaded module, which could affect another module IN A STRING OF MODULES e.g on a roof. A balcony power plant has typically two completely independent modules, each has its own MPPT. Shading one module does not impact the other due to their own MPPT: So this situation is somethingg different. I want to prevent misunderstandings or the discussion goes into a wrong direction.

jstammi, 2023-10-18:
Btw, I added this question on the HM600 wrt bypass diodes also to the akkudoktor forum https://www.akkudoktor.net/forum/postid/158368/, as he (among some other youtubers) tested a lot of micro inverters recently - but not (yet?) checking for their mppt afaik

k0017, 2023-10-18:
@jstammi : there is a danger when you test two modules parallel connected to one mppt on a hm-600. The power can be much greater than allowed on one mppt. This can break your hm 600

jstammi, 2023-10-18:
> @jstammi : there is a danger when you test two modules parallel connected to one mppt on a hm-600. The power can be much greater than allowed on one mppt. This can break your hm 600

I am aware. With 1 to east and 1 to west I think this could work - if I can trust the dc currents measured (by my hm-600) up to now from my panels (directed to the south).
But before connecting them in parallel, I will for sure 1st measure currents separately to make sure that they do not exceed hm 600 spec (else I need to upgrade the inverter e.g. to hm-1500 being limited to 600W output then)

Btw, I plan to do so only for one of the 2 ports. For the 2nd the setup shall look like 2p(east-west)-2s pv -> charger -> 24V lifepoe4 -> dpm8624 -> hm600. To adjust injected power to current needs. And have some remaining buffered for the night. Related hardware is sitting in my labor already.

> @jstammi : be aware that even in such forum they don´t know very much about this topic. Trust me. I made this experience in https://www.photovoltaikforum.com/ as well.

Maybe. But after the Deye drama wrt missing relais, a group of youtubers, with Akkudoktor among them, tested a lot of micro inverters recently. And I hope that I can somehow raise some attention to them for a another problem. So his forum is my 1st step in that direction (fingers crossed).

k0017, 2023-10-18:
It would be really glad if you were able to raise attention to this problem. I wrote to Holger Laudeley. I think, you know him. But no anwer.

k0017, 2023-10-18:
I am Sorry, the Request i obviously closed accidentially

closed 2023-10-18 by k0017

avandorp, 2023-10-19:
After logging more voltage/power data it seems to me as if the HM-600 sweeps on startup from the (high) voltage the panel is supplying when not under load and stops at the first maximum it finds. That's especially annoying as I've bought the inverter because it was advertised to start at low voltages (22 V and MPPT going from 16 to 60 V). But judging from my data it always starts at roughly 40 V. It does seem to shut off somewhere in the range from 16 to 22 V in the evening, which is somewhat ok. By starting at a high voltage and scanning downwards for the first maximum it will never find the global maximum in a shaded situation.

stefan123t, 2023-10-19:
To find the Global Maximum the inverter would need to sweep across the complete voltage range in small increments, which would be small enough to incrementally trigger the bypass Diodes of the modules, thus allowing to find all local Maxima.

Currently the HM inverter firmware offers only setting the AC side parameters using a so called Grid Profile, eg https://github.com/lumapu/ahoy/wiki/Protocol#file-name-de_vde4105_2018

I am unaware of a similar DC side PV/MPPT Profile which would allow to fine tune the input side of the inverters Maximum Power Point Tracker(s).

But this issue is actually not a feature missing in OpenDTU but instead the HM Firmware. As the firmware is not OpenSource (yet) we will have to wait until someone comes up with either an OpenHardware Project for a MicroInverter with firmware which can implement logic to find such a Global Maximum.

k0017, 2023-10-19:
Do someone have the Firmware as file? That maybe could be inspected. 
Do we know, which SOC is built in?
The firmware of the Ninebot Scooters is not open source, but there were people able to analyse the firmware and change it to their needs. It is extremely popular.

How can we find such people, that can do it?

Is there any API, that can control the HM from outside? We can adjust the power limit. If you could adjust the Voltage via API, then you could do the sweep in openDTU. Maybe there is a max voltage? I don´t believe so, but who knows. 

What else could someone think? Maybe an external MPPT, just ike an optimizer. Maybe, there are solution, we can think about?!

stefan123t, 2023-10-19:
You may consider to discuss these topics on the Discord chat https://discord.gg/WzhxEY62mB

EmJay276, 2024-03-11:
Any update on this topic? I see the same behaviour on my panels with a HM-600

avandorp, 2024-03-11:
I don't think there will be any update. If the inverters are physically capable of supporting a real MPTT algorithm and if Hoimyles releases a firmware update, then we're lucky. Otherwise it seems that one simply has to buy a converter with real MPTTs (or a separate optimizer). I just feel that the way the Hoimyles inverters are advertised that it sounds that they are more capable then they really are.