tbnobody/OpenDTU #1397: no yield per day after sunset with HMS-Series, HM is ok

kr0815, 2023-10-01:
### What happened?

I have 2 OpenDTU-Installations
- one with 4 HM-1200 Inverters
- the other one with 3 HMS-2000 Inverters

After sunset, i see on the HM-1200 inverters the yield by day
In the HMS-Installation, YPD shoes zero after sunset

### To Reproduce Bug

what shall i say :-)

### Expected Behavior

see YPD for both installations

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.9.18

AloisKlingler, 2023-10-03:
since I disabled the "poll inverter at night" this did not occur any more for me.
see https://github.com/tbnobody/OpenDTU/issues/979

kr0815, 2023-10-03:
thanks, i will give it a try :-)

I don´t understand - how to even poll the inverters at night - they don´t send anything when there is no input?

huste511, 2023-10-03:
I also have several HMS here and have never been able to see the daily yield after sunset.

I also deactivated nighttime polling.

Since OpenDTU can determine the day itself, smaller values ​​from the same inverter should be ignored - it would be even better to cumulate them if necessary.

kr0815, 2023-10-04:
I changed the settings for "poll inverter at night" - still strange , now another of the 3 HMS shows a yield per day, the others zero - musst be really a bug?

The HM-Inverters show all as expected

huste511, 2023-10-04:
Not really a bug in openDTU but an area where improvements could be made.

The HMS switches off at some point at dusk. If it gets a little brighter at some point, it starts again, usually doesn't provide any energy, but for it it's still a "new day" and it and delivers 0 values...

tbnobody, 2023-10-04:
I am currently bad out of order. But please change the eventlog. Is it possible that your hms inverters reboot at evening? In this case the daily counter is resettet. Happens when e.g. at the evening the wether is cloudy and at the end the sun still comes out a bit to power the inverter again. Hm and hms might react differently here.

kr0815, 2023-10-04:
--> tbnobody - get well soon :-)

What do you mean with changing the event log?

You are right, it could definetly be possible they reboot, show zero because of that event

tbnobody, 2023-10-04:
I mean, have look at the event log. Sry. Brain is mud. 

In the event log should see the last startup time of the inverter.

kr0815, 2023-10-04:
sorry if it sounds stupid, but where to see the event log? :-)

info - console?

tbnobody, 2023-10-04:
Ne. It's in the live view. I top of each in inverter are several buttons. One is for the event log

kr0815, 2023-10-04:
strange:

Inverter 1 - YPD 0  - inverter start 19:17
Inverter 2 - YPD 0 - inverter start 07:09
Inverter 3 - YPD 11,9 kWh - inverter start 07:09

The HM all say Inverter Start around 7:14

tbnobody, 2023-10-04:
Maybe hm require less power to keep them operational. But there is already a first walkaroumd for this issue. Will have a detailed look and modify maybe the on or other thing. Bit this feature will come in if I am back.

kr0815, 2023-10-04:
Take your time :-)   

It´s fascianting, - what i observe, HM are generally better regarding output
not much, but  it´s noticible

broth-itk, 2023-10-05:
Duplicate of https://github.com/tbnobody/OpenDTU/pull/1268 ?

I proposed a fix to cache the YieldDay values.

The problem can be observed when the interver restarts

- in the evening
- if panel is covered and disconnected for maintenance
- intentionally on interver restart

kr0815, 2023-10-06:
i guess there is another problem / bug:

See the red Message on the HM-1200, letzte Aktualisierung
With the HMS-2000 it is not red

[image]
[image]

tbnobody, 2023-10-06:
Grey means that you've disabled polling at night. And that it is currently night.

Lineflyer, 2023-11-11:
> ere.

I do also have this exact issue with my HMS.
I thought the "Zero daily yield at midnight" feature was the foreseen fix/workaround for this problem...but seems not. What is the benefit of this feature then?

For my understanding:
Is this value provided by the HMS or calculated within the DTU?
If its calculated in the DUT it should be an easy fix, if the value comes from the HMS it would need to be buffered and added within one day by the DTU.

pimisen, 2023-11-21:
Same issue here. 2 HM. No issues. One HMS YPD gone after sunset

GM-Jokemaster, 2023-11-29:
Same here HMS800 T2. (Newest Version) v23.11.16
Reset to 0 on the same Day. At Power lost at the OpenDTU (Reset at Midnight is active)
Also a Friend of mine same HMS same result.
Please try to Store the yield per Day and if NTC Time tells same Day = Add the new (now low) vlaue to the Last before "Power lost"?
TY !!

broth-itk, 2023-12-01:
> Yes, its basically that the chance of the HMS to stop transmitting due to sunset and then getting alive again for a few seconds short after seems significantly higher than with the HM series. This effect IMHO causes the YTD reset....and that is why I asked for clarification how the "reset ad midnight" works. I thought it should exactly prevent this problem. Otherwise it should be enhanced to cover this problem of multiple "restarts" of transmission during a day.

Agree, I noted this problem long ago.
This is why I implemented caching of the YTD value (see above comment from Oct 5 and pull request).

At daychange the offset value needs to get resetted. Due to new implementations this could maybe done elsewhere.
Haven't looked at the code since then.

tbnobody, 2023-12-01:
It's already in my dev branch. But I am currently out of order. Therefor. Please keep patient.

broth-itk, 2023-12-01:
@tbnobody: Patience is not a problem for me. 

Gute Besserung!

pimisen, 2023-12-01:
> It's already in my dev branch. But I am currently out of order. Therefor. Please keep patient.

Also: "Gute Besserung"! :-)

pimisen, 2023-12-01:
Maybe that is of use for anybody: HMS and HM seem to be different. Home Assistant Yield curves look kinda different: 
[image]

Lineflyer, 2023-12-01:
> It's already in my dev branch. But I am currently out of order. Therefor. Please keep patient.

Thanks for the update and all the best!

Lineflyer, 2023-12-05:
@pimisen 
Yes, this visualization shows the "bug". The production is reset to zero on the HMS while it just stays stable on the HM and only resets on the following sunrise. The proposed fix should make your HMS curve look like the others.

tbnobody, 2023-12-16:
This should be fixed with 1de3b4816619b7675fbb9917c287590086eafe10

You can enable this feature per inverter.

kr0815, 2023-12-16:
I installed the new version ( download file was [opendtu-generic_esp32.bin]

I don´t see this new feature?

[image]

tbnobody, 2023-12-16:
Clear your browser cache or use strg+f5 on this page....

Lineflyer, 2023-12-17:
Just installed v23.12.16 and enabled the feature. Can provide feedback tomorrow whether it works with my HMS-1600.

tbnobody, 2023-12-17:
Please also provide the console output at the time the inverter goes offline in the evening.

tbnobody, 2023-12-17:
> maybe this is worth another bug ticket?

Already fixed... will release new version soon. And this code has to run even if the inverter is disabled because the inverter gets maybe disabled just at night. in this case the value has to be set as well...

chris299, 2023-12-17:
@fred777 @tbnobody same here today: although configured, my HMS-1600 got 0 for the day now, because it restarted in the late afternoon.
[image]

[image]

[image]

tbnobody, 2023-12-17:
As mentioned before. Need to see the console log...

chris299, 2023-12-17:
@tbnobody I understand, but how can I catch the important part? I don't know when it restarts, and a lot is coming through, even if the inverters are offline:
[image]

tbnobody, 2023-12-17:
you could either run [putty](https://github.com/tbnobody/OpenDTU/wiki/FAQ#q-how-to-observe-the-esp32-boot-process-by-looking-at-the-serial-console) at the time when the inverter goes offline (should be around the same time every day) and enable the logging function.
Another method  is just using the WebUI. It doesn't matter if there is a lot of logging output. I will crawl through the output.

kr0815, 2023-12-17:
same here didn´t work as expected today

yesterday it worked

Maybe a bug that it works just one time after installation? :-)

broth-itk, 2023-12-17:
If the console output is important, it might be useful to have it sent to a syslog server.
Many years ago I implemented syslog on an embedded device and I can share the code if desired.

Lineflyer, 2023-12-17:
> you could either run [putty](https://github.com/tbnobody/OpenDTU/wiki/FAQ#q-how-to-observe-the-esp32-boot-process-by-looking-at-the-serial-console) at the time when the inverter goes offline (should [...]

Will also try to provide a log next days.
I was not aware I can connect via SSH to the DTU. What port does it listen on?

tbnobody, 2023-12-17:
Ahhhhh I just read the code multiple times again. I think I found the issue.

tbnobody, 2023-12-17:
Does someone of you compile the code?

This would be my guess:

```diff
diff --git a/lib/Hoymiles/src/parser/StatisticsParser.cpp b/lib/Hoymiles/src/parser/StatisticsParser.cpp
index df6e585..0048301 100644
--- a/lib/Hoymiles/src/parser/StatisticsParser.cpp
+++ b/lib/Hoymiles/src/parser/StatisticsParser.cpp
@@ -82,8 +82,6 @@ void StatisticsParser::clearBuffer()
 {
     memset(_payloadStatistic, 0, STATISTIC_PACKET_SIZE);
     _statisticLength = 0;
-
-    memset(_lastYieldDay, 0, sizeof(_lastYieldDay));
 }

 void StatisticsParser::appendFragment(const uint8_t offset, const uint8_t* payload, const uint8_t len)
diff --git a/lib/Hoymiles/src/parser/StatisticsParser.h b/lib/Hoymiles/src/parser/StatisticsParser.h
index 2893cbd..10f06e0 100644
--- a/lib/Hoymiles/src/parser/StatisticsParser.h
+++ b/lib/Hoymiles/src/parser/StatisticsParser.h
@@ -168,5 +168,5 @@ private:
     uint32_t _lastUpdateFromInternal = 0;

     bool _enableYieldDayCorrection = false;
-    float _lastYieldDay[CH_CNT];
+    float _lastYieldDay[CH_CNT] = {};
 };
\ No newline at end of file
```

tbnobody, 2023-12-18:
> But wait - why do you keep setting _lastYieldDay[static_cast<uint8_t>(c)] = 0; in StatisticsParser::endAppendFragment() ?

Because the offset is applied in the line above.
If you used the generic_esp32 build in the past you can use this one for testing...
[firmware.zip](https://github.com/tbnobody/OpenDTU/files/13703060/firmware.zip)

chris299, 2023-12-18:
@tbnobody I just flashed the firmware.zip. I'll report how it will work today afternoon if the inverter restarts....

broth-itk, 2023-12-18:
@chris299 You can just restart the inverter from GUI. This will have the same effect.

tbnobody, 2023-12-18:
> you can just restart the inverter from GUI.

Great idea!! I've just tested it.... and it seems to work as expected.

chris299, 2023-12-18:
> @chris299 You can just restart the inverter from GUI. This will have the same effect.

@tbnobody just restarted the inverter and it seems to work :-)

tbnobody, 2023-12-18:
The fix is included in todays version: https://github.com/tbnobody/OpenDTU/releases/tag/v23.12.18

I am going to close this issue. Please feel free to open it again if there are still issues.

chris299, 2023-12-19:
@tbnobody looks like something else went wrong now: 
the yield was not reset at midnight, although configured and it shows a dot instead of a comma.
(I'm still on the testing version from yesterday, not the 23.12.18 yet)
[image]

tbnobody, 2023-12-19:
If you compile yourself, you could try this one:

```diff
diff --git a/lib/Hoymiles/src/Hoymiles.cpp b/lib/Hoymiles/src/Hoymiles.cpp
index 4cd3ef0..e7b3d26 100644
--- a/lib/Hoymiles/src/Hoymiles.cpp
+++ b/lib/Hoymiles/src/Hoymiles.cpp
@@ -133,10 +133,12 @@ void HoymilesClass::loop()
             if (currentWeekDay != lastWeekDay) {

                 for (auto& inv : _inverters) {
+                    // Have to reset the offets first, otherwise it will
+                    // Substract the offset from zero which leads to a high value
+                    inv->Statistics()->resetYieldDayCorrection();
                     if (inv->getZeroYieldDayOnMidnight()) {
                         inv->Statistics()->zeroDailyData();
                     }
-                    inv->Statistics()->resetYieldDayCorrection();
                 }

                 lastWeekDay = currentWeekDay;
```

chris299, 2023-12-19:
> If you compile yourself, you could try this one:

@tbnobody unfortunately I'm not able to compile it myself....

BTW: Thank you very much for your effort and commitment! 👍 🙏

tbnobody, 2023-12-19:
I am pretty sure that fixes the issue. will be included in todays release.

broth-itk, 2023-12-19:
I'm running the latest version. A smaller inverter had 15Wh daily yield. After manual restart, it value drops to zero:

[image]

tbnobody, 2023-12-19:
> Can I provide more information?

Please tell me your version number. "latest" is not enough.
Please also provide the inverter config of this specific inverter.

broth-itk, 2023-12-19:
@tbnobody: OpenDTU v23.12.18

I've just noticed a new setting "Tagesertragskorrektur", wasn't aware about that.

As far as I was concerned, the code looks to work as designed - thanks!

stefan123t, 2023-12-20:
@kr0815 @chris299 @Lineflyer could you test again with OpenDTU v23.12.18/19 ?
According to @broth-itk the code has been fixed and this issue could probably be closed if you report back too.

chris299, 2023-12-20:
23.12.19 worked fine for me so far @stefan123t

stefan123t, 2023-12-21:
@tbnobody suggest to close this issue as fixed in 23.12.18/19.

Lineflyer, 2023-12-21:
Yes, it also works with my HMS-1600 without further issues now!

closed 2024-01-13 by tbnobody