tbnobody/OpenDTU #1405: [Request] .btn-group margin-top: 0.25rem;

readonly24, 2023-10-03:
### Is your feature request related to a problem? Please describe.

Ist's not a problem, it's just a cosmetic thing

### Describe the solution you'd like

I would insert a margin-top: 0.25rem for .btn-group so the buttons have a little space when there is more than one row (depends on the browser window size)

`.btn-group {
  border-radius: var(--bs-border-radius);
  margin-top: 0.25rem;
}`

[image]

### Additional context

I'm not a web develeoper, maybe there is a better solution for this.

T2023om, 2023-10-04:
Hi, ich nehme mal die deutsche Sprache.
Ich bin auch der Meinung, dass an den 5 Buttons eine Veränderung vorgenommen werden sollte.
Ich würde zuerst die Button im Menü ausblendbar machen, sodass nur noch die jeweils Interessanten bei Bedarf angezeigt werden.
Damit kann man verhindern, dass mehr als eine Zeile für diese bis auf die Sekunden statischen Daten verwendet werden. Das ist besonders für kleinere Bildschirme interessant, siehe oben sogar 3 Zeilen.
Auf meinem ipad  springt zur Zeit die Anzeige von einer auf zwei Zeilen und zurück alle 5 Sekunden, wenn die Aktualisierungszeit anders als  "1"  ist, sodass ich die SW Vers. 23.8.28 wieder verwende.
Ein etwas größerer Abstand zwischen den Buttons muss bei einer Zeile nicht unbedingt sein, bei mehreren erleichtert es dir Bedienung

closed 2023-10-09 by tbnobody

T2023om, 2023-10-10:
> (Issue #1405)
>
>
>
> Closed #1405 <https://github.com/tbnobody/OpenDTU/issues/1405>  as completed
> via
> <https://github.com/tbnobody/OpenDTU/commit/76dfaa7d3b01962c40547547682abf94
> aab14d99> 76dfaa7.
>
> 
> <https://github.com/tbnobody/OpenDTU/issues/1405#event-10594020471> , or
> unsubscribe
> Q2I5AVCNFSM6AAAAAA5RPZC5KVHI2DSMVQWIX3LMV45UABCJFZXG5LFIV3GK3TUJZXXI2LGNFRWC
> 5DJN5XDWMJQGU4TIMBSGA2DOMI> .
> <https://github.com/notifications/beacon/BCZ3442VPX2T32YSXOF22IDX6Q2I5A5CNFS
> M6AAAAAA5RPZC5KWGG33NNVSW45C7OR4XAZNWJFZXG5LFIV3GK3TUJZXXI2LGNFRWC5DJN5XKUY3

Guten Morgen,

 

danke für die Info. Leider ist mein Problem damit nicht gelöst.

 

Anbei zwei 2 Bilder auf einem iPad, die den Sachverhalt zeigen und das macht
keinen Spaß da zuzusehen.

Bei 4 Button passiert die ständige Umschaltung noch nicht, aber bei 5 ist
scheinbar mindestens ein Pixel in der Zeile zu wenig.

 

Ich habe heraus gefunden, falls mein Vorschlag zur Ausblendung einzelner
Button per Menü zu aufwendig ist, eine Änderung  in:

 

<div class="d-flex flex-wrap">

<div style="padding-right: 2em;">HM-1500</div>

<div style="padding-right: 2em;">Seriennummer: 116190746898</div>

<div style="padding-right: 2em;">Aktuelles Limit: 1.500 W | 100 %</div>

<div style="padding-right: 2em;">Letzte Aktualisierung:  vor 1.587 Sekunden
/ 10.10.2023, 13:01:19</div>

</div>

 

das Problem für mich auch so lösen kann. Dazu ist der padding-right  Wert
auf 1 zu ändern.

 

<div style="padding-right: 1em;">Letzte Aktualisierung:  vor 195 Sekunden
</div>

 

Meine Versuche haben keine negativen Auswirkungen auf die Anzeige ergeben
und vielleicht hilft es ja auch anderen Nutzern.

 

Danke für die Unterstützung

commit 76dfaa7d (2023-10-09): Fix #1405: Add some border around icons when in multi line mode