tbnobody/OpenDTU #1407: Setting for "reset daily yield on midnight" missing

Lineflyer, 2023-10-05:
### What happened?

I learned from https://github.com/tbnobody/OpenDTU/commit/ec9af886d54fa7c87f4a75c4baee475392efcff9, that there should now be a setting to enforce the daily yield only to be reset on midnight.

Briefly looking into the commit, I thought that this setting would be an inverter setting.
I could however not find such setting in v23.9.18 in my inverter settings:

[image]

### To Reproduce Bug

Open inverter settings for existing inverter

### Expected Behavior

It should show a setting for "Reset daily yield" (or do I have a problem in understanding that feature?)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

d419fd4

StefanOberhumer, 2023-10-06:
Think you find it under:

EN:
"Menu Settings | Inverter Settings", then "Edit inverter" (Pencil-Icon), within the popup on the top select "Advanced"

DE:
"Menü Einstellungen | Wechselrichter", dann "Wechselrichter bearbeiten" (Stiftsymbol), dann im oben Popup auf "Erweitert"

StefanOberhumer, 2023-10-06:
Sorry - seems I missed you screenshot!

I installed also https://github.com/tbnobody/OpenDTU/commit/d419fd479461a933ffa13420e5448e0d45cec4b6 (v23.9.18) and I have the feature option available.

I had a similar problem with an other feature last time. 
Resetting Browser-Cache helped in my case.

Lineflyer, 2023-10-06:
That did the trick, it needed a browser cache reset. Problem solved.

closed 2023-10-06 by Lineflyer