tbnobody/OpenDTU #1411: [Request] Add adjustable interval time for powermeter http request

ntfrnd, 2023-10-07:
### Is your feature request related to a problem? Please describe.

I see in the console that the HTTP request is made every 10 sec. to read the (Tasmota-) powermeter.
This seems very slow to me, especially since the request via web browser is easily possible every second.

### Describe the solution you'd like

It would be helpful to make the request interval adjustable for the http(s)+json - powermeter from e.g. 1....30 sec. In one second steps.

tbnobody, 2023-10-07:
Are you sure you are using opendtu and not opendtu-onbattery? Opendtu does not read any power meter.

ntfrnd, 2023-10-07:
Oops, you are right :-)
I use both : OpenDTU since one year 👍 and now I test  OpenDTU_onBattery. I got the wrong browser tab, sorry. Can you delete this topic?

closed 2023-10-08 by tbnobody