tbnobody/OpenDTU #1421: Inverter Firmware update

ThomasCr, 2023-10-11:
### Describe the solution you'd like

I have read, that it is possible to request firmware update for Homiles Inverters per Mail if you have a DTU from the company. It would be possible to capture that process to a) know how to transfer the firmware to the inverter b) to get the firmware file or maybe also capture the download address.

I know every update should be considered carefully - but it would be a good thing to do it with OpenDTU.

stefan123t, 2023-12-20:
The Firmware Update has to be first captured using an original DTU WLite / Pro for us to be able to implement the necessary commands. Even though there is strong evidence in the Hoymiles Source Code of the DTU Pro that the [DOWN_DAT `0x0a`](https://github.com/lumapu/ahoy/wiki/Protocol#wie-wird-das-down_pro-0x0e--down_dat-0x0a-verwendet) and INIT `0xff` Commands are used in a so called MultiRequest, we still have no logs from the communication between the GD32 and the NRF24L01+ Module in the DTU WLite / Pro when such a Firmware Upgrade is done.

The DOWN_PRO `0x0e` to apply a new Grid Profile is also a MultiRequest Command, as it has to send more than the usual 12 Bytes payload to the inverter too. The 141 byte payload (being split into 12 packets of max. 12 bytes) of a Grid Profile is probably a safer target to be implemented thoroughly before we should try the exciting Firmware Upgrade.

Ayo91, 2024-07-04:
Any news on that? Will it be possible to update inverters?

stefan123t, 2024-07-15:
@Ayo91 IMHO we first need to implement #900 in order to have the grid profile update understood and working for:
1. known grid profiles
2. (optional) custom modified grid profile values

Before implementing the actual Firmware upgrade, we should trace the process for a known FW image by Hoymiles.

Though that would need to be traced first by someone with an Hoymiles DTU, eg. by nivadis/Discord.

stefan123t, 2024-08-15:
@ThomasCr any progress on implementing the Grid Profile update first, I think we have sufficiently traced it to make it work.
Regarding the Firmware upgrade I heard from users of the DTUBI (HMS-WiFi models have a built-in DTU) that it may take an hour (or more) to download the latest firmware from the Hoymiles CDN and upload it to the inverter, though they had to start over three or more times because some checksum may have failed to match. 
Also Note that HMS-WiFi models do not use a radio connection between DTU and inverter, i.e. they use only WiFi between your Router and the Internet with the S-Miles Cloud to download the firmware image and push it via a local UART connection from the DTUBI to the inverter. Hence there should be less / little transmission errors IMHO.

stefan123t, 2024-10-20:
Huch ich sollte mal die Logs vom [Firmware Update](https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-1928827458) von @broth-itk auswerten und wie versprochen dokumentieren.

Also nach allem was wir wissen beginnt das Firmware Update mit dem Command 0x0e DOWN_PRO von der DTU, also alle Zeilen mit `b'7e0e...'`.

Hier die beiden Log Files von @broth-itk mit den aufgeräumten Zeilen, die Kommandos auf dem UART / der Seriellen Schnittstelle zwischen MCU und NRF / CMT Modul beginnen immer mit `7e` und enden auf `7f`, wenn das nicht der Fall ist, dann wurde beim Logging etwas umgebrochen, was nicht sein muß / sein sollte.

[Firmware Upgrade.zip](https://github.com/user-attachments/files/17452123/Firmware.Upgrade.zip)
[log_1_regex.txt](https://github.com/user-attachments/files/17452119/log_1_regex.txt)
[log_2_regex.txt](https://github.com/user-attachments/files/17452120/log_2_regex.txt)

Ich habe den folgenden Regex dazu verwendet `(?<!7f)'\n[AB]: [0-9-]{10} [0-9:.]{15}: b'(?!7e)`, vielleicht kann man das noch in das Python Script einarbeiten ?

<details><summary>Firmware Update</summary>
<p>

```hex
A: 2024-02-04 10:28:19.619589: b'7e0e8072978580729785810600001110121000272ba574577f'
A: 2024-02-04 10:28:19.699604: b'7e0e80729785807297858102000010000df9c1a87f'
A: 2024-02-04 10:28:19.811663: b'7e0e80729785807297858102000010000df9c1a87f'
A: 2024-02-04 10:28:19.907703: b'7e0e80729785807297858102000010000df9c1a87f'
A: 2024-02-04 10:28:20.003693: b'7e0e80729785807297858102000010000df9c1a87f'
A: 2024-02-04 10:28:20.099804: b'7e0e80729785807297858102000010000df9c1a87f'
A: 2024-02-04 10:28:20.195817: b'7e0e807297858072978581020000040008fe413e7f'
A: 2024-02-04 10:28:20.275879: b'7e0e807297858072978581020000040008fe413e7f'
A: 2024-02-04 10:28:20.355931: b'7e0e80729785807297850118300000000830100008ff7d5e92aff92d6f7f'
A: 2024-02-04 10:28:20.403932: b'7e0e807297858072978582272b07e7025d040010121000c1e30b7f'
A: 2024-02-04 10:28:20.483982: b'7e0e807297858072978581043010000048fedb46f3737f'
A: 2024-02-04 10:28:20.564106: b'7e0e80729785807297850120301800fcf70008fcf70008fcf70008047f'
A: 2024-02-04 10:28:20.612065: b'7e0e807297858072978502fcf70008fcf70008fcf70008fcf700080c7f'
A: 2024-02-04 10:28:20.676063: b'7e0e807297858072978583fcf70008f513687f'
A: 2024-02-04 10:28:20.756097: b'7e0e80729785807297850120302800fcf70008fcf70008fcf70008347f'
A: 2024-02-04 10:28:20.804123: b'7e0e807297858072978502fcf70008fcf70008fd830008fd8d0008027f'
A: 2024-02-04 10:28:20.852142: b'7e0e807297858072978583faef0008b3c1e27f'
A: 2024-02-04 10:28:20.932206: b'7e0e80729785807297850120303800fd1f0008fb710008fce30008587f'
A: 2024-02-04 10:28:20.980229: b'7e0e807297858072978502fca70008fdb50008fdbf0008fdc90008697f'
A: 2024-02-04 10:28:21.028247: b'7e0e807297858072978583fdd300087a9e4f7f'
A: 2024-02-04 10:28:21.108318: b'7e0e80729785807297850120304800fddd0008fde70008fdf10008697f'
A: 2024-02-04 10:28:21.156344: b'7e0e807297858072978502fdfb0008fe050008fd970008fda10008c77f'
A: 2024-02-04 10:28:21.204364: b'7e0e807297858072978583fdab0008af4d317f'
A: 2024-02-04 10:28:21.284397: b'7e0e80729785807297850120305800eff60008fcf70008fa450008e27f'
A: 2024-02-04 10:28:21.332410: b'7e0e807297858072978502fe2d0008fe370008fcf70008fd790008997f'
A: 2024-02-04 10:28:21.396444: b'7e0e807297858072978583fe0f000861899c7f'
A: 2024-02-04 10:28:21.476479: b'7e0e80729785807297850120306800fb850008fb990008fbad0008357f'
A: 2024-02-04 10:28:21.524483: b'7e0e807297858072978502fbc10008fbd50008fbe90008fbfd00080c7f'
A: 2024-02-04 10:28:21.572526: b'7e0e807297858072978583fcf7000859af787f'
A: 2024-02-04 10:28:21.652522: b'7e0e80729785807297850120307800fb7b0008fb8f0008fba30008c37f'
A: 2024-02-04 10:28:21.700545: b'7e0e807297858072978502fbb70008fbcb0008fbdf0008fbf300085c7f'
A: 2024-02-04 10:28:21.748573: b'7e0e807297858072978583fcf700083738817f'
A: 2024-02-04 10:28:21.828624: b'7e0e80729785807297850120308800fb490008fb530008fb670008197f'
A: 2024-02-04 10:28:21.876632: b'7e0e807297858072978502fcf70008fcf70008fcf70008fcf700080c7f'
A: 2024-02-04 10:28:21.924654: b'7e0e807297858072978583fcf700089e03137f'
A: 2024-02-04 10:28:22.004705: b'7e0e80729785807297850120309800fc070008fc110008fcf70008927f'
A: 2024-02-04 10:28:22.052747: b'7e0e807297858072978502fcf70008fa950008fa9f0008fcf70008067f'
A: 2024-02-04 10:28:22.116808: b'7e0e807297858072978583fcf700083165da7f'
A: 2024-02-04 10:28:22.196849: b'7e0e8072978580729785012030a800fd470008fd510008fd5b00080f7f'
A: 2024-02-04 10:28:22.244873: b'7e0e807297858072978502fd650008fcf70008fcf70008fcf700089f7f'
A: 2024-02-04 10:28:22.292896: b'7e0e807297858072978583fcf700081f7ced7f'
A: 2024-02-04 10:28:22.372957: b'7e0e8072978580729785012030b800fb0d0008fb170008fb2100086f7f'
A: 2024-02-04 10:28:22.420982: b'7e0e807297858072978502fb2b0008fb350008fb3f0008fcf70008dd7f'
A: 2024-02-04 10:28:22.469002: b'7e0e807297858072978583fcf70008baab9f7f'
A: 2024-02-04 10:28:22.549035: b'7e0e8072978580729785012030c800fc890008fc7d5f0008fc9d0008487f'
A: 2024-02-04 10:28:22.597133: b'7e0e807297858072978502fc930008fcf70008fcf70008fcf70008687f'
A: 2024-02-04 10:28:22.645121: b'7e0e807297858072978583fcf70008b8e5d37f'
A: 2024-02-04 10:28:22.725128: b'7e0e8072978580729785012030d800fd290008fd330008fcf70008de7f'
A: 2024-02-04 10:28:22.773139: b'7e0e807297858072978502fcf70008fa810008fa8b0008fcf70008067f'
A: 2024-02-04 10:28:22.837177: b'7e0e807297858072978583fcf700086dc82b7f'
A: 2024-02-04 10:28:22.917209: b'7e0e8072978580729785012030e800fa3b0008fa1d0008fa270008047f'
[... cut ...]
A: 2024-02-04 10:38:22.536419: b'7e0e80729785807297858100000001e4c1ab7f'
A: 2024-02-04 10:38:22.632439: b'7e0e80729785807297858100000001e4c1ab7f'
A: 2024-02-04 10:38:22.728493: b'7e0e80729785807297858100000001e4c1ab7f'
A: 2024-02-04 10:38:22.824572: b'7e0e80729785807297858100000001e4c1ab7f'
A: 2024-02-04 10:38:22.936679: b'7e0e80729785807297858100000001e4c1ab7f'
A: 2024-02-04 10:38:23.032673: b'7e0e80729785807297858100000001e4c1ab7f'
A: 2024-02-04 10:38:23.128725: b'7e0e80729785807297858100000001e4c1ab7f'
```
</p>
</details>