tbnobody/OpenDTU #1428: [Request] mDNS / DNS-SD / Bonjour to announce opendtu-<hash>.local Name

stefan123t, 2023-10-14:
### Is your feature request related to a problem? Please describe.

Users without a proper DHCP + DNS Server eg Speedport 723V Typ B routers may not be able to figure out the IP assigned to OpenDTU by the routers DHCP after some reboot.

### Describe the solution you'd like

Announce the name opendtu-<hash>.local via mDNS in the local network. That way users can access the DTU more easily.

To be honest I change my name to opendtu or opendtu-fusion in my FritzBox but other / novice users may not have that option in their routers.

### Describe alternatives you've considered

Adding a static mapping to the DHCP Server to assign a static IP and Name to the ESP MAC address. But this feature is not available in all routers. I have flashed freetz over my old and nowadays outdated FritzBox 3xxx just to have a proper DHCP with this functionality. But not all users may be willing / able to flash custom firmware on their routers to workaround this issue ;)

### Additional context

See PR #695 for the necessary changes and the AhoyDTU code where it is implemented too

stefan123t, 2023-10-14:
By the way get well soon first!

CommanderRedYT, 2023-10-20:
If you want I can submit a new PR with

- Webapp mDNS Enable/Disable
- Announce the name so opendtu-xxxxx.local will be possible
- Add records that show devices that OpenDTU has a website 
- Custom record so one can scan the network for OpenDTU auto discovery.

Open questions would be:

- Enabled by default?

stefan123t, 2023-10-20:
@CommanderRedYT the name of your OpenDTU is already (pre)set under **Settings > Network Settings**
Hostname: `OpenDTU-%06X`

Here `%06X` is replaced with the last 3-bytes/6-chars of your ESP32 chips MAC-Address in Hex representation.
I regularly change this to either opendtu or opendtu-fusion to distinguish my two DTU targets, 
though I know other developers do have more than just two DTUs.
I enjoy the simplicity of speaking names without hex chars, which I won't remember easily :older_man:.

The question for me is more how does that work when setting the Hostname in my case, even though mDNS is not used in OpenDTU yet ?

CommanderRedYT, 2023-10-20:
> The question for me is more how does that work when setting the Hostname in my case, even though mDNS is not used in OpenDTU yet ?

Hostname Solution only works when DNS and DHCP are communicating afaik

CommanderRedYT, 2023-10-20:
I have added my PR here: https://github.com/tbnobody/OpenDTU/pull/1441

tbnobody, 2023-11-15:
Implemented in 7c875187a8357e73df17a8f38b65e477bdf657c3

closed 2023-11-15 by tbnobody