tbnobody/OpenDTU #1455: OpenDTU on Olimex ESP32-POE-ISO - No Connection

commyonline, 2023-10-29:
### What happened?

Hallo zusammen,
ich bekomme mit dem Olimex-Board keine Verbindung zum HM-1500. Der WR ist gebraucht gekauft und hat ein Zertifikat für die Drosselung auf 600W dabei.
Das Board war nur, ist aktuell geflashed und hat die .json File als Konfig geladen. Die Belegung mit dem NRF Board ist entsprechend angepasst.
Im Ersten Versuch hatte ich in der Belegung einen Fehler (MISO und CE vertauscht) kann dadurch etwas beschädigt sein?

Wenn ich den WR mittels 2 Modulen provisorisch auf der Terrasse starte leuchtet die grüne LED im 4sec Rhythmus. Startet ich nun die OpenDTU Hardware blinkt der WR im 2sec Rythmus. (Status DTU erkannt, oder?)
Allerdings sagt die OpenDTU, dass keine Verbindung hergestellt werden konnte. Bzw. Das diese seit dem Boot nicht mehr stattgefunden hat:

[image]

Allerdings bekomme ich keine Daten und keine Verbindung auf der OpenDTU angezeigt.
In der Konsole läuft Folgendes:
14:23:12.997 > All missing
14:23:12.997 > Nothing received, resend whole request
14:23:12.997 > TX SystemConfigPara Channel: 61 --> 15 83 77 60 84 80 12 17 60 80 05 00 65 3E 5C B9 00 00 00 00 00 00 00 00 89 75 27 
14:23:13.248 > RX Period End
14:23:13.248 > All missing
14:23:13.248 > Nothing received, resend whole request
14:23:13.248 > TX SystemConfigPara Channel: 75 --> 15 83 77 60 84 80 12 17 60 80 05 00 65 3E 5C B9 00 00 00 00 00 00 00 00 89 75 27 
14:23:13.498 > RX Period End
14:23:13.498 > All missing
14:23:13.498 > Nothing received, resend whole request
14:23:13.498 > TX SystemConfigPara Channel: 3 --> 15 83 77 60 84 80 12 17 60 80 05 00 65 3E 5C B9 00 00 00 00 00 00 00 00 89 75 27 
14:23:13.702 > RX Period End
14:23:13.702 > All missing
14:23:13.702 > Nothing received, resend count exeeded
14:23:13.702 > Fetch inverter: 116183776084
14:23:13.702 > Request SystemConfigPara
14:23:13.755 > TX RealTimeRunData Channel: 23 --> 15 83 77 60 84 80 12 17 60 80 0B 00 65 3E 5C C1 00 00 00 00 00 00 00 00 45 78 90 
14:23:13.758 > Websocket: [/livedata][5] disconnect

### To Reproduce Bug

Auf dem Olimex habe ich ein NRF Board aus der Bucht mit ca. 40mm Leitungen gelötet. Verbindungen mehrfach nachgeprüft. Der NRF wird im OpenDTU erkannt und Status ist konfiguriert/verbunden/nRF24L01+.
Zusätzlich auf Anregung aus anderen Issus habe ich es außerdem ohne POE getestet und die Hardware per USB versorgt. Auch ein Keramikkondensator 1uF am NRF Board hilft nichts.

[image]

### Expected Behavior

Hier die Systemdaten. Vielleicht hat jemand einen Tipp oder sieht das Problem.

[image]
[image]
[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.10.9

coltc50, 2023-10-30:
Also bei mir klappt die Verbindung auch nur mit einem alten Huawei-Handy-Netzteil per USB. Ich habe allerdings einen Adafruit HUZZAH32 - ESP32 Feather am NRF24L01+ hängen. Habe schon alle möglichen Netzteile und USB-Kabel durch. Komischerweise ging es auch am USB Port einer alten FB7340. Ich habe aber am Gartenhaus auf dem die Module liegen eine FB7362 in Betrieb und hier wird über den USB keine Verbindung aufgebaut. Kondensatoren habe ich auch schon durchprobiert...

Sorry, hilft Dir zwar nicht weiter, aber wie gesagt probiere mal alle möglichen Handynetzteile durch. Mit Glück kannst Du mit einem eine Verbindung herstellen.
Mich würde mal interessieren warum das Funkmodul da so wählerisch ist...

broth-itk, 2023-10-30:
Das ist ein 100nF Kondensator. 104 = 10 + 4 Nullen = 10 0000 = 100000pF = 100nF = 0.1µF
Bitte Elko parallel schalten und eine kleine Induktivität (z.B. Ferrit) in die Stromversorgung des Moduls (Zuleitung).

Entweder der DC/DC Wandler auf dem Board macht Störungen und hindert den Empfang oder das Modul crasht weil es nicht genügend Energie zum Senden bekommt.

Vorausgesetzt natürlich, dass Anschluss und Konfiguration passt.

turrican944, 2023-11-01:
Moin

[image]

Nehme mal einen etwas größeren Kondensator. Ich hatte so eine olimex auch Mal am Laufe und die lief problemlos allerdings hatte ich 100uF oder so.
Da ich die HM Wechselrichter durch einen HMS getauscht habe, hab ich den olimex umgebaut, funktioniert mit dem hms aber genauso so.

turrican944, 2023-11-03:
Moin
Die habe ich fertig lassen bei JLCPCB.
Have noch welche liegen Schreib mir Mal ne Email.

Gerd-Watzmann, 2023-11-04:
Hallo, ich habe auch das Problem dass der Olimex ESP32 POE ISO über Ethernet angebunden die Wechselrichter nicht findet.

Er findet sie aber, wenn ich in der Netzwerkkonfiguration auf "save" klicke (ohne irgendetwas zu verändern). Für einige Minuten sind die Daten dann vollständig da, bevor die Verbindung zu den Wechselrichtern wieder verloren geht.

Der Kondensator spielt in diesem Fall keine Rolle (die Verbindung funktioniert ja grundsätzlich einwandfrei), der Fehler liegt irgendwo in der Software.

Dieses Problem habe auch andere User, und es gibt irgendwo einen Thread dazu.

Ist es evtl. das?

commyonline, 2023-11-06:
Hi @Gerd-Watzmann,

der "Trick" Save im Netzwerk-Menü sorgt bei mir nicht für eine kurzfristige Verbindung. Allerdings vermute ich auch, dass es nicht (nur) ein Hardware Problem ist. Bei jedem Neustart der Hardware scheint er kurz die Verbindung zum WR gehabt zu haben.
In den kommenden Tagen teste ich noch etwas mit einer Ferritperle, Kondensator und dem Verdrillen von MOSI/MOSI.
Alternativ bestelle ich noch ein anderes Funkmodul.

Gerd-Watzmann, 2023-11-06:
Ich hatte das selbe (also nicht nur das gleiche) Funkmodul an der gleichen Stelle mit der gleichen USB-Stromversorgung, aber einem ESP32 via WLAN im Betrieb - das lief wochenlang ohne die allergeringsten Probleme durch. Daher hätte ich zumindest in meinem Fall zunächst die Software im Verdacht.

turrican944, 2023-11-07:
Hast du es mit einem dicken Elko nun probiert? Wenn nein mach Mal einen ran.

broth-itk, 2023-11-07:
Auf dem Board ist ja schon ein anderer DC/DC Wandler der aus den 50V PoE die Versorgung für den Controller erzeugt.

Wie schon vor einer Woche geschrieben macht es ggf. Sinn die Versorgung des Funkmoduls über einen LC Filter zu "glätten".

Hilfsweise das Funkmodul mit einem externen Netzteil versorgen. GND verbinden nicht vergessen...

Gerd-Watzmann, 2023-11-07:
Nachdem ich den Puffer-Elko auf 470 uF vergrößert habe, läuft es jetzt zumindest durch, wenn auch mit regelmäßigen DTU-Fehlern. Sendeleistung auf max., Stromversorgung noch über USB (verschiedene Netzteile und Kabel ausprobiert). 

Also scheint das tatsächlich der Weg zu sein, den ich weiter verfolgen muss. Wie hieß es schon in den 1970ern in der Elektor: "jedes Gerät ist so gut wie seine Speisung" ;-)

Besten Dank erstmal soweit!

Verwirrend war natürlich die Sache mit dem Sichern der Netzwerkeinstellungen, s.a.:

https://github.com/tbnobody/OpenDTU/issues/450

broth-itk, 2023-11-07:
Bitte Hinweise auf Seite 7 des Datenblatts von CMT2300A beachten, gilt aber auch für andere Funkmodule:

https://www.cdebyte.com/pdf-down.aspx?id=429

[image]
[image]
[image]
[image]
[image]

Gerd-Watzmann, 2023-11-07:
Das ist schon alles klar soweit, sind ja die üblichen Hersteller-Disclaimer ;-) Es waren halt zwei Dinge etwas verwirrend: das selbe (nicht nur das gleiche) Funkmodul machte an einem normalen ESP32 Board keinerlei Probleme, an der selben Stelle und mit der selben Stromversorgung über USB. Ob nun mit oder ganz ohne (!) Puffer-Elko. Und da war dann die Sache mit dem "speichern"-Button der Netzwerkeinstellungen, der für jeweils 2 Minuten einen einwandfreien, sauberen Funkkontakt ermöglichte (s. diesbezüglichen Thread). Das hat _schon_ ein bisschen was von Voodoo ;-)

broth-itk, 2023-11-07:
Die Hinweise sind durchaus ernst gemeint und sollten auch berücksichtigt werden. Ignoranz führt zu diversen Fehlern.

Auf dem ESP32-POE-ISO Board gibt es einen DC/DC Wandler der Störungen verursacht, leitungsgekoppelte und elektromagnetische durch den internen Übertrager.

Gerd-Watzmann, 2023-11-07:
Wie gesagt, Stromversorgung in beiden Fällen über USB. 
POE nehme ich erst in Angriff, wenn alles sauber läuft. 
Und nochmal: danke, jetzt alles gut ;-)

commyonline, 2023-11-07:
Das stimmt und dessen bin ich mir bewusst. Allerdings lässt sich das Board auch per USB versorgen ("ohne PoE") und auch dort arbeitet das Funkmodul nicht. Auch "ohne Ethernet", mittels Übertragung im ESP-eigenen WLAN kommt keine Verbindung zum WR zustande. Einen Elko hatte ich bereits dran (220uF).
Ich werde nochmal das Board wechseln, Leitungen kürzen, 100nF+Elko anlöten, Längsinduktivität in VCC einfügen und MISO/MOSI verdrillen.
Melde mich danach.

commyonline, 2023-11-09:
Hallo zusammen,
ich habe nochmal folgendes umgebaut und getestet: [Ferrit-Perle](https://www.reichelt.de/emv-ferrit-entstoerfilter-axial-21-ohm-10-mhz-bead-8-21-p105526.html) in Serie zu VCC, auf dem NRF-Board 100nF Kerko + 220uF Elko. Zusätzlich an der Stiftleiste des Olimex VCC 1uF Kerko und Leitungen VCC+GND sowie MISO+MOSI verdrillt. Fotos folgen.
Das Modul läuft über USB (Powerbank) und Zugang erfolgt über ESP-eigenes default-WLAN. NRF+-Chip ist erkannt und Konfiguriert. Letzte Verbindung zum WR ist jedesmal zum Bootzeitpunkt. Siehe folgende Screenies. Die Konsole gibt nun allerdings weniger aus als bisher.
Eine Sache ist mir noch aufgefallen: Beim Speichern der Netzwerkeinstellungen kommt folgene Warnung: apiresponse.undefined (siehe Foto)

Habe nun erstmal ein fertiges OpenDTU bestellt um meine Anlage zu betreiben. Über Einschätzungen wäre ich trotzdem dankbar.

[image]

[image]

[image]

Gerd-Watzmann, 2023-11-10:
Man sollte das Thema "Antenne" / Polarisation auch nicht völlig unterschätzen. Ich z.B. habe nur ca. 3m Abstand zwischen den beiden Wechselrichtern und der DTU, aber es liegt eine Stahlbetondecke dazwischen. Ich bekomme mit aufgesteckter Stummelantenne in aufrechter Position an einer bestimmten Stelle eine zuverlässige Verbindung. Eine schräge oder liegende Antenne funktioniert nicht wirklich, und mit den NRF-Boards mit Printantenne kommt grundsätzlich überhaupt keine Verbindung zustande. Das selbe Verhalten mit unterschiedlichen ESP-Boards, sowohl unter AhoyDTU wie auch OpenDTU, und Sendeleistung auf "max". Letzteres hilft wohl nur in einer Richtung - oder gibt es eine Möglichkeit, die Sendeleistung der Wechselrichter zu beeinflussen?

commyonline, 2023-11-15:
Hallo nochmal.
Ich kann einen Erfolg berichten: Nachdem ich ein anderes Funkmodul (SparkFun Transceiver Breakout, nRF24L01+) verbaut habe funktioniert die Verbindung zum WR.
Dieses Modul hat einen eigenen Spannungsregler (LDO). Dies scheint der Entscheidende Punkt zu sein.

Weiterer Funfakt: Mit dem "alten Funkmodul" konnte ich zu keinem Zeitpunkt eine Verbindung aufbauen. Allerdings konnte ich erfolgreich Power-Limits an den WR senden. Auch wenn die Rückmeldung nicht dargestellt wurde - der WR hatte das Limit zuverlässig übernommen.

Gerd-Watzmann, 2023-11-15:
Ja, die Funkmodule ... YMMV kann man da nur sagen. Und das auch ganz ohne die Problematik der Spannungsversorgung - die kommt "on top", aber die kann man sich ja mit den Elkos und ggf. anderen Maßnahmen entschärfen.

BTW: Kann man den Sendepegel am WR eigentlich auch regulieren? Es ist ja nur die halbe Miete, wenn man ihn an der DTU auf "max" stellt ...

closed 2025-04-19 by github-actions[bot]