tbnobody/OpenDTU #1468: need reboot to get connection again every morning

001oc, 2023-11-02:
### What happened?

opendtu can't establish communication to the inverters next morning

### To Reproduce Bug

Hi guys,

the whole summer everything works great but now, since some weeks, I have to reboot or sometimes cold reboot (unplug esp32 and replug it again) to get my opendtu up and running every morning.

Opendtu is running and tries to communicate to the inverters but isn't able to establish a communication.

The symptoms are that there is no answer from inverter.

What I do to reinitiate communication is switch off mqtt temporarily and reboot (mostly unplug esp and replug it). After this procedure opendtu starts communication with inverter latest after 2 minutes, mostly much faster, sometimes nearly immediatly.

Without switching off mqtt and simply cold rebooting helps not very often, maybe after 10-20 attempts it comes up again. But that is not reproducable. What helps for sure is switching off mqtt.

I have played arround with different combinations of polling times from inverter and mqtt but nothing helps. I even played arround with many combinations of cutting off communication at night or let it runn the whole night. Nothing helps but switching of mqtt temporarily.

In my setting mqtt (or better opendtu) is trying to set limit_nonpersistent_absolute the whole time as I am using opendtu and node red in "backoffice" to realize a zero feed-in. Of course I could (and sometimes do) control sending mqtt events at node red side. But I think it is more logical, that the "problem" should be solvend at opendtu side. Node red shouldnt be in charge of solving a problem that opendtu or my inverters have with constantly beeing bothered by mqtt events that couldnt be proceees at "client-side".

I think it would be helpful if opendtu stops processing mqtt events from "outside2 as longs as there is no stable connection to any of the inverters. I think opendtu knows if there is a stable connection to any of the inverters it should talk to. If there is no responce from any of the inverters opendtu could simply ignore any external mqtt eventts as it will not bbring it uo anyway.

I am quit sure that this will help alot to prevent opendtu failing at its "morning routine". I have even tested switching of mqtt manually at night and there was no problem at all for opendtu to find its inverters next morning and after switching on mqtt at the next morning (after opendtu established communication to its inverters) everythin works easy and smootly.

It is a bit strange, that the problem wasnt there in summer and startet some weeks ago - but maybe autum has nothing to do with teh symptoms .-)

By the way, my opendto will talk to 2 * HMS2000 - if that seems to be relevant.

### Expected Behavior

after loosing connection to inverters stop processing mqtt events until stable connection is reestablished (next morning)

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

23.10.9 (latest)

### Relevant log/trace output

```Shell
I will send it tommorow
```

001oc, 2023-11-03:
23:39:41.101 > Nothing received, resend whole request
23:39:41.101 > TX AlarmData 865.00 MHz --> 15 80 72 47 92 80 16 37 68 80 11 00 65 44 25 25 00 00 00 00 00 00 00 00 86 C1 0C 
23:39:42.249 > RX Period End
23:39:42.249 > All missing
23:39:42.249 > Nothing received, resend whole request
23:39:42.249 > TX AlarmData 865.00 MHz --> 15 80 72 47 92 80 16 37 68 80 11 00 65 44 25 25 00 00 00 00 00 00 00 00 86 C1 0C 
23:39:42.975 > RX Period End
23:39:42.975 > All missing
23:39:42.975 > Nothing received, resend count exeeded
23:39:43.046 > TX SystemConfigPara 865.00 MHz --> 15 80 72 47 92 80 16 37 68 80 05 00 65 44 25 25 00 00 00 00 00 00 00 00 92 D5 18 
23:39:44.193 > RX Period End
23:39:44.193 > All missing
23:39:44.193 > Nothing received, resend whole request
23:39:44.193 > TX SystemConfigPara 865.00 MHz --> 15 80 72 47 92 80 16 37 68 80 05 00 65 44 25 25 00 00 00 00 00 00 00 00 92 D5 18 
23:39:44.250 > RX Period End
23:39:44.250 > All missing
23:39:44.250 > Nothing received, resend whole request
23:39:44.250 > TX SystemConfigPara 865.00 MHz --> 15 80 72 47 92 80 16 37 68 80 05 00 65 44 25 25 00 00 00 00 00 00 00 00 92 D5 18 
23:39:44.281 > RX Period End
23:39:44.281 > All missing
23:39:44.281 > Nothing received, resend whole request
23:39:44.281 > TX SystemConfigPara 865.00 MHz --> 15 80 72 47 92 80 16 37 68 80 05 00 65 44 25 25 00 00 00 00 00 00 00 00 92 D5 18 
07:02:41.929 > RX Period End
07:02:41.929 > All missing
07:02:41.929 > Nothing received, resend whole request
07:02:41.929 > TX ActivePowerControl 865.00 MHz --> 51 80 72 47 92 80 16 37 68 81 0B 00 2A F8 00 00 89 88 E6 
07:02:43.721 > Received MQTT message on topic: MS2054/solar/116480727391/cmd/limit_nonpersistent_absolute
07:02:43.721 > Limit Non-Persistent: 1100 W
07:02:43.721 > Received MQTT message on topic: MS2054/solar/116480724792/cmd/limit_nonpersistent_absolute
07:02:43.721 > Limit Non-Persistent: 1100 W
07:02:44.006 > RX Period End
07:02:44.006 > All missing
07:02:44.006 > Nothing received, resend count exeeded
07:02:44.073 > TX ActivePowerControl 865.00 MHz --> 51 80 72 73 91 80 16 37 68 81 0B 00 2A F8 00 00 89 88 D1 
07:02:46.103 > RX Period End
07:02:46.103 > All missing
07:02:46.103 > Nothing received, resend whole request
07:02:46.103 > TX ActivePowerControl 865.00 MHz --> 51 80 72 73 91 80 16 37 68 81 0B 00 2A F8 00 00 89 88 D1 
07:02:49.043 > RX Period End
07:02:49.043 > All missing
07:02:49.043 > Nothing received, resend whole request
07:02:49.043 > TX ActivePowerControl 865.00 MHz --> 51 80 72 73 91 80 16 37 68 81 0B 00 2A F8 00 00 89 88 D1 
07:02:51.063 > RX Period End
07:02:51.063 > All missing
07:02:51.063 > Nothing received, resend whole request
07:02:51.063 > TX ActivePowerControl 865.00 MHz --> 51 80 72 73 91 80 16 37 68 81 0B 00 2A F8 00 00 89 88 D1 
07:02:53.374 > RX Period End
07:02:53.374 > All missing
07:02:53.374 > Nothing received, resend whole request
07:02:53.374 > TX ActivePowerControl 865.00 MHz --> 51 80 72 73 91 80 16 37 68 81 0B 00 2A F8 00 00 89 88 D1 
07:02:53.478 > Received MQTT message on topic: MS2054/solar/116480727391/cmd/limit_nonpersistent_absolute
07:02:53.478 > Limit Non-Persistent: 1100 W
07:02:53.531 > Received MQTT message on topic: MS2054/solar/116480724792/cmd/limit_nonpersistent_absolute
07:02:53.531 > Limit Non-Persistent: 1100 W
07:02:55.189 > RX Period End
07:02:55.189 > All missing
07:02:55.189 > Nothing received, resend count exeeded

till 23:39:44.281 everthing work "normal", after 07:02:41.929 chaos starts. This loop never stops and prevents opendtu to establish a working connection to the inverters.

broth-itk, 2023-11-03:
Do you poll at night?

It looks like there is a big gap between transmission and reception (timeout):

23:39:44.281 > TX SystemConfigPara 865.00 MHz --> 15 80 72 47 92 80 16 37 68 80 05 00 65 44 25 25 00 00 00 00 00 00 00 00 92 D5 18
07:02:41.929 > RX Period End

Do you use a big capacitor to buffer the supply of the rf module?

001oc, 2023-11-03:
@broth-itk thanks for the reply.

That was even my first thought too, if it could be the power supply. I have tried a couple of at least 10 different power supplies and ended up, for being sure that this wouldnt be the problem, with a switching one that could deliver 100 watts 5V/20A. On the board itself I started with a 100uF end endet with a 1F Supercap with low ESR, witch makes the board run for aprox. two minutes with disconnected power supply. I think that should meet all criterias for preventing brown-outs :-).

I think that is not the problem. For me it looks like that the inverters are not willing to take power limit commands before being awake. For me it looks like that they shut down when getting a priorized power command before having stable connection to opendtu. If I prevent mqtt power limiting commands manually the inverter starts communication with opendtu every monring with ease. After the communication between opendtu and the inverters (without mqtt) is established, I can start mqtt and opendtu runs smootly even with power regulating commands. When I leave mqtt running the whole night, normaly opendtu is not able to start communication next morning.

stefan123t, 2023-11-03:
I understand that you are trying to continuously send power limit commands to the inverter. Even if the inverter is currently unreachable, due to low power / sun light conditiona.

Is there no status flag for the inverter on the MQTT topics which you can easily use as a condition for sending limit commands from NodeRed in the first place?

I agree that OpenDTU could do the thinking for you, because if the inverter receives the *same* power limit again, it will seize communication with the DTU for several minutes. Dunno what happens if you send it commands continuously during low power conditions. So there should be a HeartBeat or Status Check right before the Power Limit being sent to the inverter.

001oc, 2023-11-04:
@stefan123t 
thanks for your answer. You are right, under "normal conditions" I am sending "senseless" and of course at this time even "useless" power limit commands to opendtu ... and opendtu tries to send this sense- and useless commands to the not stable communicating inverters, which makes the inverters somewhat grumpy. You are not completely right when you suppose that the inverters stop communication just for a while - they stop communicating at all for the time being. If they would stop just for some minutes to come up with a stable communication I wouldnt ask for a solution.

And you are right, I can solve the problem quiet easly within node red ... but I dont think that it is the right place to fix the problem - I will explain that matter from my perspektive somewhat later, frist the way I am doing things right now. I check for some variables I, sense- and useless, take care of in node red like that:

`if ((msg.PowerState.SolarWR1StatusReachable != 0) && (msg.PowerState.SolarWR1StatusProducing != 0)) ...`

I am looking for the first inverters reachability and if it is producing. Both are values opendtu is aware of - as I get them from opendtu. I currently have 2 HMS 2000 running right now - I am expecting to run at least 4 more till next summer. They both are running at nearly the same place with having 4 panels each. As of the fact that they are near by each other for my current usecase it is enough to check only one inverters availability to be quiet sure both are availabel. But, and now we come closer to the point where I think it is the wrong place where I right now "solve" the problem, what if I have x inverters running and they dont have the same angle to the sun and as of that have diffent times when they are available - are you still having the opinion it is the best idea to solve this problem within node red for some x inverters individually - I dont think so.

My point of view is that the only "right" place to solve such problems is within the inverters themself - for me it makes no sense that they are not able to just drop commands they can't execute. As of the fact that we dont have enough influence to the firmware of the inverters, we should solve the problem at the best/nearest place we have but the inverters themselfs. The second best place in my opinion is the opendtu. I think that it is the worst place to solve such problems at nodered or, even worser, at the mqtt-server. I think neither node red nor the mqtt-server should be in charge of checking the inner conditions of the inverters. Nodered has to read the powermeters parameter comming in by mqtt, decide if there is the need to change powerlimits and send corresponding command to opendtu. Maybe but not neccesarally nodered could checking if powerlimit commands are executed, but it should be even enough if it sends completely dull constantly powerlimit commands as long as they seem to be needed from a most lowbrain perspective. That is a simple and robust "control loop". Everything above that, from algorithm perspective, is not of the business of nodered or the mqtt-server. Opendtu knows best about the current conditions of the inverters, as of that it has to decide if it makes sense to send a powerlimit command to the inverters or even not. If one would make it most comfortable opendtu could send a feedback message if the command is executed or not and even why it is not executed - but it makes not much sense to hold nodered responible for not scaring the sensitive and whiny inverters for beeing not very robust concerning powerlimit commands. Nodered, in my opinion, is not responsible for making decissions about the mental stability of a bitchy inverter - that is the job of opendtu.

stefan123t, 2023-11-06:
@001oc I appreciate your reasoning that the inverter should discard commands, which it does IMHO eventually.
Also the OpenDTU has some built in checks to prevent sending superfluous commands to the inverter, but obviously not enough. Actually we check the response after some timeout whether the power limit has been received.
Now in the current situation if you continuously send new and sometimes superfluous power limit commands the DTU will have to wait for the timeout before it can re-check the power limit of the inverter. Hence even the normal operation of the OpenDTU may be impacted by your current use of the MQTT power limit topic.

stefan123t, 2024-01-24:
@001oc I can see from your log files that you are using a CMT2300A based HMS/HMT inverter.
These models are known to seize communication for 15 minutes / 900 seconds timeout (!) before resuming to the initial base channel, i.e. 865/915MHz.
Did you power off your DTU for that time or somehow made sure it was reset and trying to reach the inverter after the state occurred ?
If not, some users reported that only removing DC **and** AC power from the inverter allowed them to reset it to normal communication mode. 
In general HMS/HMT models seem to be quite a bit more picky about communication errors or being sent superfluous commands.

@tbnobody can we include a kill switch which prevents superfluous commands being sent to the inverter via MQTT ?
* during night-time or when the inverter is unreachable we do not need to send commands.
* do we have a queue for power_limit commands, i.e. will we send them later or do we need to empty that queue once in a while, i.e. after wake-up / reachable again ?
* if the power_limit setting is below a certain threshold (5%, 30W) difference from current / previous setting, we may consider to skip the command as being too fine grained.
* can we return a power_limit command failure using MQTT being async ?

tbnobody, 2025-02-07:
> @tbnobody can we include a kill switch which prevents superfluous commands being sent to the inverter via MQTT ?

This was implemented in one of the last versions.. going to close this issue.