tbnobody/OpenDTU #1476: openDTU with CMT2300A suddenly stops communicating with HMS 1600

Zaphodorange, 2023-11-05:
### What happened?

After almost one month with v23.9.18 the DTU just does not want to communicate with the Inverter at all. I tried with the (also / still) available DTU lite from Hoymiles without problem. I resetted the DTU to factory results (as stated in another bug) to no success. I did not recently update the DTU nor did right now since the updates after 23.9.18 seem not to touch any of my needed and used functionality.
And the power supply is capable of doing up to 2,4 Amps / was not changed to working setup.

### To Reproduce Bug

Reset, factory defaults and entering all configuration again does not help. DTU lite from hoymiles is working. Setup ran without problem 4 weeks before.

### Expected Behavior

just connect as the weeks before

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.9.18

### Relevant log/trace output

```Shell
RX Period End
17:03:01.559 > All missing
17:03:01.559 > Nothing received, resend whole request
17:03:01.559 > TX AlarmData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 11 00 65 47 BC B2 00 00 00 00 00 00 00 00 A6 92 E2 
17:03:01.559 > TX SPI Timeout
17:03:02.330 > RX Period End
17:03:02.330 > All missing
17:03:02.330 > Nothing received, resend whole request
17:03:02.330 > TX AlarmData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 11 00 65 47 BC B2 00 00 00 00 00 00 00 00 A6 92 E2 
17:03:02.330 > TX SPI Timeout
17:03:03.101 > RX Period End
17:03:03.101 > All missing
17:03:03.101 > Nothing received, resend whole request
17:03:03.101 > TX AlarmData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 11 00 65 47 BC B2 00 00 00 00 00 00 00 00 A6 92 E2 
17:03:03.101 > TX SPI Timeout
17:03:03.854 > RX Period End
17:03:03.854 > All missing
17:03:03.854 > Nothing received, resend count exeeded
17:03:03.903 > TX SystemConfigPara 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 05 00 65 47 BC B2 00 00 00 00 00 00 00 00 B2 86 F6 
17:03:03.903 > TX SPI Timeout
17:03:04.086 > RX Period End
17:03:04.086 > All missing
17:03:04.086 > Nothing received, resend whole request
17:03:04.086 > TX SystemConfigPara 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 05 00 65 47 BC B2 00 00 00 00 00 00 00 00 B2 86 F6 
17:03:04.086 > TX SPI Timeout
17:03:04.310 > RX Period End
17:03:04.310 > All missing
17:03:04.310 > Nothing received, resend whole request
17:03:04.310 > TX SystemConfigPara 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 05 00 65 47 BC B2 00 00 00 00 00 00 00 00 B2 86 F6 
17:03:04.310 > TX SPI Timeout
17:03:04.555 > RX Period End
17:03:04.555 > All missing
17:03:04.555 > Nothing received, resend whole request
17:03:04.555 > TX SystemConfigPara 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 05 00 65 47 BC B2 00 00 00 00 00 00 00 00 B2 86 F6 
17:03:04.555 > TX SPI Timeout
17:03:04.758 > RX Period End
17:03:04.758 > All missing
17:03:04.758 > Nothing received, resend whole request
17:03:04.758 > TX SystemConfigPara 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 05 00 65 47 BC B2 00 00 00 00 00 00 00 00 B2 86 F6 
17:03:04.758 > TX SPI Timeout
17:03:04.960 > RX Period End
17:03:04.960 > All missing
17:03:04.960 > Nothing received, resend count exeeded
17:03:04.960 > Fetch inverter: 116491545678
17:03:04.960 > Request SystemConfigPara
17:03:05.005 > TX ChannelChangeCommand 868.00 MHz --> 56 91 54 56 78 80 18 45 48 02 15 21 14 14 1E 
[... 186 lines cut ...]
17:03:22.218 > All missing
17:03:22.218 > Nothing received, resend whole request
17:03:22.218 > TX RealTimeRunData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 0B 00 65 47 BC C9 00 00 00 00 00 00 00 00 8E 9F A6 
17:03:22.218 > TX SPI Timeout
17:03:22.720 > RX Period End
17:03:22.720 > All missing
17:03:22.720 > Nothing received, resend count exeeded
17:03:22.766 > TX AlarmData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 11 00 65 47 BC C9 00 00 00 00 00 00 00 00 54 84 7D 
17:03:22.766 > TX SPI Timeout
17:03:23.500 > RX Period End
17:03:23.500 > All missing
17:03:23.500 > Nothing received, resend whole request
17:03:23.500 > TX AlarmData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 11 00 65 47 BC C9 00 00 00 00 00 00 00 00 54 84 7D 
17:03:23.500 > TX SPI Timeout
17:03:24.270 > RX Period End
17:03:24.270 > All missing
17:03:24.270 > Nothing received, resend whole request
17:03:24.270 > TX AlarmData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 11 00 65 47 BC C9 00 00 00 00 00 00 00 00 54 84 7D 
17:03:24.270 > TX SPI Timeout
17:03:25.078 > RX Period End
17:03:25.078 > All missing
17:03:25.078 > Nothing received, resend whole request
17:03:25.078 > TX AlarmData 865.00 MHz --> 15 91 54 56 78 80 18 45 48 80 11 00 65 47 BC C9 00 00 00 00 00 00 00 00 54 84 7D 
17:03:25.078 > TX SPI Timeout
```

tbnobody, 2023-11-05:
> TX SPI Timeout

Seems to be bad cable connection or defective module

Zaphodorange, 2023-11-05:
Thanks for the comment !
I will open the Hardware and try to fix it. I can perform this only tomorrow - will do a follow up in this thread.

4irm, 2023-11-08:
Hi there, I would like to add that I am also not able to establish any connection to my new HMS 1600. My 2nd inverter (HM 1500) works flawlessly. I double-checked every cable, etc. - Hence, I do not believe we are talking about a hardware defect here.

msaphoto06, 2023-11-08:
I have an identical problem on an esp32 with CMT
The connection is unstable.
No matter the setting 0db +20db
I am at -65rssi when I transmit..

Off -14:39:39
On -14:37:12
On -13:18:42
Off - 12:36:34
On - 10:55:12

Zaphodorange, 2023-11-08:
Hello, at least for what I opened the bug for is fixed; once I unplugged an re-plugged the 2,4Ghz board it started working again; thanks @tbnobody for the help ! 
As for @4irm : you are aware that the HMS uses 2,4 Ghz and the HMT 800 Mhz ?! You need two RF modules and I do not know how good they work in parallel ?!

Lineflyer, 2023-11-09:
> As for @4irm : you are aware that the HMS uses 2,4 Ghz and the HMT 800 Mhz ?! You need two RF modules and I do not know how good they work in parallel ?!

Just a correction:
HM uses 2,4 GHz
HMS and HMT use 800 MHz

But in case of @4irm you are right: The HM and HMS do not use the same frequency. He needs 2 RF modules.

closed 2025-04-19 by github-actions[bot]