tbnobody/OpenDTU #1479: [Request] Support for HMS-XX00W-2T

fsoelch, 2023-11-05:
### Is your feature request related to a problem? Please describe.

newer versions of the HMS does have integrated DTUs
it't would be grate to support those in OpenDTU as well in order to collect the inverter information offline as well.

### Describe the solution you'd like

DennisOSRM did a grate job in building a rust based solution for this Problem. Thankfully, he publised the code here: https://github.com/DennisOSRM/hms-mqtt-publisher

stefan123t, 2023-12-17:
The .protobuf files describe the so called NetworkCommands of the DTU Pro and DTU BI which are used between the S-Miles Cloud and S-Miles Installer Apps. They were included in the original source code drop on gitee.com.

There is also a python library to connect both the DTU-BI (Built-in DTU) in newer HMS-XX00**W**-2T HMS-WiFi models
based on those .protobuf definition files by rovo89: https://github.com/rovo89/hoymiles_proto

Such a solution within the OpenDTU project would need to convert the .protobuf-based, generated code from DennisOSRM / rovo89 from python to C++ in order to run directly on the ESP32.

stefan123t, 2023-12-19:
@tbnobody I do not have a HMS-xxxx**W** HMS-WiFi Model, but there were offers on the Discord to send one to the developers if need be.
Currently WiFi Information (Access Point) shows the AP mode as disabled after some time in the STA station mode.

For the DTU-BI there are three options:
* S-Miles Installer App:
   If we add an option to open this or another (third) WiFi connection in STA mode and connect to Port 10081 and IP 10.10.100.254 of the DTU-BIs Access Point we can send / request NetworkCommands using the .protobuf definitions and / or the Python library by rovo89 https://github.com/rovo89/hoymiles_proto.
* S-Miles Cloud endpoint:
   rovo89 mentioned that he did the following:
   - 'Hoymiles S-Installer App installiert
   - DIY-Account erstellt
   - Mit dem Account angemeldet
   - "Netzwerkkonfig" ausgewählt
   - (Firmware-Upgrade - Details lasse ich mal aus)
   - SSID und Passwort meines IOT-WLANs eingegeben --> Verbindung erfolgreich
   - Keine Anlage angelegt - meine Erwartung ist, dass es dadurch weniger Kommunikationsversuche gibt.
   - In den DHCP-Einstellungen des Routers eine feste IP für den WR eingetragen
   - In der Router-Firewall den ausgehenden Traffic von dieser IP geblockt und geloggt.
   - Innerhalb von 3 Minuten gab es Kontaktversuche zu 8.211.32.133 (dataeu.hoymiles.com) Port 10081 sowie 223.5.5.5 und 8.8.8.8 (je Port 53, also DNS).
   - https://github.com/henkwiedig/Hoymiles-DTU-Proto geclont
   - Python-Module installiert und protos komppiliert
   - IP des WR in GetConfig.py eingetragen und gestartet
   According to rovo89 we could spoof the 8.211.32.133 (dataeu.hoymiles.com) Port 10081 endpoint and let the DTU-BI deliver all its data to his man-in-the-middle DTU Relay python script or implement the same relaying on the OpenDTU and create a WiFi Bridge.
* reconfigure S-Miles Cloud endpoint
  Similar to the previous we can use the NetworkCommand to set the endpoint URL (DNS and Port) of the S-Miles Cloud in the Hoymiles DTU-BI. This is also possible in the DTU Pro Source Code NetworkCommand to change / update e.g. from dataeu.hoymiles.com to datacn.hoymiles.com / dataus.hoymiles.com.

adcrafter27, 2024-09-21:
Hello, I have this script running with an esp and an HMS-800W-2T
that works without any problems. The values ​​are picked up by the inverter within the local network.
Can't you implement something similar? or collaboration with the user

stefan123t, 2024-09-21:
@adcrafter27 according to the platformio.ini of the project it depends on nanopb v0.4.8 inorder to transpile the protobuf definitions to some C++  headers referenced and in the dtuHoymiles.cpp code. This is a s similar to the approach rovo89 suggested above for the python code.

tbnobody, 2025-01-09:
I don't have any plans to implement the inverters with WiFi... at least on a ESP32.

closed 2025-01-09 by tbnobody