tbnobody/OpenDTU #1487: Add "Total Yield Total", "Total Yield Day", "Total Power" to MQTT

uwebaierl, 2023-11-08:
### Is your feature request related to a problem? Please describe.

I have two micro-inverter connected. Both send data via MQTT to Home Assistant for their individual panels. Unfortunately, "Total Yield Total", "Total Yield Day" and "Total Power" are calculated within OpenDTU, but are not send via MQTT to Home Assistant.

### Describe the solution you'd like

Add "Total Yield Total", "Total Yield Day" and "Total Power" to MQTT.

turrican944, 2023-11-13:
Hi
Total Yield Total", "Total Yield Day" and "Total Power" are send via MQTT, but there is no Autodiscovery in HA (yet).
add the topics manual in MQTT.yaml

ac/yieldtotal is Yield Total
ac/yiedday is Yield Day 
....

https://github.com/tbnobody/OpenDTU/blob/master/docs/MQTT_Topics.md

DPO99, 2023-11-13:
But this is not what he wants. He wants the summary (landing page) of all inverters and there is no mqtt value at this moment.

WolfgangMe, 2023-11-16:
> ... there is no mqtt value at this moment.

Yes for all Inverters it is:
/<solar>/ac/power
/<solar>/ac/yieldtotal
/<solar>/ac/yieldday

The problem is, that these topics are not added Home Assistant MQTT-Auto-Discovery. So you have to add it manualy to the configuartion.yaml (as @turrican944 already mentioned)

closed 2023-11-21 by uwebaierl