tbnobody/OpenDTU #1488: Power not set to 0 when not reachable at EOD

mhin76, 2023-11-08:
### What happened?

At EOD when my inverter is getting offline OpenDTU still shows the last received power. I have set the "Zero runtime data". That works during the day when one of the inverter is too far away, but at EOD that is not working.

[image]

### To Reproduce Bug

Set the flag "Zero runtime data", wait until EOD when Inverter is offline.

### Expected Behavior

to set Power to 0.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.9.18

### Anything else?

I am using HM-800 and HM-1500 inverter.
Using a pre-compiled binary from Diamex (shop that provides also the hardware).

tbnobody, 2023-11-09:
Have you disabled the polling at night?

mhin76, 2023-11-09:
yes that is turned off. ah, you mean maybe DTU puts itself to night time before my inverter had a chance to time out and therefore the zero runtime data will never be triggered?

tbnobody, 2023-11-09:
> you mean maybe DTU puts itself to night time before my inverter had a chance to time out and therefore the zero runtime data will never be triggered?

That would be my guess yes. I will correct change the behavior of setting the runtime data to zero. currently it only happens on a active inverter (polling enabled). But as we see in your case, it might happen that the night time can occour before the inverter gets unreachable.

mhin76, 2023-11-10:
Perfect guess!! Now all power are set to zero after activating polling at night. Thanks for that hint!

commit fa7641f7 (2023-11-15): Fix: Power not set to 0 when not reachable at EOD

If the inverter polling was disabled before it got unreachable the values where not set to zero
Fix: #1488
closed 2023-11-10 by mhin76