tbnobody/OpenDTU #1499: v23.10.9: crash after a few days of normal operation

justinotherguy, 2023-11-14:
### What happened?

I've been using opendtu for a few months now - more than happy :)

I've been using v23.4.28 last and did an upgrade to v23.10.9 last week.
Since then the controller has crashed twice after a few days of regular operation.
By "crash" I mean:
- no more mqtt connection
- web interface not available anymore

### To Reproduce Bug

- install v23.10.9 binary (opendtu-generic_esp32.bin)
- leave it running
- wait for a few days :)

### Expected Behavior

controller remains available via web ui and keeps sending messages via mqtt

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.10.9

### Anything else?

my hw:
Chip-Modell: ESP32-D0WDQ6
Chip-Revision: 1
Chip-Kerne: 2
CPU-Frequenz: 240 MHz

I have done a downgrade to v23.9.18 to see whether this changes anything; in case you need more details from v23.10.9 running, I can always change it back.

Thank you in advance! :)

wuppiwuppi, 2023-11-15:
I can confirm this. Since the latest version v23.10.9 I have to reboot the device regularly to get it back to work (2-3 days). I did regular updates before that (maybe not every single version) and did not observe issues like this. 

My setup is similar:
ESP32-D0WD-V3
nRF24L01+
HM-600

001oc, 2023-12-02:
If you have mqtt running it is worth to test if it becomes better (solve the problem) when shutting of mqtt for testing purpose. If you read https://github.com/tbnobody/OpenDTU/issues/1468 you will see that this is my workarround to at least a similar looking problem. Currently opendtu is sending use- and senseless commands while inverters are not communicating at all (at night). Maybe that is disturbing opendtu, the inverter or whatever. My workarround right now ist to do the job of opendtu and test for proper communicating inverters and prevent sending anything to the inverter if there is no communication between inverter and opendtu.

mr-p666, 2024-02-21:
With the latest version some issues seems to be fixed. https://github.com/tbnobody/OpenDTU/issues/1587#issuecomment-1956618406

justinotherguy, 2024-02-21:
> With the latest version some issues seems to be fixed. [#1587 (comment)](https://github.com/tbnobody/OpenDTU/issues/1587#issuecomment-1956618406)

thx for the heads-up!
I've just installed the current version (v24.2.16) with unchanged config.
We'll see, if its running stable now.

Thx to the project for the update, too! 👍

wuppiwuppi, 2024-02-22:
yes, after upgrad to the latest version - issue seems fixed. 
Uptime > 10d. 

Upgrading without OTA took some "time" ;)

justinotherguy, 2024-03-07:
> We'll see, if its running stable now.

Alright: Since the update the dtu is running wonderfully; I have had an unexpected reboot after 1 or 2 days; been looking at it yesterday - 9 days of operation and I thought "let's wait until tomorrow, so I can also report 10 days of continuous operation"; well: it's there - but it has rebooted :D

So, in short:
- v24.2.16 runs super stable for me, the initial issue seems to be fixed
- I cannot tell whether the versions before v23.10.9 also had spurious reboots - but I don't really mind, as the device reboots and then operates normally again

Thank you! :)

I'll close the issue.

closed 2024-03-07 by justinotherguy