tbnobody/OpenDTU #1501: HTTP error upon OTA upgrade

andreas-bulling, 2023-11-15:
### What happened?

I am running 23.10.9 and tried to upgrade to the latest version but upon reaching 100% I see the following error

` [HTTP ERROR] Bad Request`

### To Reproduce Bug

upgrade to the latest firmware

### Expected Behavior

no error

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.10.9

RayAUT, 2023-11-16:
Same issue with latest firmware

Archibald-Harry-Tuttle, 2023-11-16:
> ### What happened?
>
> I am running 23.10.9 and tried to upgrade to the latest version but upon reaching 100% I see the following error
>
> ` [HTTP ERROR] Bad Request`
>
> ### To Reproduce Bug
>
> upgrade to [...]

same on my side

munter1506, 2023-11-16:
I have the same Error
"[HTTP ERROR] Bad Request"

tbnobody, 2023-11-16:
* What .bin file are you trying to flash?
* Please provide a screenshot of Info --> System
* Have you tried a power cycle?

Gerri1, 2023-11-16:
Same with me too!
[image]

The file is uploaded, but then the message appears.

System Informationen
Firmwareinformationen
Hostname	OpenDTU-96B38C
SDK-Version	v4.4.4
Konfigurationsversion	0.1.25
Firmwareversion / git Hash	v23.10.9
PIO Umgebung	generic
Firmware-Aktualisierung	Neue Version verfügbar! Zeige Änderungen!
Reset Grund CPU 0	Timer Group0 Watch dog reset digital core
Reset Grund CPU 1	for APP CPU, reset by PRO CPU
Anzahl der Konfigurationsspeicherungen	49
Betriebszeit	0 days 12:19:21
Hardwareinformationen
Chip-Modell	ESP32-D0WD-V3
Chip-Revision	3
Chip-Kerne	2
CPU-Frequenz	240 MHz
Speicherinformationen
Typ	Verwendung	Frei	Benutzt	Größe
Heap	
45 %
	155 kB	126 kB	281 kB
LittleFs	
6 %
	300 kB	20 kB	320 kB
Sketch	
98 %
	25 kB	1.511 kB	1.536 kB
Funkmodulinformationen
nRF24 Status	konfiguriert
nRF24 Chip-Status	verbunden
nRF24 Chip-Type	nRF24L01+
CMT2300A Status	nicht konfiguriert
CMT2300A Chip-Status

tbnobody, 2023-11-16:
@Gerri1 especially for you I can say that you have the wrong partition scheme. Please read the important upgrade notes on the start page <https://github.com/tbnobody/OpenDTU#-important-upgrade-notes->

Your Sketch partition shows just 1.536 kB. With the new partition table (as delivered since March 2023) the size should be 1.920 kB.

Gerri1, 2023-11-16:
I had no problems with the updates up to and including v23.10.9!
I'll probably have to remove that part.

tbnobody, 2023-11-16:
> I had no problems with the updates up to and including v23.10.9!

That might be right... Because these firmware images where most propably small enough to fit into the old partitions.

Gerri1, 2023-11-16:
Thanks for the info.
When I have a lot of time again, I'll flash it again!

munter1506, 2023-11-16:
> @Gerri1 especially for you I can say that you have the wrong partition scheme. Please read the important upgrade notes on the start page [...]

Thanks! This solved the issue in my case :)

Gerri1, 2023-11-16:
The new FW is now running for me too! 👍😊

tbnobody, 2023-11-16:
@awarda-rh you cannot just use the opendtu-generic.bin because this bin is compiled with the generic environment while you need a firmware compiled with opendtufusionv2 environment. I don't know (other then compile it yourself) how to get this specific version.

awarda-rh, 2023-11-16:
Thanks, @tbnobody , you see I am still a newby in the OpenDTU world.

I did some more digging on github and found this in the [OpenDTUFusionDocs](https://github.com/markusdd/OpenDTUFusionDocs/blob/main/OTA.md):

_ATTENTION for pre-flashed boards
If you bought an OpenDTUFusion Board that had a pre-flashed firmware, you might lose radio module connection after using the generic OTA image when you OTA for the first time, as the baked-in pin assignments get overridden._

My device currently is using the "OpenDTU Fusion v2 with CMT2300A and NRF24" device profile, see attached screenshot.
[image]

I think, I have to perform these three steps to update the pre-flashed AllianceApps OpenDTU Fusion:

1.) Use this file to OTA update the firmware:
https://github.com/tbnobody/OpenDTU/releases/download/v23.11.15/opendtu-generic_esp32s3_usb.bin

2.) After the update restore the configuration from this file: https://raw.githubusercontent.com/tbnobody/OpenDTU/master/docs/DeviceProfiles/opendtu_fusion.json

3.) Then I have to select "OpenDTU Fusion v2 with CMT2300A and NRF24" in Settings > Device Manager.

I haven't executed this plan yet. Does this sound like a reasonable plan, or am I still missing something?

Marc9485, 2023-11-16:
Bei mir ging heute in Sachen Verbindung nichts🤷🏻‍♂️[image]

Marc9485, 2023-11-16:
[image]

tbnobody, 2023-11-16:
> Bei mir ging heute in Sachen Verbindung nichts🤷🏻‍♂️

1. Hat dieser Fehler nichts mit dem Error on OTA Upgrade zu tun. Öffne bitte in diesen Fällen einen neuen Issue
2. Wurde dieser Fehler mit dem heutigen Update behoben.

Marc9485, 2023-11-16:
[image]
Scheint behoben zu sein mit dem heutigen update. Danke dir

3DJupp, 2024-01-05:
I also experienced this, some times I still have to do the update via VS Code, thats a fix for then.
(Partition table was already the new 1920 kB one)

ThiloGa, 2024-02-24:
Hi, 
> I think, I have to perform these three steps to update the pre-flashed AllianceApps OpenDTU Fusion:
>
> 1.) Use this file to OTA update the firmware: [...]

were you successfull?

greetingd

xtermi2, 2024-06-30:
I have a similar issue upgrading the Firmaware OTA. I also get `OTA Error` `HTTP ERROR] Bad Request`

I downloaded the latest release `v24.6.29` an tried `opendtu-generic_esp32.bin`.

My Sketch partition is already 1920 kB.

In the console I could not see any log message regarding this error while upgrade:
```
Fetch inverter: 114190515818
```

Here my system info:
[image]

closed 2025-04-19 by github-actions[bot]