tbnobody/OpenDTU #1508: OpenDTU does not wake up from night mode - it keeps in "night mode"

StefanOberhumer, 2023-11-17:
### What happened?

In the morning the display keeps powered off. 
In menu Info/NTP says "Night".
After changing location (Lat/Lon) display wakes up!

### To Reproduce Bug

Use precompiled binary, configure a NTP server

### Expected Behavior

In the morning the display keeps powered off. 
In menu Info/NTP says "Night".
After changing location (Lat/Lon) display wakes up!

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

527c2eaf4a1f07223d012f4cbfc4428dc0226966

StefanOberhumer, 2023-11-17:
Will check if 6f685a4ab5bad96b625f11a03a5e4c6f6db87c42 fixes the issue!

BabyFnord, 2023-11-17:
Had this too, for the first time w/ v23.11.15, but never with v23.10.9 and prior versions. I've now upgraded to v23.11.16 and will check …

DocSnyd3r, 2023-11-17:
Issue #1507 should ne the same problem, can be merged/closed

closed 2023-11-17 by StefanOberhumer

BabyFnord, 2023-11-18:
> Seems @tbnobody (THX) already fixed it with [6f685a4](https://github.com/tbnobody/OpenDTU/commit/6f685a4ab5bad96b625f11a03a5e4c6f6db87c42) (included in v 23.11.16)

Yes, that did it. Thanks @tbnobody 👍