tbnobody/OpenDTU #1512: Total Yield made available through API should respect negative yield offset

andreas-bulling, 2023-11-17:
### Is your feature request related to a problem? Please describe.

OpenDTU offers, amongst others, the total yield through its API. However, while a negative yield offset can be set per string to correct the yield shown on the web interface (e.g. to correct for yield that a previous owner of the inverter has accumulated, i.e. "reset" it to zero), this offset doesn't seem to be respected/applied when providing total yield through the API.

As a result, for example, the dbus-opendtu project cannot retrieve and show this offset-corrected yield in the Venus OS/Cerbo remote console. See https://github.com/henne49/dbus-opendtu/issues/121

### Describe the solution you'd like

Respect/apply a negative yield offset configured for a string when offering data through the API

tbnobody, 2023-11-18:
With API you mean the http interface? It shows exactly the same as the webinterface because the webinterface only uses the API and nothing else. There is no different data source. So if the web interface shows the correct values all output is the same.

andreas-bulling, 2023-11-18:
I don't have any knowledge about the openDTU internals I am afraid. I reported the fact that the openDTU web interface and the Venus OS/Cerbo inverter information generated by dbus-opendtu show different values in the above dbus-opendtu issue. Its author said that this is not due to dbus-opendtu but the values it obtains through the openDTU API.

Maybe the two of you want to communicate directly to see where the negative offset that can be set per string is not applied?

henne49, 2023-11-18:
I can confirm, that the negative value/offset works it is shown in Web UI, in JSON under api/livedata/status. With that we also use it correctly in https://github.com/henne49/dbus-opendtu.

Please verify your opendtu version is the latest.

andreas-bulling, 2023-11-19:
I am using the latest OpenDTU version but it does not work:

[image]

[image]

tbnobody, 2023-11-19:
What do you expect to see? The values shown in your second screenshot are exactly the same as in the opendtu web interface. they are showing the daily values. the offsets are applied to the total values (which is the only thing which makes sense).

tbnobody, 2023-12-07:
Going to close this issue aas we cannot reproduce it and no further response.