tbnobody/OpenDTU #1514: [Request] Allow to configure LWT QoS

tobiasdierich, 2023-11-18:
### Is your feature request related to a problem? Please describe.

I tried connecting my openDTU to AWS IoT Core. Unfortunately, AWS does not support QoS 2 which is used for the LWT (see https://docs.aws.amazon.com/iot/latest/developerguide/mqtt.html#mqtt-differences). AWS just closes your tcp connection if you try to connect with a LWT QoS of 2. At the moment, there is no way to configure which QoS to use for the LWT

### Describe the solution you'd like

Add a config parameter to allow users to change the QoS of the LWT.

I'd be happy to submit a PR.

tbnobody, 2023-12-16:
Imlemented in ee4811bbe75d572f454945f573c814dc8c30142c

closed 2023-12-16 by tbnobody

sabouflage, 2024-01-26:
I'm currently trying to connect an openDTU to AWS IoT Core and currently the most recent version (24.1.21). Found this issue as there is not that much information about connecting openDTU to AWS and troubleshooting. Did it work in general @tobiasdierich ?

Tend to open a seperate issue if not, as i'm currently receiving a lot 
```
Connecting to MQTT...
Disconnected from MQTT.
Disconnect reason:TCP_DISCONNECTED
```
and on AWS side just meaningless
```
{
    "timestamp": "2024-01-26 12:53:30.040",
    "logLevel": "ERROR",
    "traceId": "***",,
    "accountId": "***",
    "status": "Failure",
    "eventType": "Connect",
    "protocol": "MQTT",
    "principalId": "***",,
    "sourceIp": "***",,
    "sourcePort": "***",
}
```

tobiasdierich, 2024-01-26:
@sabouflage Yup, it worked for me. The error message just posted is what I got before changing the LWT QoS to 1. Did you double check that you are using LWT QoS != 2?

sabouflage, 2024-01-30:
Thanks for the confirmation. Yes, found this thread, and the tiny information hidden in https://docs.aws.amazon.com/iot/latest/developerguide/mqtt.html#mqtt5-properties (No QoS2 for AWS IoT MQTT v5 LWT support) and disabled the QoS2 after upgrading.

Unfortunately, I still have the same problem with the following configuration
[image]

No issues for connecting, if I use the same Certificate for connecting via a different MQTT Client (Mqttx) to IoT Core or connecting with the openDTU to an open MQTT Broker (broker.hivemq.com, via TLS).

Did you use a different authentication method or self-generated certificates?

sabouflage, 2024-01-30:
For TLS-secured connections there was still a hardcoded QoS of "2" - fixed that in https://github.com/tbnobody/OpenDTU/pull/1706 @tbnobody.

Fixed it manually - works as intended. Thanks for the support @tobiasdierich - tuned a little bit more with MQTTx to reproduce the error with the QoS level to finally find the right trace. And also thanks to this great project @tbnobody! :)