tbnobody/OpenDTU #1526: Gesamtleistung hängt sich auf wenn Umrichter nicht erreichbar

pyro2k9, 2023-11-26:
### What happened?

Wenn der Umrichter nicht mehr erreichbar ist, bleibt die Leistung insb. die Gesamtleistung auf dem letzten Wert stehen - also im Prinzip "hängen". Dies ist auch im Webpanel zu sehen.
In dem Zuge wird auch keine MQTT Nachricht mit Leistung 0W abgesetzt.

### To Reproduce Bug

Ist immer so, wenn der Umrichter nicht mehr erreichbar ist, also wenn es dunkel ist...

### Expected Behavior

Gesamtleistung sollte auf 0W gesetzt werden und dementsprechend auch eine MQTT Nachricht abgesetzt werden.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.11.16

### Relevant log/trace output

```Shell
n.r.? Sonst bitte nachfordern.
```

Wunni73, 2024-01-12:
Ja sogar die Statistik bleibt hängen
[image]

Juergen2453, 2024-01-12:
evtl. dort nicht richtig eingestellt?

[image]

Wunni73, 2024-01-12:
Super, ich Teste es gleich!
>Nulle Laufzeit Daten<    Habe ich eingestellt wenn der Wechselrichter keine Spannung mehr hat und keine Daten mehr sendet dann wird genullt.

Wunni73, 2024-01-12:
[image]
[image]
:-(  die Leitung ist jetzt Null aber es ist auch der Tag und die Gesamtleistung 0 :-(

tbnobody, 2025-02-07:
Going to close this issue due to no activity and a lot of code changes since then. Feel free to open a new issue if the problem persists.

closed 2025-02-07 by tbnobody