tbnobody/OpenDTU #1535: + new mqtt topic [serial]/status/rssi

hamstie, 2023-11-30:
### Describe the solution you'd like

it would be nice to have the rssi value of the dtu (not wifi)

SuperNinja73, 2023-12-22:
as @tbnobody said : "Inverter RSSI is (not yet) published. (One reason is, that the nRF modules don't provide a detailed RSSI information)"

but I have an HMS1000 so a cmt2300A and it seems to me that RSSI is supported as i saw here  :
https://github.com/lumapu/ahoy/wiki/RF-Module-CMT2300A---E49-900M20S

SuperNinja73, 2023-12-22:
we can imagine a RSSI triangle like this : 
[image]

tbnobody, 2024-10-05:
Implemented in edfe06e31eae286fbb88da69e9b65420cfa8628a

closed 2024-10-05 by tbnobody