tbnobody/OpenDTU #1567: Unbekanntes Modell

Karli67, 2023-12-15:
### What happened?

Unbekanntes Modell! Bitte melden Sie die "Hardware Teilenummer" und das Modell als Problem.

HMS-2000-4T
200051441TM.SBRO.DE.HM

### To Reproduce Bug

Modell wurde nicht erkannt 

### Expected Behavior

Wurde nicht erkannt

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V23.11.16

tbnobody, 2023-12-15:
Bitte einen Screenshot des kompletten Dialogs einfügen.

Karli67, 2023-12-15:
[image]

Karli67, 2023-12-15:
Ist das so OK?

tbnobody, 2023-12-15:
Welche OpenDTU Version verwendest du? Frage nur, weil die Hardwareteilenummer eigentlich zu kurz aussieht. War der Meinung ich hatte das in einer Version schonmal abgefangen.

Karli67, 2023-12-15:
[image]

Karli67, 2023-12-15:
Ich habe den Wechselrichter noch nicht am Netz angeschlossen.

Karli67, 2023-12-17:
Nach dem Update und dem Anschluss an das Netz wurde der Wechselrichter erkannt.
Danke

closed 2023-12-17 by tbnobody