tbnobody/OpenDTU #1572: Datums- und IP Anzeige im Display Pluszeichen statt Punkt

jblond4711, 2023-12-18:
### What happened?

Auf meinem 2.42“ Display werden Pluszeichen beim Datum und der IP Adresse anstelle von Punkten angezeigt.

### To Reproduce Bug

Aufs Display gucken

### Expected Behavior

OpenDTU installiert

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.12.17

tbnobody, 2023-12-18:
What display are you using? Not just the size, the complete name with it's controller.

jblond4711, 2023-12-18:
Ich hoffe ein Bild reicht?

[image]

tbnobody, 2023-12-18:
Currently the following i2c displays are supported:
* 2 = SSD1306
* 3 = SH1106

You are using a SSD1309...

tbnobody, 2023-12-18:
If you used the generic_esp32 binary in the past you can try to flash the attached firmware. Make sure to change to the `display.type` to 4 in your device profile.

[firmware.zip](https://github.com/tbnobody/OpenDTU/files/13703914/firmware.zip)

jblond4711, 2023-12-18:
Ich verwende die opendtu-generic.bin, wenn ich die generic_esp32 flashe gibt es eine Fehlermeldung bei dem Versionsstand.
Müsste ich die generic_esp32 am PC komplett neu flashen?
Kann ich den Display Type 4 auch in der generic.bin einstellen?

jblond4711, 2023-12-18:
Gut, dann gucke ich morgen nochmal genauer da ich in den Einstellungen kein Display Typ 4 gefunden habe.
Reicht auch das OTA flashen?
Ich war nämlich schon dabei nochmal komplett neu zu flashen aber da es dann dunkel war und ich dann ja nichts mehr auf dem Display gesehen hätte.

tbnobody, 2023-12-18:
OTA flashen und im Device Profile (das pin_mapping.json) den Display Typ ändern.

jblond4711, 2023-12-19:
Habe jetzt alles so eingestellt wie von dir geschrieben.
Leider sind die Punkte vom Datum und der IP auf dem Display weiterhin Pluszeichen. Es wird mir eine neue Firmware angezeigt soll ich die mal flashen oder weiterhin die von oben verwenden?

tbnobody, 2023-12-19:
> Es wird mir eine neue Firmware angezeigt soll ich die mal flashen oder weiterhin die von oben verwenden?

Die offizielle Firmware kann keinen DisplayType 4. (Den hattest du schon gesetzt oder?) Kann man auch unter Einstellungen --> Hardware kontrollieren. Wenn dann alles korrekt ist, aber die Anzeige immer noch falsch ist würde ich hier noch eine andere Firmware einstellen. Es gibt dummerweise 2 Treiber für das von dir beschriebene Display.

jblond4711, 2023-12-19:
Den DisplayType habe ich auf 4 gesetzt.
Dann las uns (mich) mal bitte noch die Firmware mit dem anderen Treiber testen.
[image]

tbnobody, 2023-12-19:
The attached file uses U8G2_SSD1309_128X64_NONAME2_F_HW_I2C

If this  does not either it's maybe a issue in the U8G2 Library.

[firmware.zip](https://github.com/tbnobody/OpenDTU/files/13720257/firmware.zip)

huste511, 2023-12-20:
Wenn Du so adhoc alternative Versionen bereitstellen kannst, könnte man dann nicht generell mehr Displays unterstützen?

Hatte mir in den letzten Monaten aus div. Quellen große und kleine OLEDs in verschiedenen Farben bestellt.
Bei allen habe ich die Plus-Zeichen. - Ich dachte schon da wäre normal und gewollt.
Auch im Netz findet man viele Bilder in der Variante.

Displays sind u.a. diese hier;
https://de.aliexpress.com/item/1005003484018034.html
https://de.aliexpress.com/item/1005003093058462.html

jblond4711, 2023-12-20:
Schade, mit der von Dir zuletzt bereitgestellten Firmware sind immer noch die Pluszeichen da.
Würde es eventuell was ändern das Datum und die IP Anzeige etwas zu vergrößern? Platz ist auf dem großen Display ja noch vorhanden.
Mich wundert nur das bei der Uhrzeit der Doppelpunkt richtig angezeigt.

tbnobody, 2023-12-23:
> Wenn Du so adhoc alternative Versionen bereitstellen kannst, könnte man dann nicht generell mehr Displays unterstützen?

Man muss es  halt selber kompilieren... können auch die wenigsten.

tbnobody, 2023-12-23:
> Würde es eventuell was ändern das Datum und die IP Anzeige etwas zu vergrößern? Platz ist auf dem großen Display ja noch vorhanden.

Auf dem 0.96" display ist da fast kein platz

jblond4711, 2023-12-24:
Gäbe es die Möglichkeit das Displayabhängig zu machen?  Also per Option einzustellen welche Displaygröße man verwendet?

huste511, 2023-12-24:
Die div. Displaygrößen (0,96 ; 1,3 ; 2,42 usw.)  haben doch alle die gleiche Auflösung von 128x64 Pixel.
Was sollte das also bewirken?

ampere-dieter, 2023-12-25:
Habe auch ein 2,42" Display verwendet. Geht sowohl mit der Standard firmware als Type#2 (SSD1306) als auch mit der firmware von hier als Type#4 (SSD1309) Display.  Man muss aber die seitlichen Pins benutzen, bei den oberen Pins sind SDA und SCL nicht angeschlossen. Hier der Link:  https://de.aliexpress.com/item/1005005241315177.html?spm=a2g0o.detail.0.0.51f813b8LyBDGH&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40000.327270.0&scm_id=1007.40000.327270.0&scm-url=1007.40000.327270.0&pvid=7e0c04e9-f1eb-4562-a43f-5e2db23c36e8&_t=gps-id:pcDet

[image]

Frohe Weihnacht, Dieter

huste511, 2023-12-25:
Bei Type 4 hat Du Punkte oder Pluszeichen? Doppelpunkte sehen normals aus?
Bei den Displays kommt es nicht nur drauf an wo die Pins sind, sondern wie Lötbrücken etc. sitzen. Viele kann man ändern.

btw: Hatte man versucht Bilder von Displays mit Punkten in Google zu finden. >95% sind wohl aber mit Pluszeichen unterwegs ... ;-)

ampere-dieter, 2023-12-25:
Hallo huste511,

[image]

Bei beiden firmware.zip Versionen hier in Fred wird bei meinem Display bei der kleinen Schriftgröße (Datum, Zeit) der Dezimalpunkt als Pluszeichen dargestellt und nicht als Punkt. Das gilt für Konfiguration als Type_2 als auch bei Type_4 Display. Stört mich aber nicht. Da das gleiche auch bei der default firmware auftritt, d.h. Mini-Pluszeichen statt Punkt, bin ich wieder auf die default firmware zurückgegangen. Meine 4 Panels liegen sowieso noch in der Hütte und müssen noch montiert werden. Mit einem  einstellbaren DC-Spannungswandler hab ich ein Panel am Wechselrichter simuliert und ihn eingerichtet. Ging problemlos. Z. Zt. bin ich am grübeln wie ich eine Nulleinspeisung hinkriege. Geht das mit OpenDTU oder muß man OpenDTU-onBattery verwenden? Am digitalen Zähler hängt eine Hichi Lesekopf. Mein alter analoger Zähler wurde leider ausgewechselt....
Dieter

tbnobody, 2023-12-26:
I've just doublechecked on my display. Yes, these small dots look like plus signs. But that is related to the font and not a issue. On a small display you just don't see the plus sign as good as on a larger one. But the font cannot be changed as the resolution of the displays is the same. Therefor I am going to close this issue as it is not really a issue.

closed 2023-12-26 by tbnobody