tbnobody/OpenDTU #1579: OpenDTU acquires LAN IP-address via DHCP although DHCP is set to off and static IP-address is defined

nakott, 2023-12-19:
### What happened?

I'm using a OpenDTU with LAN-connector. All worked well with opendtu-generic.bin 231116. I updated to 231217.
OpenDTU acquires LAN IP-address via DHCP although DHCP is set to off and static IP-address is defined. 

I reverted back to 231116 and is again good and after installing 231217, DHCP=off is ignored again. 

### To Reproduce Bug

Turn off DHCP, define static IP configuration, restart OpenDTU and check Info->Network section Network Interface (Ethernet)

### Expected Behavior

Static IP-address defined unter Settings->Network should be used and shown under Info->Network

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

fbc39ef

### Anything else?

I'm using this OpenDTU
https://www.derdtushop.de/shop?store-page=Open-DTU-OpenDTU-fur-Hoymiles-HM-Reihe-Lan-Plug-%26-Play-ready-mit-Gehause-p580008627

Here, I defined the static IP to 192.168.178.220

[image]

And here is, what the OpenDTU actually has: 192.168.178.185. My Fritzbox uses DHCP for addresses between 20 and 200.

[image]

tbnobody, 2023-12-19:
Found the issue. will release a new version in the evening.

closed 2023-12-19 by tbnobody

nakott, 2023-12-20:
Works perfectly. Many Thanks :-)

commit 3b923885 (2023-12-19): Fix #1579: Static IP in Ethernet mode did not work correctly