tbnobody/OpenDTU #1582: inverter limit is not transmitted

huste511, 2023-12-21:
### What happened?

If you want to set a new power limit for the inverter, the settings will be accepted but not transmitted.

There is also no error message.

### To Reproduce Bug

The transfer of commands is deactivated under settings -> inverter.

### Expected Behavior

There should at least be a note that the command cannot be executed due to the corresponding settings.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

e.g. v23.12.19 and before

mroenne2022, 2023-12-21:
I use the same version from Dec 19.th , but my DTU works fine and i can set new powerlimit without any problems.

huste511, 2023-12-21:
So your DTU transmits commands even though it is switched off?

mroenne2022, 2023-12-21:
Oh sorry, no the transfer of commands is not deactivated. It is only during the night deactivated and i send powerlimits only in the daytime.

tbnobody, 2024-03-25:
If this issue is still valid, please provide a snippet of the console log when you are setting the limit. Otherwise please close this issue.

tbnobody, 2024-03-31:
Closed due to inactivity. Feel free to open a new issue if the problem persists

closed 2024-03-31 by tbnobody