tbnobody/OpenDTU #1588: Firmware Update only possible after reboot

de-sascha, 2023-12-25:
### What happened?

If OPENDTU running some time and I need to update the firmware file the WEB IF not update. The percentage status bar went fast from 0-100% without apply the update. Workaround is reboot OpenDTU and then the Update can be apply. 

### To Reproduce Bug

Because after reboot I can apply the update. Assuming may this s an issue on the uptime. Because after reboot the firmware.bin can be uploaded. 

### Expected Behavior

May be fix the issue no reboot is required. 

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

v23.12.24

tm-107, 2024-02-29:
Can't reproduce it here, firmware update works fine.

MetaChuh, 2024-02-29:
@de-sascha 

is this issue solved for you too with v24.2.16 and can be closed ?

de-sascha, 2024-03-01:
Yes seams fixed in the meantime

closed 2024-10-15 by tbnobody