tbnobody/OpenDTU #1592: Firmware Update  v23.12.24 cannot be updated

wesseli01, 2023-12-26:
### What happened?

After the upload of the new FW Version  [opendtu-generic_esp32.bin](https://github.com/tbnobody/OpenDTU/releases/download/v23.12.24/opendtu-generic_esp32.bin) and the system was restarted - the new version is not active

### To Reproduce Bug

Upload 

### Expected Behavior

FW Version  [opendtu-generic_esp32.bin](https://github.com/tbnobody/OpenDTU/releases/download/v23.12.24/opendtu-generic_esp32.bin)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

OpenDTU-48FB10

### Relevant log/trace output

```Shell
FW Version  [opendtu-generic_esp32.bin](https://github.com/tbnobody/OpenDTU/releases/download/v23.12.24/opendtu-generic_esp32.bin)
```

tbnobody, 2023-12-28:
Please provide a screenshot of Info --> System --> Memory Information

maltejahn, 2023-12-29:
Hi, probably the same issue. I already tried it 3 times to Update the Firmware. But it still says that a new firmware is available. I used the opendtu-generic_esp32.bin v23.12.24 
[image]

But.. with the v23.12.28 it works again

Juergen2453, 2023-12-29:
Bei mir läuft die Version schon über einen Tag.

[image]

wesseli01, 2023-12-30:
I have updated with the new version - v23.12.28. It is working now. Thank you for the fixing.

wesseli01, 2023-12-30:
The topic can be closed now.

closed 2023-12-30 by wesseli01