tbnobody/OpenDTU #1600: values do not update in live view when reopen openDTU in browser or webapp

fletch247, 2023-12-29:
### What happened?

values like total power do not update when you reopen openDTU live view page. (on iphone) in the browser this is no problem since you can pull down the website to refresh. this is not possible when using openDTU page as a webapp.

### To Reproduce Bug

open openDTU as webapp. switch to another app. go back to openDTU webapp. total power will not change anymore until you go to a different openDTU page or force quit the webapp.

### Expected Behavior

auto refresh page everytime webapp is opened.
pull down to refresh should be possible (maybe not possible due to how iOS webbapps work).

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.12.28

### Anything else?

grafana dashboard webapp works perfectly

tbnobody, 2023-12-30:
I figured out the same problem. Unfortunatly I don't know how to detect when the webapp is opened (from a cached state). If someone has an idea, please help! :)

As a workaround, just move e.g. to the about view and then back to the live view.

stefan123t, 2023-12-30:
Could be related to this old thread on `cache.manifest` or hinting with dynamic-links (adding `?v=1`) to prevent iOS in webapp mode that the JavaScript / CSS files should better not be cached in the first place.

[How do you force an iPad home screen bookmarked web app to refresh?](https://stackoverflow.com/questions/8792255/how-do-you-force-an-ipad-home-screen-bookmarked-web-app-to-refresh)

[iOS: "Added to home screen" website won't refresh](https://stackoverflow.com/questions/10336135/ios-added-to-home-screen-website-wont-refresh)

```html
<META HTTP-EQUIV="CACHE-CONTROL" CONTENT="NO-CACHE">
<META HTTP-EQUIV="PRAGMA" CONTENT="NO-CACHE">
```

Or this one using timers to detect when the Focus is returned to the WebApp from background

[How to detect in iOS webapp when switching back to Safari from background?](https://stackoverflow.com/questions/4656387/how-to-detect-in-ios-webapp-when-switching-back-to-safari-from-background)

and pull-to-refresh (tm) gesture in iOS PWA / SPA's:

[iOS PWA: How to (re-)enable pull to refresh?](https://stackoverflow.com/questions/75972895/ios-pwa-how-to-re-enable-pull-to-refresh)

kwaaak, 2024-01-01:
It will update if you leave it open for some time (iPhone Safari webapp). 45 seconds, a minute, two minutes. Maybe helps when trying to find a mechanism to force the update

tbnobody, 2024-01-29:
Pull to refresh is implemented since a few versions. Also a status indicator for the websocket. I would therefor close this issue. Feel free to reopen if the current solution is not working or is insufficient.

closed 2024-01-29 by tbnobody

fletch247, 2024-01-30:
the indicator helps! just found the pull to refresh today.