tbnobody/OpenDTU #1601: Since last 2 updates after restart power off/on no time synchronisation any more

his1220, 2023-12-30:
### What happened?

Since last 2 updates after restart power off/on there is no time synchronisation any more.
I have to manually sync it.

### To Reproduce Bug

simple power off/on

### Expected Behavior

time synchronisation should work after repower

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	v23.12.28

pos-ei-don, 2023-12-30:
Here is the Logoutput from today.
As soon as i restart the device, it works again.

```
RX Period End
10:31:11.312 > All missing
10:31:11.312 > Nothing received, resend whole request
10:31:11.312 > TX RealTimeRunData Channel: 23 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 5E 00 00 00 00 00 00 00 00 AA F0 FA 
10:31:11.922 > RX Period End
10:31:11.922 > All missing
10:31:11.922 > Nothing received, resend whole request
10:31:11.922 > TX RealTimeRunData Channel: 40 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 5E 00 00 00 00 00 00 00 00 AA F0 FA 
10:31:12.098 > Websocket: [/livedata][3] disconnect
10:31:12.530 > RX Period End
10:31:12.530 > All missing
10:31:12.530 > Nothing received, resend whole request
10:31:12.530 > TX RealTimeRunData Channel: 61 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 5E 00 00 00 00 00 00 00 00 AA F0 FA 
10:31:13.034 > RX Period End
10:31:13.034 > All missing
10:31:13.034 > Nothing received, resend count exeeded
10:31:13.230 > TX SystemConfigPara Channel: 75 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 5E 00 00 00 00 00 00 00 00 64 FF 35 
10:31:13.446 > RX Period End
10:31:13.447 > All missing
10:31:13.447 > Nothing received, resend whole request
10:31:13.447 > TX SystemConfigPara Channel: 3 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 5E 00 00 00 00 00 00 00 00 64 FF 35 
10:31:14.906 > RX Period End
10:31:14.906 > All missing
10:31:14.906 > Nothing received, resend whole request
10:31:14.906 > TX SystemConfigPara Channel: 23 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 5E 00 00 00 00 00 00 00 00 64 FF 35 
10:31:15.067 > RX Period End
10:31:15.067 > All missing
10:31:15.067 > Nothing received, resend whole request
10:31:15.067 > TX SystemConfigPara Channel: 40 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 5E 00 00 00 00 00 00 00 00 64 FF 35 
10:31:15.279 > RX Period End
10:31:15.279 > All missing
10:31:15.279 > Nothing received, resend whole request
10:31:15.279 > TX SystemConfigPara Channel: 61 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 5E 00 00 00 00 00 00 00 00 64 FF 35 
10:31:15.478 > RX Period End
10:31:15.478 > All missing
10:31:15.478 > Nothing received, resend count exeeded
10:31:15.681 > Fetch inverter: 116182806256
10:31:15.681 > Request SystemConfigPara
10:31:15.886 > TX RealTimeRunData Channel: 75 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 63 00 00 00 00 00 00 00 00 3B FC 5A 
10:31:16.089 > RX Period End
10:31:16.089 > All missing
10:31:16.089 > Nothing received, resend whole request
10:31:16.089 > TX RealTimeRunData Channel: 3 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 63 00 00 00 00 00 00 00 00 3B FC 5A 
10:31:16.311 > RX Period End
10:31:16.311 > All missing
10:31:16.311 > Nothing received, resend whole request
10:31:16.311 > TX RealTimeRunData Channel: 23 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 63 00 00 00 00 00 00 00 00 3B FC 5A 
10:31:16.924 > RX Period End
10:31:16.924 > All missing
10:31:16.924 > Nothing received, resend whole request
10:31:16.924 > TX RealTimeRunData Channel: 40 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 63 00 00 00 00 00 00 00 00 3B FC 5A 
10:31:17.532 > RX Period End
10:31:17.532 > All missing
10:31:17.532 > Nothing received, resend whole request
10:31:17.532 > TX RealTimeRunData Channel: 61 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 63 00 00 00 00 00 00 00 00 3B FC 5A 
10:31:18.038 > RX Period End
10:31:18.038 > All missing
10:31:18.038 > Nothing received, resend count exeeded
10:31:18.240 > TX SystemConfigPara Channel: 75 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 63 00 00 00 00 00 00 00 00 F5 F3 95 
10:31:18.450 > RX Period End
10:31:18.450 > All missing
10:31:18.450 > Nothing received, resend whole request
10:31:18.450 > TX SystemConfigPara Channel: 3 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 63 00 00 00 00 00 00 00 00 F5 F3 95 
10:31:18.762 > RX Period End
10:31:18.762 > All missing
10:31:18.762 > Nothing received, resend whole request
10:31:18.762 > TX SystemConfigPara Channel: 23 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 63 00 00 00 00 00 00 00 00 F5 F3 95 
10:31:19.072 > RX Period End
10:31:19.072 > All missing
10:31:19.072 > Nothing received, resend whole request
10:31:19.072 > TX SystemConfigPara Channel: 40 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 63 00 00 00 00 00 00 00 00 F5 F3 95 
10:31:19.381 > RX Period End
10:31:19.381 > All missing
10:31:19.381 > Nothing received, resend whole request
10:31:19.381 > TX SystemConfigPara Channel: 61 --> 15 82 80 62 56 80 11 86 48 80 05 00 65 8F E3 63 00 00 00 00 00 00 00 00 F5 F3 95 
10:31:19.585 > RX Period End
10:31:19.585 > All missing
10:31:19.585 > Nothing received, resend count exeeded
10:31:19.993 > Fetch inverter: 116182806256
10:31:19.993 > Request SystemConfigPara
10:31:20.290 > TX RealTimeRunData Channel: 75 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 68 00 00 00 00 00 00 00 00 0B 8F 12 
10:31:20.708 > RX Period End
10:31:20.708 > All missing
10:31:20.708 > Nothing received, resend whole request
10:31:20.708 > TX RealTimeRunData Channel: 3 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 68 00 00 00 00 00 00 00 00 0B 8F 12 
10:31:21.318 > RX Period End
10:31:21.318 > All missing
10:31:21.318 > Nothing received, resend whole request
10:31:21.318 > TX RealTimeRunData Channel: 23 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 68 00 00 00 00 00 00 00 00 0B 8F 12 
10:31:21.929 > RX Period End
10:31:21.929 > All missing
10:31:21.929 > Nothing received, resend whole request
10:31:21.929 > TX RealTimeRunData Channel: 40 --> 15 82 80 62 56 80 11 86 48 80 0B 00 65 8F E3 68 00 00 00 00 00 00 00 00 0B 8F 12 

```

tm-107, 2024-02-29:
Can't reproduce it here, the time synchronization works fine.

MetaChuh, 2024-02-29:
@his1220 

is this issue solved for you too with v24.2.16 and can be closed ?

tbnobody, 2025-02-07:
Going to close this issue due to no activity and a lot of code changes since then. Feel free to open a new issue if the problem persists.

closed 2025-02-07 by tbnobody