tbnobody/OpenDTU #1606: [Request] Add Grid Profile DE_VDE4105_2011 (2.0.0) 0x0301 and CN_NBT32004_2018 (2.0.1) 0x0100 if still Unknown

stefan123t, 2023-12-31:
### Is your feature request related to a problem? Please describe.

Releate to unkown Grid Profiles in AhoyDTU https://github.com/lumapu/ahoy/issues/1304

Grid Profile for inverter TSUN TSOL-400 (wird als HM-350 erkannt)
Unknown Profile
`03 01 20 00 00 08 08 FC 07 30 00 01 0A 55 00 01 09 E2 10 00 13 88 12 8E 00 01 14 1E 00 01 20 00 00 01 30 07 02 8A 0A 55 07 30 14 1E 12 8E 00 50 00 1E 40 00 07 D0 03 E8 70 00 00 01`

Und hier noch ein "Unknown Profile" für den grid profile parser (MI-1500, 3rd gen.):
`01 00 20 01 00 03 08 FC 07 A3 00 10 0A 8C 00 10 04 7E 00 0A 0C 21 00 0A 10 02 13 88 13 56 1B 58 13 9C 00 01 12 8E 00 0A 20 00 00 01 30 03 01 2C 0A 8C 07 A3 13 9C 13 56 40 00 0B B8 0B B8 50 00 00 00 13 8B 01 90 00 10 60 00 00 01 0A 28 0A 8C 03 E8 70 02 00 01 27 10`

@tbnobody bitte das zweite einpflegen, falls es noch fehlen sollte. Danke!

Maybe you want to add that in OpenDTU too 
See initial requests in https://github.com/tbnobody/OpenDTU/pull/1527 or here https://github.com/tbnobody/OpenDTU/pull/1590 ?

### Describe the solution you'd like

Wir haben das selbe Grid Profile bereits einmal auf der OpenDTU Wiki Seite dokumentiert:
https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser

`03 01 20 00 00 08 08 fc 07 30 00 01 0a 55 00 01 09 e2 10 00 13 88 12 8e 00 01 14 1e 00 01 20 00 00 01 30 07 02 8a 0a 55 07 30 14 1e 12 8e 00 50 00 1e 40 00 07 d0 03 e8 70 00 00 01 1a 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

Ich habe es mal mit der dort angegebenen CRC16 Checksum `1a 45` ins Binärformat konvertiert und mit `./dtu.py` in der S-Miles Cloud hochgeladen um zu sehen was Hoymiles dazu sagt: `DE_VDE4105_2011` (2.0.0)

#### Grid Profile

Name | Value | Unit
-- | -- | --
**DE_VDE4105_2011** |   | -
H/LVRT |   | -
H/LFRT |   | -
Islanding Detection (ID) |   | -
Reconnection (RT) |   | -
Ramp Rates (RR) |   | -
Active Power Control (APC) |   | -
Nominal Voltage (NV) | 230 | V
Low Voltage 1 (LV1) | 184 | V
LV1 Maximum Trip Time (MTT) | 0.1 | s
High Voltage 1 (HV1) | 264.5 | V
HV1 Maximum Trip Time (MTT) | 0.1 | s
10mins Average High Voltage (AHV) | 253 | V
Nominal Frequency | 50 | Hz
Low Frequency 1 (LF1) | 47.5 | Hz
LF1 Maximum Trip Time (MTT) | 0.1 | s
High Frequency 1 (HF1) | 51.5 | Hz
HF1 Maximum Trip time (MTT) | 0.1 | s
ID Function Activated | 1 |  
Reconnect Time (RT) | 65 | s
Reconnect High Voltage (RHV) | 264.5 | V
Reconnect Low Voltage (RLV) | 184 | V
Reconnect High Frequency (RHF) | 51.5 | s
Reconnect Low Frequency (RLF) | 47.5 | s
Short Interruption Reconnect Time (SRT) | 8 | s
Short Interruption Time (SIT) | 3 | s
Normal Ramp up Rate (RUR_NM) | 20 | Rated%/s
Soft Start Ramp up Rate (RUR_SS) | 10 | Rated%/s
APC Function Activated | 1 |  

### Describe alternatives you've considered

I have used this code to convert the Hex Dump into a binary `grid_profile.bin` for upload via `./dtu.py`.
https://github.com/rovo89/hoymiles_proto/pull/1

Maybe you can make use of the script/s `hex2profile.py` and `profile2hex.py` based on the below code for uploading other Unknown Grid Profiles to the S-Miles Cloud too.

##### hex2profile.py

```python3
#!/bin/python3

import struct

def reverse_bytes(binary_data):
    byte_list = list(binary_data)
    for i in range(0, len(byte_list), 2):
        byte_list[i], byte_list[i+1] = byte_list[i+1], byte_list[i]
    reversed_data = bytes(byte_list)
    return reversed_data

def binary_to_hex_dump(file_path):
    with open(file_path, 'rb') as file:
        data = file.read()
    hex_dump = ''
    for byte in reverse_bytes(data):
        hex_code = format(byte, '02X')
        hex_dump += hex_code + ' '
    return hex_dump.strip()

def hex_dump_to_binary(hex_dump, output_file):
    hex_dump = hex_dump.replace(" ", "")
    hex_bytes = [hex_dump[i:i+2] for i in range(0, len(hex_dump), 2)]
    binary_data = bytes.fromhex("".join(hex_bytes))
    reversed_data = reverse_bytes(binary_data)
    with open(output_file, "wb") as file:
        file.write(reversed_data)

#hex_dump = "03 00 20 00 00 0A 08 FC 07 30 00 1E 0B 3B 00 01 04 0B 00 1E 09 E2 10 00 13 88 12 8E 00 01 14 1E 00 01 20 00 00 01 30 03 02 58 0A C8 07 A3 13 92 12 8E 40 00 07 D0 03 E8 50 08 00 01 13 9C 01 90 00 10 01 F6 13 74 70 02 00 01 27 10 80 00 00 00 08 5B 01 2C 08 B7 09 41 09 9D 01 2C 90 00 00 00 FF A1 B0 00 00 00 01 F4 00 5F A0 02 00 00 00 00"
hex_dump = "03 01 20 00 00 08 08 fc 07 30 00 01 0a 55 00 01 09 e2 10 00 13 88 12 8e 00 01 14 1e 00 01 20 00 00 01 30 07 02 8a 0a 55 07 30 14 1e 12 8e 00 50 00 1e 40 00 07 d0 03 e8 70 00 00 01 1a 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"

#file_path = 'grid_profile_original.bin'
#hex_dump = binary_to_hex_dump(file_path)
#print(hex_dump)

file_path = 'grid_profile.bin'
hex_dump_to_binary(hex_dump, file_path)

hex_dump = binary_to_hex_dump(file_path)
print(hex_dump)
```

### Additional context

Habe es mal per ./dtu.py in die S-Miles Cloud geladen: `CN_NBT32004_2018` (2.0.1) kommt mir chinesisch vor !

Name | Value | Unit
-- | -- | --
**CN_NBT32004_2018** |   | -
H/LVRT |   | -
H/LFRT |   | -
Islanding Detection (ID) |   | -
Reconnection (RT) |   | -
Ramp Rates(RR) |   | -
Frequency Watt (FW) |   | -
Volt Watt (VW) |   | -
Active Power Control (APC) |   | -
Nominal Voltage (NV) | 230 | V
Low Voltage 1 (LV1) | 195.5 | V
LV1 Maximum Trip Time (MTT) | 1.6 | s
High Voltage 1 (HV1) | 270 | V
HV1 Maximum Trip Time (MTT) | 1.6 | s
Low Voltage 2 (LV2) | 115 | V
LV2 Maximum Trip Time (MTT) | 0.1 | s
High Voltage 2 (HV2) | 310.5 | V
HV2 Maximum Trip Time (MTT) | 0.1 | s
Nominal Frequency | 50 | Hz
Low Frequency 1 (LF1) | 49.5 | Hz
LF1 Maximum Trip Time (MTT) | 700 | s
High Frequency 1 (HF1) | 50.2 | Hz
HF1 Maximum Trip time (MTT) | 0.1 | s
Low Frequency 2 (LF2) | 47.5 | Hz
LF2 Maximum Trip Time (MTT) | 0.1 | s
ID Function Activated | 1 |  
Reconnect Time (RT) | 30 | s
Reconnect High Voltage (RHV) | 270 | V
Reconnect Low Voltage (RLV) | 195.5 | V
Reconnect High Frequency (RHF) | 50.2 | s
Reconnect Low Frequency (RLF) | 49.5 | s
Normal Ramp up Rate (RUR_NM) | 30 | Rated%/s
Soft Start Ramp up Rate (RUR_SS) | 30 | Rated%/s
FW Function Activated | 0 |  
Start of Frequency Watt Droop (Fstart) | 50.03 | Hz
FW Droop Slope (Kpower_Freq) | 40 | Pn%/Hz
Recovery Ramp Rate (RRR) | 0.16 | Pn%/s
VW Function Activated | 1 |  
Start of Voltage Watt Droop (Vstart) | 260 | V
End of Voltage Watt Droop (Vend) | 270 | V
VW Droop Slope (Kpower_Volt) | 10 | Pn%/V
APC Function Activated | 1 |  
Power Ramp Rate (PRR) | 100 | Pn%/s

stefan123t, 2023-12-31:
I also checked the spanish `ES_RD1699` (2.0.0) from Spain which appeared on the wiki lately

`10 00 20 00 00 01 08 FC 07 A3 00 0C 09 E2 00 0C 0A 55 00 10 10 00 13 88 12 C0 00 24 13 BA 00 04 20 00 00 01 30 07 07 08 09 E2 07 A3 13 B0 12 C0 00 32 00 1E 40 00 07 D0 03 E8 70 00 00 01 04 BD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

Name | Value | Unit
-- | -- | --
**ES_RD1699** |   | -
H/LVRT |   | -
H/LFRT |   | -
Islanding Detection (ID) |   | -
Reconnection (RT) |   | -
Ramp Rates (RR) |   | -
Active Power Control (APC) |   | -
Nominal Voltage (NV) | 230 | V
Low Voltage 1 (LV1) | 195.5 | V
LV1 Maximum Trip Time (MTT) | 1.2 | s
High Voltage 1 (HV1) | 253 | V
HV1 Maximum Trip Time (MTT) | 1.2 | s
High Voltage 2 (HV2) | 264.5 | V
HV2 Maximum Trip Time (MTT) | 0.16 | s
Nominal Frequency | 50 | Hz
Low Frequency 1 (LF1) | 48 | Hz
LF1 Maximum Trip Time (MTT) | 3.6 | s
High Frequency 1 (HF1) | 50.5 | Hz
HF1 Maximum Trip time (MTT) | 0.4 | s
ID Function Activated | 1 |  
Reconnect Time (RT) | 180 | s
Reconnect High Voltage (RHV) | 253 | V
Reconnect Low Voltage (RLV) | 195.5 | V
Reconnect High Frequency (RHF) | 50.4 | Hz
Reconnect Low Frequency (RLF) | 48 | Hz
Short Interruption Reconnect Time (SRT) | 5 | s
Short Interruption Time (SIT) | 3 | s
Normal Ramp up Rate (RUR_NM) | 20 | Rated%/s
Soft Start Ramp up Rate (RUR_SS) | 10 | Rated%/s
APC Function Activated | 1 |

stefan123t, 2023-12-31:
@tbnobody also in der S-Miles Cloud kommt das oben unter **Additional context** `CN_NBT32004_2018` Grid Profile in der S-Miles Cloud heraus. Ohne Fehlermeldung!
Bitte beachten die S-Miles Cloud Web Oberfläche gibt die Section Header alle einmal am Anfang aus (?), dann kommen die Detailwerte ohne Section Überschriften. M.E. sehr gewöhnungsbedürftig ?

tbnobody, 2023-12-31:
Naja, das schon, aber wie oben beschrieben, kommt das 0x00 zweimal (nach entsprechnder byte länge) vor. Wenn das so nicht passt, dann stimmt der ganze Ansatz, wie es geparsed wird, aktuell nicht.

stefan123t, 2024-01-12:
I have analysed all the sixteen Grid Profiles from the [Wiki](https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser)
I have put all Grid Profile hex codes in an Excel Sheet for reference and numbered them 1 .. 16
To verify the names of the Grid Profiles and the contents I have uploaded them using the hex2profile.py Script and dtu.py from @rovo89.
Find the results as a zip of the Grid Profiles in markdown form (in case you want to include them into documentation).

[image]
[S-Miles_Cloud_Grid_Profiles.xlsx](https://github.com/tbnobody/OpenDTU/files/13919182/S-Miles_Cloud_Grid_Profiles.xlsx)
[S-Miles_Cloud_Grid_Profiles.zip](https://github.com/tbnobody/OpenDTU/files/13919183/S-Miles_Cloud_Grid_Profiles.zip)

For two profiles with the 0x0a00 magic bytes I have received no data.
For the most other profiles I had to remove the trailing 0x00 byte to convert them into grid_profile.bin for the dtu.py to upload.

tbnobody, 2024-05-29:
@stefan123t had just some time to look in detail through your data. Thank you for that!

Could you please double check profile no 3?
In the result I saw:
```
Recovery High Frequency (RVHF) | 502 | Hz
```

This is off by factor 10 which would match to #2021 where the value is too small. If the 502 is correct, than the profile is maybe wrong and the profile as shown in #2021 is right (and the divider has to be 10 instead of 100)

When looking at profile 4+5+6+7, the current parser outputs 5,02 but your data show 50,2 --> of by factor 10 again

I would say, that profile 3 contains a wrong value for RVHF in the inverter (off by factor 10) and the parser has to be adjusted from divider 100 to divider 10. But please verify profile 3 again.

tbnobody, 2024-06-10:
Hopefully I added all the information to the grid profile parser. In this case, I would close this issue.

closed 2024-06-10 by tbnobody