tbnobody/OpenDTU #1612: Connection problems with openDTU since dec23 releases

Gloomyeye, 2024-01-03:
### What happened?

Since the december firmware updates, I noticed that I have a lot of disconnections. Only a (hard) reset of the esp32 helps to retrieve new values and a new connection to the openDTU esp32.

[image]

### To Reproduce Bug

I can not say exactly why the connection problems occur. All was working with the november release.

### Expected Behavior

No connection errors

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.12.31

### Anything else?

**Ausgangsituation**
- ESP-WROOM-32 + NRF24L01+
- Verbindungsdaten für WLAN korrekt hinterlegt
- FritzBox 7590 WLAN Router (v.7.57) + Mesh (mit FritzBox Repeater)
- PiHole v5 ist als DNS aktiv
- 2,4 GHz + 5 GHz sind aktiviert (WP2+WPA3 Verschlüsselung aktiv)
- Abstand von ESP32 zum Router minimal (< 3m)

Gloomyeye, 2024-01-16:
Seit Tagen verliere immer mal wieder ich die komplette Verbindung zum ESP32, wo OpenDTU installiert ist. Das ist sehr ärgerlich, da nur ein manueller Neustart per Stromkabelziehen den ESP32 wieder aufweckt. Ich habe an der Installation (außer einem Firmware Update) nichts geändert.

tbnobody, 2024-01-19:
What do you see on the [serial console](https://tbnobody.github.io/OpenDTU-docs/firmware/howto/serial_console/) if this issue occurs?

Gloomyeye, 2024-01-26:
So far, I flashed my esp32 with a complete new firmware (without OTA) and I noticed that only a few disconnection with 10 sec. or so appeared in the Home assistant log for the connection status. Maybe this way helped me with the issue. I will continue to look in the log the next days.

Gloomyeye, 2024-02-17:
So far, no issues anymore with the latest version.

closed 2024-02-17 by Gloomyeye