tbnobody/OpenDTU #1613: Access Point not working after firmware update

DcFamas, 2024-01-03:
### What happened?

Hey there,
I cannot connect to the Access Point of the DTU anymore, when I installed and firmware update;

[image]

### To Reproduce Bug

These are my exact steps:

- for example flash v23.12.24
[image]
- powercycle the device
- connect to the device via. AP
- upload my pin_mapping.json

`[
    {
        "name": "NRF24 with SSD1306",
        "nrf24": {
            "miso": 19,
            "mosi": 23,
            "clk": 18,
            "irq": 16,
            "en": 4,
            "cs": 5
        },
        "eth": {
            "enabled": false
        },
        "display": {
            "type": 2,
            "data": 21,
            "clk": 22
        }
    },
    {
        "name": "CMT2300A with SSD1306",
        "nrf24": {
            "miso": -1,
            "mosi": -1,
            "clk": -1,
            "irq": -1,
            "en": -1,
            "cs": -1
        },
        "cmt": {
            "clk": 12,
            "sdio": 14,
            "cs": 27,
            "fcs": 26,
            "gpio2": -1,
            "gpio3": -1
        },
        "eth": {
            "enabled": false
        },
        "display": {
            "type": 2,
            "data": 21,
            "clk": 22
        }
    },
    {
        "name": "NRF24 + CMT2300A with SSD1306",
        "nrf24": {
            "miso": 19,
            "mosi": 23,
            "clk": 18,
            "irq": 16,
            "en": 4,
            "cs": 5
        },
        "cmt": {
            "clk": 12,
            "sdio": 14,
            "cs": 27,
            "fcs": 26,
            "gpio2": -1,
            "gpio3": -1
        },
        "display": {
            "type": 2,
            "data": 21,
            "clk": 22
        }		
    }
]`

- change to the specific configuration, for example NRF and CMT
- change language of the display to german and the orientation to none
- upload the latest firmware (here v23.12.31 opendtu-generic.bin)

[image]

- my computer isn't connected to the device anymore, I need to reconnect. Screen is still on and showing "offline". I cannot reconnect

### Expected Behavior

That I can use the AP and a firmware update.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.12.31 and v.23.12.24

### Anything else?

I'm using this dev board;
[image]

The CMT module by ebyte, this NRFmodule
[image]

I'm using this configuration for a long time, there must be something in the firmware. :/

Thank you for the upcoming help!

DcFamas, 2024-01-03:
flashed directly the newest;

[image]

tbnobody, 2024-01-03:
What do you see on the serial console if you cannot connect to the AP? And keep in mind, if you are chaning the default `admin` password, the password for the AP is also updated.

DcFamas, 2024-01-03:
Grabbed another ESP32/DTU;

[image]

For now its;
[image]

Uploaded the pin_mapping.json;
[image]

Changed the hardware configuration (to CMT);
[image]

Changed language and screen rotation;
[image]

Startet firmwareupdate to latest:
[image]

Connection not possible anymore:
[image]

I haven't changed the password in the process. :/ Thanks for the quick response!

DcFamas, 2024-01-03:
Just to be sure;
powercycled the device:

[image]

and tried with smartphone:
[image]

DcFamas, 2024-01-03:
Now I got some time and tested furthermore:

- Installed 231116 and updated to newest, not working
- Installed 23109 and updated to newest, not working
- Installed 2388 and updated to newest, not working
- Installed 2388 and updated to 231116, works fine
   -> then I installed 231216, works fine
   -> then 231217, works fine
   -> then 231218, works fine
   -> then 231219, works fine
   -> then 231224, works fine

aaaaand here we go;
v23.12.28 is the problem. :(

tbnobody, 2024-01-03:
Seems to be this issue: https://github.com/espressif/arduino-esp32/issues/8828

In version v23.12.28 the core has been updated from 2.0.9 to 2.0.14
Have to find some workaround for this issue.

stefan123t, 2024-01-04:
@tbnobody did you consider adding some debug code for the NVS details as described here by lbernstone:

https://github.com/espressif/arduino-esp32/issues/8828#issuecomment-1800731737

> The namespace for all this is `nvs.net80211`. Here's what mine has after running `OTAWebUpdater.ino`.
> ```text
> Namespace: nvs.net80211
> Entries: 27
> key 'ap.sndchan', type '1' 
> key 'ap.authmode', type '1' 
> key 'ap.sae_h2e', type '1' 
> key 'ap.chan', type '1' 
> key 'ap.ssid', type '66' 
> key 'ap.passwd', type '66' 
> key 'ap.hidden', type '1' 
> key 'ap.max.conn', type '1' 
> key 'bcn.interval', type '2' 
> key 'ap.p_cipher', type '1' 
> key 'ap.ftm_r', type '1' 
> key 'ap.pmf_e', type '1' 
> key 'ap.pmf_r', type '1' 
> key 'ap.pmk_info', type '66' 
> ```
> To get the `ap.passwd`:
> ```text
>   prefs.begin("nvs.net80211");
>   char* pw[128];
>   prefs.getBytes("ap.passwd", pw, 128);
>   Serial.printf("pw: %s\n", pw);
> ```
> See if that has been corrupted, or if deleting that key or the namespace, lets you start over.

DcFamas, 2024-01-05:
Tested, works great! Thank you. :)

tbnobody, 2024-01-05:
@stefan123t no because as written [here](https://github.com/espressif/arduino-esp32/issues/8828#issuecomment-1801530555) the password etc seems to be correct in NVS. So I didn't want to dive more into this topic.

stefan123t, 2024-01-05:
@tbnobody I have seen your fix using a new config version number to rewrite the prefs config table. 
Thanks that should workaround this, nice thinking!

For me it seems to have something to do with the prefs section of the flash memory area.

But as you said they will have to sort this out upstream.

commit c1fc907e (2024-01-04): Fix: Access Point not working after firmware update

Fixes #1613