tbnobody/OpenDTU #1614: Die Anzeige der Seriennummer aus der Statuszeile entfernen

T2023om, 2024-01-03:
### Is your feature request related to a problem? Please describe.

Die Seriennummer in der Statuszeile ist nicht notwendig und verbraucht viel Platz, da die ausführlichen Daten im Button "Wechselrichter Informationen" vorhanden und einfach erreichbar sind.

Ein weiterer Vorteil ist, dass ein Zeilenumbruch erst bei wesentlich kleineren Displays auftritt und damit die Statuszeile länger einzeilig schmal bleibt.

### Describe the solution you'd like

app.js entsprechend löschen oder modifizieren, 

<tr>
<th scope="row">Seriennummer: </th>
<td style="text-align: right;"> </td>
<td style="text-align: right;">22222.........</td>
</tr>

### Describe alternatives you've considered

Bei Bedarf könnte die Seriennummer aber auch noch in Spalte "Allgemein" angezeigt werden.
[image]

broth-itk, 2024-01-04:
Please no serial number in "General" section. I switched to English by purpose to have small compact tables with no line breaks.
The general section is using much space anyways.

I'd emphasize to think about removing the whole "General/Allgemein" table and place those two fields elsewhere.
Proposal: In the title field where the "Serienummer" is shown now but in a 2nd row:

Instead of

`HMS-2000-4T Serial Number: 11xxxxxxxxxxxx Current Limit: 2,000 W | 100%  Data Age: 9 seconds`

new:

```
HMS-2000-4T Current Limit: 2,000 W | 100%  Data Age: 9 seconds
Serial Number: 11xxxxxxxxxxxx  Temperature	9.8	°C

```

T2023om, 2024-01-04:
@ broth-itk
I would prefer to keep one line only as long as possible to hold the aeria smal.

Aus diesem Grund könnte in der deutschen Beschriftung der Eintrag
"Letzte Aktualisierung: vor 5 Sekunden"     in 
"Aktualisiert: vor 5 Sekunden"                    geändert werden.

broth-itk, 2024-01-04:
You can go even shorter:

`HMS-2000-4T  Serial Number: 116481313070 Current Limit: 2,000 W | 100% Data Age: 1,083 seconds / 1/4/2024, 16:44:29`

`HMS-2000-4T SN: 116481313070 Limit: 2,000 W | 100% Age: 1,083 seconds / 1/4/2024, 16:44:29`

.... if a one-liner is goal :)

T2023om, 2024-05-02:
@MetaChuh

Danke für deine Rückmeldung. 

Das Problem ist doch, dass unter bestimmten Bildschirm-Bedingungen die Änderung der Anzahl der Zeichen bereits um 1 Stelle zum Zeilensprung führt. Im Laufe einer Nacht vergrößert sich die Zeichenanzahl (Sekunden) sogar um bis zu 5 Stellen.

Ich denke die SN auszublenden ist komplizierter als mein Vorschlag in der deutschen Beschriftung das  Feld zu ändern:

"**Letzte Aktualisierung:**"        z.B. vor 5 Sekunden
in
"**Aktualisiert:**"                        z.B  vor 15 Sekunden"

d.h. in app.js:
DataAge:e=>{const{normalize:t}=e;return t(["**Letzte Aktualisierung:** "])} 
wird zu 
DataAge:e=>{const{normalize:t}=e;return t(["**Aktualisiert**: "])}

Das spart sofort 9 Zeichen und daraus entsteht kein Zeilensprung mehr.

Wenn das plausibel ist mache ich auch den pull request.

Lg
T2023om

T2023om, 2024-05-02:
Bei Tablett oder PC Displays in landscape hat dieser hack die größte Wirkung.
Im Porträt Mode bleibt es bei den Zeilensprüngen bei all meinen Geräten.  ;)

T2023om, 2024-08-06:
Hi zusammen, 
neben den ständigen Verbesserungen, die ich gern nutze, bin ich mit der neuen Anordnung der Daten auf der Statusseite frustriert.
[image]
Jetzt springen zwar die Zeilen nicht mehr wenn die Anzahl der Sekunden größer 9 wird, dafür werden gleich 2 Zeilen auf dem Display genutzt und damit passt die letzte Datenzeile nur noch bedingt auf das Display.
Ich habe verschieden Vorschläge gemacht das zu vermeiden, jetzt mache ich keine mehr. 
Dafür frage ich mich, wer die diese fette blaue Zeile als notwendig und unveränderlich und schön für die Funktion der ansonsten so tollen openDTU SW hält.
Aber vielleich fehlen mir nur dafür nur Informationen um das zu verstehen.

LG
T2023om

stefan123t, 2024-08-06:
@T2023om na es ist halt die Überschrift und Toolbar zu einem Wechselrichter, deshalb sind dort alle wichtigen und relevanten Daten zum Inverter angegeben.
Die Seriennummer (zumindest die erste 7 Ziffern/Zeichen mit 4 stelliger Modell und 3 stelliger Jahres ID, Kalenderwoche der Herstellung) brauchen wir regelmäßig, so brauchen wir nicht extra fragen ;). Und wenn ein Nutzer mehrere Inverter mit der selben Modellnummer / Charge hat, dann ist die eigentlich 5 stellige laufende Serial ID der jeweiligen KW auch nicht verkeht.
Die Farbe des Balkens ändert sich je nach Zustand / Empfang.

Das Layout passt sich auch der entsprechenden Bildschirmgröße an, nennt man “responsive design”, damit es sowohl auf dem Handy, Tablet oder PC und für Screenreader / Braillezeile lesbar ist. Vielleicht sieht es ja beim Entwickler mit anderen Fenster / Monitoreinstellungen ganz gut aus ?

broth-itk, 2024-08-06:
An der Stelle möchte ich nochmal freundlich an meinen Kommentar vom 4. Januar hinweisen ;-)

Ich denke das Layout ließe sich deutlich kompakter darstellen wenn man die statischen Texte etwas kürzt.

[image]

stefan123t, 2024-08-06:
@broth-itk ja kürzer geht immer. Man könnte auch die Texte durch Icons (material/fontawesome) ersetzen.

Für die ebenfalls sinnvolle und mW in den kommenden Jahren auch gesetzlich vorgeschriebene Barrierefreiheit / Accesssibility muss man dann aber eine versteckten Hinweistext anbringen sonst überliest der Screenreader das Icon bzw “EsEn 123356789012” statt “Seriennummer 123456789012” vor, dann ist es zwar kurz aber unverständlich.

Am Code auf bzw vom ESP32 übertragenen Datenvolumen wird dadurch halt leider auch nicht wirklich viel gespart. 

Man könnte auch statt seconds die deutlich kürzere und internationale SI-Einheit s angeben, wenn man unbedingt ein paar Zeichen sparen will. Keine Ahnung ob das in den Translation Files geht ?

[image]

broth-itk, 2024-08-06:
@stefan123t 
Mit den Bestimmungen zur Accesssibility kenne ich mich nicht aus, stelle aber mal in den Raum ob dies auch ein muss bei nicht-kommerziellen Projekten sein muss. Die Sinnhaftigkeit will ich damit aber keinesfalls in Frage stellen.

Bei meinem Vorschlag ging es letztlich nur darum, die Zeile bzw. den Text innerhalb der Zeile ohne große Änderungen möglichst kurz und kompakt zu gestalten.
Ganz schmale Bildschirme (Dein Screenshot) werden davon nicht viel profitieren, bei größeren müsste man es tatsächlich mal ausprobieren was passiert / ob das dem Wunsch des OPs entspricht / dem Projekt eine Verbesserung bringt

T2023om, 2024-08-10:
Alle meine verschiedenen Vorschläge hatten das Ziel die statischen Anteile der Statuszeile so klein wie möglich zu halten. Im allgemeinen reicht eine Verkürzung um 1 oder 2 Zeichen um dieses Ergebnis zu erreichen.Vielleicht reicht es auch die SN gegen einen kurzen aussagekräftigen Namen zu ersetzen, da die SN im Untermenü vorhanden ist.Danke für die konstruktive Mitarbeit LGT2023om

stefan123t, 2024-08-11:
@T2023om ich bin prinzipiell bei Dir dass man einige der Texte kürzen kann und evtl auch sollte. 

Zum Beispiel Serial number -> Serial ID statt bloß SN. Den Screenreader muss man eben einfach berücksichtigen wenn man es abkürzt.

Serial number -> Serial ID
Current Limit -> Limit
Data Age -> Age
Seconds -> s

Im Deutschen klappt das beim Vorlesen evtl nicht so gut Seriennummer -> Serien Nr. aber das wäre mE immer noch akzeptabel da Nr. eine gebräuchlich Abkürzung ist:

Seriennummer -> Serien Nr.
Aktuelles Limit -> Limit
Letzte Aktualisierung vor -> Aktuell vor
Sekunden -> s

Du siehst dabei die Barrierefreiheit zu berücksichtigen ist eigentlich gar nicht so schwierig und muss bei einem Hardware Betriebssystem wie OpenDTU ab Mitte 2025 eingehalten werden.
Gerade bei OpenSource finde ich es richtig und wichtig das nach bestem Wissen auch zu machen und Menschen mit Schwierigkeiten in Ihrem Alltag auch zu unterstützen.