tbnobody/OpenDTU #1627: Connection to Inverter lost every couple of minutes

bastiitsab, 2024-01-08:
### What happened?

Since today my openDTU shows the state Offline. Red bar in the webgui. After restarting the device I get a working connection, but after some seconds/minutes the connection is getting lost again. The location of the devices is not changed. 
Already updated to the latest openDTU firmware -> no change. 

### To Reproduce Bug

Restart openDTU. 

### Expected Behavior

No connection drop.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.1.4

### Relevant log/trace output

```Shell
10:06:24.213 > RX Period End
10:06:24.213 > All missing
10:06:24.213 > Nothing received, resend whole request
10:06:24.213 > TX ChannelChangeCommand 868.00 MHz --> 56 91 95 65 07 80 14 05 60 02 15 21 14 14 F7 
10:06:24.398 > RX Period End
10:06:24.398 > All missing
10:06:24.398 > Nothing received, resend whole request
10:06:24.398 > TX ChannelChangeCommand 868.00 MHz --> 56 91 95 65 07 80 14 05 60 02 15 21 14 14 F7 
10:06:24.485 > RX Period End
10:06:24.485 > All missing
10:06:24.485 > Nothing received, resend count exeeded
10:06:24.600 > TX RealTimeRunData 865.00 MHz --> 15 91 95 65 07 80 14 05 60 80 0B 00 65 9B BB 11 00 00 00 00 00 00 00 00 C2 43 DC 
10:06:24.655 > Interrupt received
10:06:24.796 > Frame kaputt
10:06:24.860 > Interrupt received
10:06:24.931 > Frame kaputt
```

### Anything else?

also restarted the inverter already.

tbnobody, 2024-01-08:
Which inverter are you using? And "Frame kaputt" normally only occours if you use multiple DTUs.

bastiitsab, 2024-01-08:
> Which inverter are you using? And "Frame kaputt" normally only occours if you use multiple DTUs or if you have a bad connection from the ESP to the nRF Module

sorry forgot to mention. 
HMS-1600. 
Only one Inverter and DTU is installed

Sn0w3y, 2024-03-16:
> Another possibility would be that your inverter is sending on a frequency that is also in use by one of your neighbors. In this case you could try to change the sending frequency e.g. to 866Mhz. (It will take up to 15min before the inverter uses the new frequency)

Where can one change the frequency of the Inverter?

Arkadier, 2024-05-01:
Same issue here. My Open DTU looses the connection to HM800. I have to restart the unit always.

closed 2025-01-21 by bastiitsab