tbnobody/OpenDTU #1628: Current limit problem

lazouze, 2024-01-08:
### What happened?

Good morning.

The current limits of micro inverters are at 1000%. I tried to put 100%. But the values return to 1000% after a short time.

I have the impression that this problem appeared after putting a value in the power adjustment box in Hoymiles app. This one was originally empty. But it can no longer be empty.

[image]

[image]

### To Reproduce Bug

Le but est permanent. Si je met une valeur entre 0 et 100% celle-ci revient à 1000%. 
si je règle une autre valeur dans l’app hoymiles, la valeur dans openDTU est variable. Exemple 320%…etc

### Expected Behavior

J’aimerai que mes micro onduleur arrêtent d’être bridés / limités en production. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V24.1.4

tbnobody, 2024-01-08:
Please provide the output of `http://<DTU IP>/api/limit/status`

I also see quite a lot entries in your eventlog. Please provide a screenshot from then as well.
Are you using two DTUs at the same time? This is a not suppported behavior and could lead to all possible strange effects!

lazouze, 2024-01-08:
I have 1 DTU-pro
And 1 OpenDTU (soon 2)

I don't understand :
« Please provide the output of http://<DTU IP>/api/limit/status »

tbnobody, 2024-01-08:
> I have 1 DTU-pro
> And 1 OpenDTU

And exactly this is not supported. The inverter just cannot talk to two DTUs (and it doesent matter which one) at the same time.

lazouze, 2024-01-08:
ok so if I want to use my 2 openDTUs, I have to unplug the DTU-Pro and if I want to use the DTU-Pro, I have to unplug my 2 openDTUs. That's right ?

tbnobody, 2024-01-08:
But you must make sure that each inverter  only gets polled by one OpenDTU as well. One inverter one DTU.

lazouze, 2024-01-08:
OK. I have 20 micro inverters because of the limit of 10, there will be 10 micro inverters on one DTU and 10 on the other.
(For the moment, all micro inverters remain at 100. fingers crossed).

tbnobody, 2024-01-08:
Is your DTU-Pro already off?

You should also look at the console at watch for lines containg `SystemConfigPara`
You should see something like this:

```
09:29:59.823 > Request SystemConfigPara
09:29:59.823 > TX RealTimeRunData Channel: 23 --> 15 71 60 35 46 80 12 23 04 80 0B 00 65 9B B2 86 00 00 00 00 00 00 00 00 DB 19 41 
09:29:59.882 > Interrupt received
09:29:59.882 > RX Channel: 75 --> 95 71 60 35 46 71 60 35 46 01 00 01 01 5C 00 33 00 32 00 B3 00 AD 00 0C 93 04 4C | -80 dBm
09:29:59.904 > Interrupt received
09:29:59.904 > RX Channel: 75 --> 95 71 60 35 46 71 60 35 46 02 00 0C 8B 4F 00 0B 00 0B 01 5C 00 31 00 02 00 AC 9D | -80 dBm
09:29:59.981 > Interrupt received
09:29:59.981 > RX Channel: 61 --> 95 71 60 35 46 71 60 35 46 03 00 06 00 0C B4 71 00 00 12 C1 00 0A 00 00 08 D7 5F | -80 dBm
09:30:00.004 > Interrupt received
09:30:00.004 > RX Channel: 61 --> 95 71 60 35 46 71 60 35 46 84 13 88 01 F7 00 C7 00 16 03 A2 FF FC 00 02 19 56 42 | -80 dBm
09:30:00.355 > RX Period End
09:30:00.355 > Success
```

Several of these outputs would be interessting.

lazouze, 2024-01-08:
No the Dtu pro is not currently turned off yet

lazouze, 2024-01-08:
12:28:15.030 > RX Channel: 23 --> 95 90 52 92 35 90 52 92 35 81 00 01 04 E2 00 00 03 E8 FF FF FF FF 01 68 6E 2C 33 | -80 dBm
12:28:15.047 > RX Period End
12:28:15.047 > Success
12:28:15.047 > Fetch inverter: 114190529229
12:28:15.063 > TX RealTimeRunData Channel: 75 --> 15 90 52 92 29 80 14 31 48 80 0B 00 65 9B DC 4E 00 00 00 00 00 00 00 00 55 4C 7F 
12:28:15.134 > Interrupt received
12:28:15.239 > RX Channel: 3 --> 95 90 52 92 29 90 52 92 29 01 00 01 01 54 00 10 00 37 01 57 00 11 00 3C 00 00 9C | -80 dBm
12:28:15.253 > Interrupt received
12:28:15.302 > RX Channel: 23 --> 95 90 52 92 29 90 52 92 29 02 30 FB 00 00 31 55 00 0A 00 0B 08 B2 13 8A 00 6E 74 | -80 dBm
12:28:15.316 > Interrupt received
12:28:15.353 > RX Channel: 23 --> 95 90 52 92 29 90 52 92 29 83 00 00 00 05 03 E8 FF F8 00 AB D4 A0 20 | -80 dBm
12:28:15.372 > RX Period End
12:28:15.372 > Success
12:28:15.466 > TX AlarmData Channel: 3 --> 15 90 52 92 29 80 14 31 48 80 11 00 65 9B DC 4E 00 00 00 00 00 00 00 00 8F 57 A4 
12:28:15.921 > RX Period End
12:28:15.922 > All missing
12:28:15.922 > Nothing received, resend whole request
12:28:15.922 > TX AlarmData Channel: 23 --> 15 90 52 92 29 80 14 31 48 80 11 00 65 9B DC 4E 00 00 00 00 00 00 00 00 8F 57 A4 
12:28:16.739 > RX Period End
12:28:16.739 > All missing
12:28:16.739 > Nothing received, resend whole request
12:28:16.739 > TX AlarmData Channel: 40 --> 15 90 52 92 29 80 14 31 48 80 11 00 65 9B DC 4E 00 00 00 00 00 00 00 00 8F 57 A4 
12:28:17.549 > RX Period End
12:28:17.549 > All missing
12:28:17.549 > Nothing received, resend whole request
12:28:17.549 > TX AlarmData Channel: 61 --> 15 90 52 92 29 80 14 31 48 80 11 00 65 9B DC 4E 00 00 00 00 00 00 00 00 8F 57 A4 
12:28:17.569 > Interrupt received
12:28:17.590 > RX Channel: 23 --> 95 90 52 92 29 90 52 92 29 01 00 01 80 01 00 01 75 42 75 42 00 00 00 00 80 02 97 | -80 dBm
12:28:17.686 > Interrupt received
12:28:18.893 > RX Channel: 23 --> 95 90 52 92 29 90 52 92 29 02 00 9E A0 78 A0 78 FF FF F1 ED 90 02 00 9F A0 80 38 | -80 dBm
12:28:24.137 > RX Period End
12:28:24.137 > Middle missing
12:28:24.137 > Request retransmit: 5
12:28:24.137 > TX RequestFrame Channel: 23 --> 15 90 54 01 11 80 14 31 48 85 A9 
12:28:24.339 > RX Period End
12:28:24.339 > Middle missing
12:28:24.339 > Request retransmit: 5
12:28:24.339 > TX RequestFrame Channel: 40 --> 15 90 54 01 11 80 14 31 48 85 A9 
12:28:24.547 > Interrupt received
12:28:24.559 > RX Channel: 23 --> 95 90 54 01 11 90 54 01 11 05 00 DA A0 99 A0 99 FF FF F1 EE 90 02 00 DB A0 A0 1C | -80 dBm
12:28:24.572 > RX Period End
12:28:24.572 > Middle missing
12:28:24.572 > Retransmit timeout
12:28:24.572 > Fetch inverter: 114190542835
12:28:24.572 > Request SystemConfigPara
12:28:24.585 > TX RealTimeRunData Channel: 61 --> 15 90 54 28 35 80 14 31 48 80 0B 00 65 9B DC 58 00 00 00 00 00 00 00 00 35 AA 4F 
12:28:24.676 > Interrupt received
12:28:24.760 > RX Channel: 23 --> 95 90 54 28 35 90 54 28 35 01 00 01 01 5C 00 12 00 3E 01 5D 00 18 00 54 00 00 F4 | -80 dBm
12:28:24.835 > Interrupt received
12:28:24.937 > RX Channel: 3 --> 95 90 54 28 35 90 54 28 35 83 00 00 00 06 03 E8 00 00 00 E0 20 CC F7 | -80 dBm
12:28:25.001 > RX Period End
12:28:25.001 > Middle missing
12:28:25.001 > Request retransmit: 2
12:28:25.001 > TX RequestFrame Channel: 75 --> 15 90 54 28 35 80 14 31 48 82 A3 
12:28:25.085 > Interrupt received
12:28:25.099 > RX Channel: 23 --> 95 90 54 28 35 90 54 28 35 02 31 C7 00 00 32 20 00 0A 00 0F 08 B8 13 89 00 8C D0 | -80 dBm
12:28:25.108 > RX Period End
12:28:25.108 > Success
12:28:25.142 > TX AlarmData Channel: 3 --> 15 90 54 28 35 80 14 31 48 80 11 00 65 9B DC 58 00 00 00 00 00 00 00 00 EF B1 94 
12:28:25.957 > RX Period End
12:28:25.957 > All missing
12:28:25.957 > Nothing received, resend whole request
12:28:25.957 > TX AlarmData Channel: 23 --> 15 90 54 28 35 80 14 31 48 80 11 00 65 9B DC 58 00 00 00 00 00 00 00 00 EF B1 94 
12:28:26.775 > RX Period End
12:28:26.775 > All missing
12:28:26.775 > Nothing received, resend whole request
12:28:26.775 > TX AlarmData Channel: 40 --> 15 90 54 28 35 80 14 31 48 80 11 00 65 9B DC 58 00 00 00 00 00 00 00 00 EF B1 94 
12:28:27.542 > RX Period End
12:28:27.542 > All missing
12:28:27.542 > Nothing received, resend whole request
12:28:27.542 > TX AlarmData Channel: 61 --> 15 90 54 28 35 80 14 31 48 80 11 00 65 9B DC 58 00 00 00 00 00 00 00 00 EF B1 94 
12:28:29.133 > Interrupt received

tbnobody, 2024-01-08:
> No the Dtu pro is not currently turned off yet

In this case everything can happen....
You should see this in the eventlog as well (I bet you see a lot of error codes 2)

Regarding the console log above... I showed you a wrong example (Sorry!)

This is the right one:

```
09:30:00.368 > TX SystemConfigPara Channel: 40 --> 15 71 60 35 46 80 12 23 04 80 05 00 65 9B B2 86 00 00 00 00 00 00 00 00 15 16 8E 
09:30:00.453 > Interrupt received
09:30:00.453 > RX Channel: 23 --> 95 71 60 35 46 71 60 35 46 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
09:30:00.600 > RX Period End
09:30:00.600 > Success
```

But as mentioned, as long as two DTUs are accessing the same inverter, everything could happen...

stefan123t, 2024-01-09:
Salut @lazouze you have had more than one DTU speaking to the same Inverter.
The inverter will eventually reply to any of the DTUs calling it with some response.
If one DTU is receiving an answer to the question from an other DTU it may not recognize the answer correctly.
But it also may attribute the answer to be correct and meaningful and that is when you surely get odd results.

Please stop using multiple DTUs with the same inverter et **bonne chance**!

tbnobody, 2024-01-29:
Closing due to inactivity. Feel free to reopen or create a new issue if the problem persists...

closed 2024-01-29 by tbnobody