tbnobody/OpenDTU #1629: After restart from mqtt opendtu does not send data to mqtt, restart is needet

svalcken, 2024-01-08:
### What happened?

mqtt server was rebootet. After them, opendtu does not reconnect and send data.

### To Reproduce Bug

Restart mqtt server and than data will not sendto mqtt.

### Expected Behavior

automatical reconnect to mqtt and send of data to mqtt.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.12.28

tbnobody, 2024-01-08:
What do you see on the console if you restart the broker? What broker are you using? Please provide a screenshot of  **Info**  --> **MQTT**

bombjack7000, 2024-02-01:
I have that on OpenDTU v23.12.24, too, after restarting Home Assistant. I am using the standard MQTT server in Home Assistant (core-mosquitto).
Everything back to normal after restart of OpenDTU.
[image]

tbnobody, 2024-02-01:
What is the mqtt status (info --> mqtt) if you restart the broker and the client does not work proberly?

bombjack7000, 2024-02-01:
Whoa,
OpenDTU is connected, MQTT Explorer also shows the MQTT packets coming in.
The problem seems to be on the Home Assistant side!

I better report that there.

Thank you for your quick reply.
OpenDTU is a great project! 👍 👍 👍 

[image]

srett, 2024-03-23:
I just had the same issue. **Info -> MQTT** showed **Status: Enabled**, but **Connection Status: Disconnected**. **Info -> Console** only showed inverter communication, nothing about MQTT. Simply saving the MQTT settings again without changing anything made it connect to mosquitto again. So it appears connection retry is broken under some circumstances. I'm on 24.1.26.

closed 2024-03-31 by tbnobody