tbnobody/OpenDTU #1637: OpenDTU wont boot after v23.12.24 on WT32-ETH01

MrJoki007, 2024-01-09:
### What happened?

When i try to upgrade to a firmware later than [v23.12.24](https://github.com/tbnobody/OpenDTU/releases/tag/v23.12.24), openDTU wont boot anymore.
Tried OTA upgrade and fresh reinstall with ESP Flasher
Device: WT32-ETH01

### To Reproduce Bug

Install newer version than [v23.12.24](https://github.com/tbnobody/OpenDTU/releases/tag/v23.12.24)

### Expected Behavior

Normal startup

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

9250487

### Relevant log/trace output

```Shell
Using 'COM3' as serial port.
Showing logs:
[16:15:52]E (657) esp_core_dump_flash: No core dum[16:15:52]E (658) esp_core_dump_flash: No core dump partition found!
[16:15:52]
[16:15:52]Starting OpenDTU
[16:15:52]Initialize FS... done
[16:15:52]Reading configuration... done
[16:15:52]Reading PinMapping... found valid mapping done
[16:15:52]Initialize Network... 
[16:15:52]Brownout detector was triggered
[16:15:52]
[16:15:52]ets Jun  8 2016 00:22:57
[16:15:52]
[16:15:52]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:15:52]configsip: 0, SPIWP:0xee
[16:15:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:15:52]mode:DIO, clock div:2
[16:15:52]load:0x3fff0030,len:1184
[16:15:52]load:0x40078000,len:12784
[16:15:52]load:0x40080400,len:3032
[16:15:52]entry 0x400805e4
[16:15:53]E (657) esp_core_dump_flash: No core dum[16:15:53]E (658) esp_core_dump_flash: No core dump partition found!
[16:15:53]
[16:15:53]Starting OpenDTU
[16:15:53]Initialize FS... done
[16:15:53]Reading configuration... done
[16:15:53]Reading PinMapping... found valid mapping done
[16:15:53]Initialize Network... 
[16:15:53]Brownout detector was triggered
[16:15:53]
[16:15:53]ets Jun  8 2016 00:22:57
[16:15:53]
[16:15:53]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:15:53]configsip: 0, SPIWP:0xee
[16:15:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:15:53]mode:DIO, clock div:2
[16:15:53]load:0x3fff0030,len:1184
[16:15:53]load:0x40078000,len:12784
[16:15:53]load:0x40080400,len:3032
[16:15:53]entry 0x400805e4
[16:15:54]E (657) esp_core_dump_flash: No core dum[16:15:54]E (658) esp_core_dump_flash: No core dump partition found!
[16:15:54]
[16:15:54]Starting OpenDTU
[16:15:54]Initialize FS... done
[16:15:54]Reading configuration... done
[16:15:54]Reading PinMapping... found valid mapping done
[16:15:54]Initialize Network... 
[16:15:54]Brownout detector was triggered
[16:15:54]
[16:15:54]ets Jun  8 2016 00:22:57
[16:15:54]
[16:15:54]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:15:54]configsip: 0, SPIWP:0xee
[16:15:54]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:15:54]mode:DIO, clock div:2
[16:15:54]load:0x3fff0030,len:1184
[16:15:54]load:0x40078000,len:12784
[16:15:54]load:0x40080400,len:3032
[16:15:54]entry 0x400805e4
[16:15:55]E (657) esp_core_dump_flash: No core dum[16:15:55]E (658) esp_core_dump_flash: No core dump partition found!
[16:15:55]
[16:15:55]Starting OpenDTU
[16:15:55]Initialize FS... done
[16:15:55]Reading configuration... done
[16:15:55]Reading PinMapping... found valid mapping done
[16:15:55]Initialize Network... 
[16:15:55]Brownout detector was triggered
[16:15:55]
[16:15:55]ets Jun  8 2016 00:22:57
[16:15:55]
[16:15:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:15:55]configsip: 0, SPIWP:0xee
[16:15:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:15:55]mode:DIO, clock div:2
[16:15:55]load:0x3fff0030,len:1184
[16:15:55]load:0x40078000,len:12784
[16:15:55]load:0x40080400,len:3032
[16:15:55]entry 0x400805e4
[16:15:56]E (657) esp_core_dump_flash: No core dum[16:15:56]E (658) esp_core_dump_flash: No core dump partition found!
[16:15:56]
[16:15:56]Starting OpenDTU
[16:15:56]Initialize FS... done
[16:15:56]Reading configuration... done
[16:15:56]Reading PinMapping... found valid mapping done
[16:15:56]Initialize Network... 
[16:15:56]Brownout detector was triggered
[16:15:56]
[16:15:56]ets Jun  8 2016 00:22:57
[16:15:56]
[16:15:56]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:15:56]configsip: 0, SPIWP:0xee
[16:15:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:15:56]mode:DIO, clock div:2
[16:15:56]load:0x3fff0030,len:1184
[16:15:56]load:0x40078000,len:12784
[16:15:56]load:0x40080400,len:3032
[16:15:56]entry 0x400805e4
[16:15:57]E (657) esp_core_dump_flash: No core dum[16:15:57]E (658) esp_core_dump_flash: No core dump partition found!
[16:15:57]
[16:15:57]Starting OpenDTU
[16:15:57]Initialize FS... done
[16:15:57]Reading configuration... done
[16:15:57]Reading PinMapping... found valid mapping done
[16:15:57]Initialize Network... 
[16:15:57]Brownout detector was triggered
[16:15:57]
[16:15:57]ets Jun  8 2016 00:22:57
[16:15:57]
[16:15:57]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:15:57]configsip: 0, SPIWP:0xee
[16:15:57]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:15:57]mode:DIO, clock div:2
[16:15:57]load:0x3fff0030,len:1184
[16:15:57]load:0x40078000,len:12784
[16:15:57]load:0x40080400,len:3032
[16:15:57]entry 0x400805e4
[16:15:58]E (657) esp_core_dump_flash: No core dum[16:15:58]E (658) esp_core_dump_flash: No core dump partition found!
[16:15:58]
[16:15:58]Starting OpenDTU
[16:15:58]Initialize FS... done
[16:15:58]Reading configuration... done
[16:15:58]Reading PinMapping... found valid mapping done
[16:15:58]Initialize Network... 
[16:15:58]Brownout detector was triggered
[16:15:58]
[16:15:58]ets Jun  8 2016 00:22:57
[16:15:58]
[16:15:58]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:15:58]configsip: 0, SPIWP:0xee
[16:15:58]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:15:58]mode:DIO, clock div:2
[16:15:58]load:0x3fff0030,len:1184
[16:15:58]load:0x40078000,len:12784
[16:15:58]load:0x40080400,len:3032
[16:15:58]entry 0x400805e4
[16:15:58]
Serial port closed!
```

tbnobody, 2024-01-09:
> Brownout detector was triggered

This means that your power supply is very bad!

MrJoki007, 2024-01-09:
Why does it then work again, when i downgrade to the old version?
Never had a problem
Tried two different power supplys

tbnobody, 2024-01-09:
I don't know whats maybe changed in the  core. Maybe they made the Brownout detector more strict?

tbnobody, 2024-01-09:
Just to make sure, I upgraded my WT32-ETH01 from 2023.07.xx to the latest version via OTA. If I unplug the ethernet cable while powering on, everything works as expected (even if I connect the ethernet cable afterwards). But if I power the ESP while the ethernet cable is connected, I also get these messages (`Brownout detector was triggered`). I then added a 100uF capacitor on the power supply and it's working again. So definitly a power issue and a more sensitive brown out detection

tbnobody, 2024-01-29:
Any  news on this issue? Is it working or do you still have issues?

MrJoki007, 2024-01-31:
No i cant get it to work
I upgraded to the newest version and it did not come back online.
Then i tried to boot with Ethernet unplugged. Still did not work.
Then i tried unplugging 3.3V of the NRF module. Did not work ether.
Sometimes the ethernet lights are on and the orange one is blinking, but i cannot access the webinterface.
Tried with a big powerbank, 3,4A power supply, USB charging port of my laptop, that can handle 2,4a.
Tried different usb cables as well.

Did not try to install a capacitor, but that should not be needed if no ethernet or the NRF module (or both) are plugged in.

MrJoki007, 2024-01-31:
3.3V and 5V when plugging in
NRF and Ethernet connected 
Looks perfect

[image]

MrJoki007, 2024-01-31:
Just installed a 470uF capacitor (thats what ive got laying around) between 3.3V and GND, but still not working

MrJoki007, 2024-01-31:
Found out that 3,3V goes down to 2,5V after few secounds. Powering the esp now with the flasher with 5V and 3,3V extra and works now.
So is there a hardware problem on my WT32-ETH01?

closed 2024-01-31 by MrJoki007