tbnobody/OpenDTU #1646: Vue.js: changes to WebApp are ignored by browser

Fribur, 2024-01-15:
### What happened?

I notice whenever changes are made to WebApp (or I switch between OpenDTU and OpenDTU-OnBaterry firmware), the browser keeps loading the old cached version, and is not re-freshing the cache with the new version. Only cleaning browser cache or a forced re-load fixes the issue. And I notice a number of users on discord/github report stumbling into the same issue.

### To Reproduce Bug

easiest: switch between OpenDTU and OpenDTU-OnBattery firmware and load the OpenDTU WebInterface.

### Expected Behavior

Browser re-freshes cache and loads/displays the new version from OpenDTU. Is anybody familiar with the Vue.js framwork able to assess which of the techniques[ described in the accepted answer on stackoverflow ](https://stackoverflow.com/questions/69473141/vue-webapp-load-latest-build) is the right onw to resolve this issue for good?

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

any

tbnobody, 2024-01-15:
This has nothing todo with VueJS. The ETAG is set to the current git hash. So after each Firmware Update (when the hash changes) the page is reloaded when pressing F5 (or even better CTRL+F5).

Fribur, 2024-01-15:
hmm...so what is causing this bug then? (screenshot should contain new region settings, error message is also wrong): https://github.com/tbnobody/OpenDTU/issues/1540#issuecomment-1891087654

tbnobody, 2024-01-15:
The user did not reload the page. This has to be done manually after each update. And in case of this screenshots something very bad happend because also the dtu serial number field is empty (I didn't change anything at this part of the code).

nkappler, 2024-01-15:
You could use the`Expires` or `MaxAge` header, so the browser doesn't cache the webapp files indefinetly,
A better way would be to append the version number to the bundled files, so the filename changes with each update and thus the browser hasn't cached it yet.

nkappler, 2024-01-15:
and you can force a page reload in javascript after the firmware update with `location.reload()`.

nkappler, 2024-01-15:
See: https://github.com/tbnobody/OpenDTU/blob/47417276a013e94d7757467d79a83b117bc3621d/webapp/vite.config.ts#L50-L51

you'll need to add `[hash]` back to the file name...

tbnobody, 2024-01-16:
>  A better way would be to append the version number to the bundled files, so the filename changes with each update and thus the browser hasn't cached it yet.

In this case the browser has to reload the index.html file which include the javascript files by name (incl. the hash). But the index.html is currently not automatically reloaded after a firmware update because I don't know when the ESP is available again. 

On the other hand the hash is not required because the e-tag is set and the webserver just responses with 304 instead of the app.js itself if the git hash is the same as for the one already delivered.

nkappler, 2024-01-16:
One possibility would be to replace the javascript <link>s with a dynamic import...
And you could periodically ping the esp to see if it is ready

nkappler, 2024-01-16:
for example with the [fetch api](https://developer.mozilla.org/en-US/docs/Web/API/Fetch_API/Using_Fetch)

Fribur, 2024-01-18:
Cannot really contribute to solution finding. I get the point of F5 or CTRl+F5 fixes the issue…but personally have not done so in a number of years. So not sure if the anverage user knows that.  And I do not get why closing/re-opening the browser has not the same effect. I switch quite frequently between OpenDTU-Onbattery and OpenDTU, and Google Chrome always confronts me with the wrong WebInterface despite obviously _very_ different WebApps. I have a hunch  @nkappler suggestions of including a hash in the file name 
might resolve that?

tbnobody, 2024-01-19:
>  I have a hunch @nkappler suggestions of including a hash in the file name might resolve that? 

No.... it will not solve it because the file is cached in both scenarios by your browser... Will think about it how to solve this  best...

nkappler, 2024-01-19:
I've never implemented something like this but I know it's possible, the hash seems to be one part of the puzzle as it changes with each release, however the code that loads the sources with the hash in it's name cannot be cached or the cache needs to be invalidated somehow, otherwise it just loads the old file, which your browser already knows...

nkappler, 2024-01-19:
`location.reload()` can also take a boolean to reload without cache, but that seems to only be supportes by Firefox.

tbnobody, 2024-01-19:
The cache should be automatically invalidated because the e-tag is different. I've just doublechecked it. The browser sends the version number within the Etag header and gets a repsonse 304 from the ESP if it maches. Also the Cache-Control is set to "public, must-revalidate". Therefor a `location.replace("/")` should do the job. 

Have already implemented something and will test it locally with some of my further commits.

tbnobody, 2024-01-21:
This issue should be fixed in 0584eadcf20fdf1f8e65e6ccddd5e410da39b070
After the successfull upload a automatic reload is triggered if the ESP is available again. But keep in mind, it will only work with the next firmware update.

closed 2024-01-26 by tbnobody