tbnobody/OpenDTU #1650: Sniffer mode to capture firmware updates

broth-itk, 2024-01-15:
### Is your feature request related to a problem? Please describe.

OpenDTU team want captures of firmware upgrades of the inverters.

### Describe the solution you'd like

Switch OpenDTU in "sniffer" mode to listen, receive and store all communication from Hoymiles DTU to the inverters.

The captured data should be send over the network to syslog server for further analysis.

Wishlist:

- Listen mode
- Syslog implementation (for debug output on console)

Once we have (unencrypted?) firmware dumps of the inverters, we can start disassembling ;-)

Comments are welcome. Thanks!

stefan123t, 2024-01-18:
Promiscuous mode would be possible  as you can set both DTU and inverter Serial ID on the current menus. Those could be added both/all to the maximum five NRF pipes.

Difficulty would be to guess the channels used by both Hoymiles DTU and inverter. But switching between the five NRF channels might be sufficient as the messages/packets are usually repeated 15 times/transmitted 16 times. So switching channel after 2-4ms may be sufficient, especially if configurable.

broth-itk, 2024-01-18:
I don't like the idea of channel hopping for captures.
As OpenDTU has the same inverters configured as the Hoymiles DTU, we can scan all channels until we see active communication with matching serials.
That channel is locked in and dumped to syslog.

stefan123t, 2024-01-18:
The typical WR <-> DTU messages are repeated by the NRF radio 15 times, i.e. 16 packets as you can see in the following pictures.

One packet with 15 repeats takes about ~35 ms to be sent, i.e. a single packet will take about 2,5-3 ms with a ~1 ms gap between repeated packets. 
(The larger gaps between consecutive packets / messages sent from the WR -> DTU are usually about 9-10 ms each.)
 
[image]

So switching the five channels after 2-5 ms listening on each should allow you to listen on each channel 2-3 times during the complete transmission period for a single packet with all its retransmits. And also the inverter is changing the channel from time to time, which has been known as channel hopping.

Please note that it may be easier (and has been done initially with the Hoymiles DTU Lite too) to simply hook two serial2usb converters onto the RX/TX pins of an original Hoymiles DTU: BAUD 125000, 8N1 should be the correct settings for the UART communication between the main cpu and the RF chips. 

See https://github.com/tbnobody/OpenDTU/issues/1540#issuecomment-1885707736

broth-itk, 2024-01-18:
@stefan123t OK, since I have an original DTU, it might be easier to make both signals available from the outside to capture data.

As long I don't have to open the lid of the transceiver, all is fine :)

broth-itk, 2024-01-19:
@stefan123t 

Here you have it, my new HMDUSI - HoyMiles Dual USB Sniffer Interface

[image]
[image]
[image]

phol, 2024-01-19:
I think capturing the data should be very straightforward with a simple python script. If you run that twice in two separate terminal windows, I think you're good.

Make sure pyserial is installed: `pip install pyserial`

```py
import serial
import datetime

# Replace baud_rate with the actual Hoymiles baud rate
def read_serial(port, baud_rate=9600):
    try:
        ser = serial.Serial(port, baud_rate, timeout=1)
        print(f"Connected to {port}")

        try:
            with open("log.txt", "a") as file:
            while True:
            if ser.in_waiting > 0:
                data = ser.readline().decode('utf-8').rstrip()
                timestamp = datetime.datetime.now()
                log_entry = f"{timestamp}: {data}\n"
                print(log_entry)
                file.write(log_entry)

        except KeyboardInterrupt:
            print("Interrupted by user, exiting.")

        except Exception as e:
            print(f"Error: {e}")

    except serial.SerialException as e:
        print(f"Serial error: {e}")

    finally:
        try:
            ser.close()
            print("Serial port closed.")
        except:
            print("Error closing serial port.")

# Replace with your port, e.g., 'COM3' for Windows, '/dev/ttyUSB0' for Linux/macOS
read_serial('COM3')
```

**Exiting the Script:** Use `Ctrl + C` to interrupt and close the script.

stefan123t, 2024-01-19:
According to other users info the BAUD rate is 125000 and 8N1 for parity, start/stop bits.

The Grid Profile update has been traced by now, but you can still contribute the firmware update if you want.

To get a firmware dump you would need access to the JTAG/SWD port on the inverter. Some other users do have that at the moment and have contributed a firmware dump already.

broth-itk, 2024-01-28:
I modified the code a bit to provide a more or less usable output.
Attached some files with the outputs (oringial towards this final version)-

Before doing any more steps with the DTU and updates, can you please take a look to the dumps and check if this is usable at all?

Thanks!

[log_1.txt](https://github.com/tbnobody/OpenDTU/files/14075733/log_1.txt)
[log_2.txt](https://github.com/tbnobody/OpenDTU/files/14075734/log_2.txt)

```
#!/usr/bin/python3

import serial
import datetime
import binascii

# Replace baud_rate with the actual Hoymiles baud rate

def read_serial(port, baud_rate=9600, log_location='log.txt'):
    try:
        ser = serial.Serial(port, baud_rate, timeout=1)
        print(f"Connected to {port}")

        try:
            with open(log_location, "a") as file:
                while True:
                    if ser.in_waiting > 0:
                        data = ser.read(ser.in_waiting)
                        h = binascii.hexlify(data)
                        timestamp = datetime.datetime.now()
                        log_entry = f"A: {timestamp}: {h}\n"
                        print(log_entry)
                        file.write(log_entry)
                        file.flush()

        except KeyboardInterrupt:
            print("Interrupted by user, exiting.")

        except Exception as e:
            print(f"Error: {e}")

    except serial.SerialException as e:
        print(f"Serial error: {e}")

    finally:
        try:
            ser.close()
            print("Serial port closed.")
        except:
            print("Error closing serial port.")

# Replace with your port, e.g., 'COM3' for Windows, '/dev/ttyUSB0' for Linux/macOS
read_serial('/dev/ttyUSB0', 125000, "log_1.txt")

```

phol, 2024-01-28:
Nice that you got it working! I'm glad I could contribute this small bit. 
I'm unfortunately not the right person to answer your question, but hopefully this will get us closer to supporting firmware updates and setting grid profiles.

PS:
If you add the file extension to the start of a markdown code block, you enable syntax highlighting. :) 

```md
```py
```

broth-itk, 2024-01-28:
Thanks for your hint, slways something new to learn ;-)

broth-itk, 2024-01-28:
Here we go, these logs were collected during 5 to 6 hours until night came to send the inverters asleep.

[log_1.txt.gz](https://github.com/tbnobody/OpenDTU/files/14076846/log_1.txt.gz)
[log_2.txt.gz](https://github.com/tbnobody/OpenDTU/files/14076847/log_2.txt.gz)

I can't tell which of these logs are RX or TX but this log has above combined and sorted by time:

[log_combined_sorted.txt.gz](https://github.com/tbnobody/OpenDTU/files/14076856/log_combined_sorted.txt.gz)

This should make analysis easier.
Maybe someone can spot a new command or something we haven't found yet.

I'll detach the DTU and use OpenDTU instead now.

broth-itk, 2024-01-31:
Any feedback about the logs? Is it usable?

I plan to perform firmware updates on all inverters on friday.

stefan123t, 2024-02-01:
Hi @broth-itk, I did not have time to look into the logs in detail.
The hex format is a bit difficult to read, so I will have to post-process them to understand what is going on.
Before that I can not say what the content / commands are that are used in the logs.

broth-itk, 2024-02-01:
Thanks for your feedback!
Well, the proposed python script gave weird outputs and I don't think the DTU is using UTF-8 to communicate to its radio module.
The script can still be improved by 
- reading the commands one by one
- parsing/formatting the output at your convinience

But as time is precious, _if the collected dumps are complete_, postprocessing can always be done.

broth-itk, 2024-02-06:
Here you go, these logs include the firmware update of two -4T inverters to 1.0.27 (Firmware Build Date 2023-06-05 10:24:00):

[log_1.txt.fwupd.gz](https://github.com/tbnobody/OpenDTU/files/14175156/log_1.txt.fwupd.gz)
[log_2.txt.fwupd.gz](https://github.com/tbnobody/OpenDTU/files/14175157/log_2.txt.fwupd.gz)

stefan123t, 2024-02-09:
@broth-itk I have downloaded them but I will not be able to analyse them in the coming weeks.
But many thanks for making the traces, I am already excited to learn some new "commands" 
which we can add as a new feature to OpenDTU/AhoyDTU at a later stage.

broth-itk, 2024-02-09:
@stefan123t Thanks! The logs contain all data since startup of the DTU. Maybe there is something interesting - indeed.
Good luck!
Cheers!

stefan123t, 2024-10-20:
Huch ich sollte mal die Logs vom [Firmware Update](https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-1928827458) von @broth-itk auswerten und wie versprochen dokumentieren.

broth-itk, 2024-10-21:
@stefan123t ist das tatsächlich so, dass die Frames stets mit 7e beginnen und mit 7f enden? Ist sichergestellt, dass ein 7e innerhalb eines Pakets nicht auftritt ;-)

Das Script kann ich gerne anpassen und demnächst einen neuen Trace (wegen Power Distribution Logic) ziehen.
Das wird sicherlich einige interessieren und wir sollten sicherstellen, dass die Tracedaten vollständig und verwendbar sind.

stefan123t, 2024-10-21:
Hallo Bernhard @broth-itk ja die 7e/7f werden durch 7d5e/7d5f escaped, 7d selbst durch 7d5d. Ich habe es hier detailliert beschrieben: https://github.com/tbnobody/OpenDTU/issues/1421#issuecomment-2425232860

broth-itk, 2024-10-21:
Danke @stefan123t, habe das Script angepasst und werde das morgen tagsüber ausprobieren.
Bevor ich das Firmware/Grid Profile Update laufen lasse, würde ich Dir einen kurzen Trace zur Verfügung stellen.
Nicht, dass am Ende die Traces nicht passen :)

stefan123t, 2024-10-21:
@broth-itk Danke die Traces passen prinzipiell schon so wie sie jetzt sind.
Die Umbrüche und Ersetzungen der Escapes kann man ja leicht auch nachträglich machen. 
Ich habe lediglich für die einfachere Lesbarkeit ein paar Leerzeichen eingefügt.

broth-itk, 2024-10-21:
So sieht das nun aus, denke damit kann man etwas besser arbeiten:

[image]

Die Frames werden einzeln aus dem Datenstrom herausgesucht und ins Log geschrieben.
Anstelle von RX/TX habe ich schlicht << und >> ins Log geschrieben da ich schlicht nicht weiß welcher USB/Serial Adapter auf welcher Leitung hängt.
Sollte aber so passen.

Ich lasse die DTU morgen mal den ganzen Tag online und beobachte.

stefan123t, 2024-10-21:
@broth-itk ja sieht gut aus die Richtung passt. 

Z.B. das MainCmd 0x15 REQ_ARW_DAT_ALL mit dem SubCmd RealTimeRunData_Debug [sic] 0x0B ist die Anfrage nach den aktuellen Werten. Die geht von der DTU zum WR >>.

Die Antwort ist dann  0x95 (immer MainCmd | 0x80) vom WR zur DTU <<.

broth-itk, 2024-10-22:
@stefan123t 

Anbei ein großer Trace von heute morgen. 
Darin sind folgende Updates enthalten:

- HMS-1800-4T/HMS-2000-4T von V01.00.27 auf V01.01.12
- HMS-350-1T von V01.00.14 auf V01.01.00

[update_communication_log.txt.gz](https://github.com/user-attachments/files/17472953/update_communication_log.txt.gz)

stefan123t, 2024-10-28:
@AlexJacu bitte ein eigenes Issue aufmachen oder Deinen Kommentar löschen. Hier geht es nur um die Implementierung / Dokumentation des Sniffer Modus. Danke.

stefan123t, 2024-11-15:
@DanielR92 schau mal hier sind die beiden Firmware Updates von @broth-itk 

https://github.com/tbnobody/OpenDTU/issues/1421#issuecomment-2425232860

Die sind für die beiden Inverter mit den Serial IDs `80729785` und `81313070`.

Er hatte hier und weiter oben in diesem issue seine Inverter genauer beschrieben:

https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2428138345

> These are my IDs and SW

> DTU: 80164633 (v00.02.23)
HMS 1: 80729785 (1800-4T, v1.00.27)
HMS 2: 81313070 (2000-4T, v1.00.27)
HMS 3: 83199047 (350-1T, v1.00.14)

https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2260425094

> Serial 1164807xxxxx
Production Year 2022
Production Week 7
Model HMS-1800-4T
Detected max. Power 1,800 W
Bootloader Version 0.0.1
Firmware Version 1.0.27
Firmware Build Date 2023-06-05 10:24:00
Hardware Part Number 269635841
Hardware Version 01.00

> Serial 1164813xxxxx
Production Year 2022
Production Week 13
Model HMS-2000-4T
Detected max. Power 2,000 W
Bootloader Version 0.0.1
Firmware Version 1.0.27
Firmware Build Date 2023-06-05 10:24:00
Hardware Part Number 269644033
Hardware Version 01.00

> Serial 1124831xxxxx
Production Year 2022
Production Week 31
Model HMS-350-1T
Detected max. Power 350 W
Bootloader Version 0.1.2
Firmware Version 1.0.14
Firmware Build Date 2021-12-09 12:46:00
Hardware Part Number 270541056
Hardware Version 02.00

stefan123t, 2024-11-15:
Ach ja und die letzten Logs habe ich mir noch nicht genauer angesehen. Das müsste mW die Version mit der „Power Distribution Logic“ aus #2362 sein

https://github.com/tbnobody/OpenDTU/issues/1650#issuecomment-2428650265

stefan123t, 2025-06-12:
Ein weiteres Log mit ChannelChange Commands von Bernhard findet sich hier: https://github.com/tbnobody/OpenDTU/discussions/1218#discussioncomment-13409125

> Habe zunächst auf Manuell gestellt, Kanal 1, dann auf Kanal 5, Kanal 10 und schließlich wieder auf "Auto"
[communication_log.txt](https://github.com/user-attachments/files/20653846/communication_log.txt)