tbnobody/OpenDTU #1651: Fehler bei der Frequenzeinstellung bei CMT2300a

eloka080677, 2024-01-15:
### What happened?

Meine OpenDTU ist auf 868Mhz eingestellt. Schaut man sich die Einstellungen an, wird dort allerdings 868.000,00 MHz angezeigt. Zusätzlich kommt die Meldung, dass sich die eingestellte Frequenz außerhalb des gültigen Bereichs befindet. Stellt man dies per Schiebeschalter auf 868,00MHz, kommt beim Speichern die Meldung „Die Frequenz muss zwischen 860250000 und 923500000 kHz liegen und ein vielfaches von 250kHz betragen!“

Faktor 1000 zu hoch

### To Reproduce Bug

Mit alter Firmware 868MHz einstellen und auf aktuelle Firmware aktualisieren

### Expected Behavior

Anpassung der Frequenzprüfung auf 868MHz anstatt 868GHz

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

https://github.com/tbnobody/OpenDTU/releases/tag/v24.1.14

### Relevant log/trace output

```Shell
Die Frequenz muss zwischen 860250000 und 923500000 kHz liegen und ein vielfaches von 250kHz betragen!
```

tbnobody, 2024-01-15:
Bitte den Browser Cache löschen oder mit STRG+F5 neu laden. Dann sollte über der Frequenzauswahl auch die Landes/Regionsauswahl erscheinen.

eloka080677, 2024-01-15:
Danke für den Schellen Hinweis. Hier muss tatsächlich der Browser am IPad gecached haben. Jetzt bekomme ich auch das angesprochene Dropdown Feld Europa, Amerika,Brasilien. 

Vielen liebenDank

closed 2024-01-15 by eloka080677