tbnobody/OpenDTU #1663: Cannot update OpenDTU from 23.10.9 to 24.1.4.

micha1221, 2024-01-18:
### What happened?

Hi, I tried to change the firmware on my esp32 OpenDTU, but after uploading the firmware, I got a "bad http request" error.

### To Reproduce Bug

I tried it very often with the "master" and the 24.1.4 firmware, and also with latest files ... everythings the same.

### Expected Behavior

sucessful update to the latest firmware

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	v23.10.9

micha1221, 2024-01-18:
Sorry - new version 24.1.14 !

nkappler, 2024-01-18:
Are you using a vpn?
Had this issue, although with another version.
For me it was the periodically refreshing connection that caused the error and turning off the vpn helped...

ituri, 2024-01-19:
Have you tried rebooting? I'm under the impression that if your OpenDTU has been running for a couple of days, the memory will be too full to store the new firmware upgrade. Rebooting "cleans" the memory and allows for enough space for the update - at least that's my amateur theory why a reboot has always helped if updates failed beforehand.

micha1221, 2024-01-19:
> Are you using a vpn? Had this issue, although with another version. For me it was the periodically refreshing connection that caused the error and turning off the vpn helped...

No, I'm using no vpn.

micha1221, 2024-01-19:
> Have you tried rebooting? I'm under the impression that if your OpenDTU has been running for a couple of days, the memory will be too full to store the new firmware upgrade. Rebooting "cleans" the [...]

offcourse I thought of that too ... and did many times - with no effect

micha1221, 2024-01-19:
By the way: same behavior with the V24.1.18 from today ...

nkappler, 2024-01-19:
> Have you tried rebooting? I'm under the impression that if your OpenDTU has been running for a couple of days, the memory will be too full to store the new firmware upgrade. Rebooting "cleans" the [...]

I'm facing the same issue now,
Seconds after a reboot, the file upload will reach ~60% then it suddenly pauses and after some more time it aborts.

If I get a response to the update call at all, it always seems to be `OTA could not begin`

nkappler, 2024-01-19:
[image]
[image]

nkappler, 2024-01-19:
Of course the moment I try to make a screen recording it succeeded...
It was my seventh try or so in total, a few minutes after reboot.
So maybe if it doesn't work immediately after a reboot and it doesn't work a long time after a reboot, try it a few minutes after rebooting.

micha1221, 2024-01-19:
this is my screenshot:
[image]

micha1221, 2024-01-19:
and the errror:
[image]

micha1221, 2024-01-19:
OK - this was. Since I've done it seems to work.

closed 2024-01-19 by micha1221