tbnobody/OpenDTU #1664: Massive WiFi Problems after updating to Build #1176

fritzCARaldo, 2024-01-18:
### What happened?

Help, I have massive problems. Until yesterday, my OpenDTU was running without any issues. Then, I wanted to update from v24.1.4 to OpenDTU Build #1176. This process stopped at 19%. A second attempt failed at 9% and reported "OTA error - [HTTP ERROR] Bad Request." I then tried with OpenDTU Build #1169, with the same result. I tried flashing with various versions (using Tasmota 2.2.3), and that worked. HOWEVER: Since then, I can't get the OpenDTU to connect to my WLAN (correct data, open WLAN). It takes the data but doesn't connect and keeps the Access Point open. After leaving it without power for a long time yesterday, it suddenly worked. However, after a while, the OpenDTU became unreachable again, and I had to disconnect it from power for an extended period (simple restart doesn't work). This morning, I can't get it to work again. I then used the online ESP tool to execute "Factory_Reset_And_Bootloader_Repair.bin" and flashed an OpenDTU firmware from 2 weeks ago. However, the issue persists: I can't get the device to (stably) connect to my WLAN?! PLEASE HELP! I'm slowly losing hope...

DEUTSCH: Hilfe ich habe massive Probleme. Bis gestern lief mein OpenDTU problemlos. Dann habe ich von v24.1.4 auf OpenDTU Build #1176 updaten wollen. Dieses stoppte bei 19%. Erneuter Versuch bricht bei 9% ab und meldet "OTA-Fehler - [HTTP ERROR] Bad Request". Habe es dann mit OpenDTU Build #1169 versucht, mit dem gleichen Resultat. Habe dann mit verschiedenen Versionen geflashed (mit Tasmota 2.2.3) und das hat auch geklappt. ABER: Seither bekomme ich das OpenDTU nicht mehr vom Access Point in mein WLAN (Daten korrekt, WLAN offen). Er übernimmt zwar die Daten, verbindet sich aber nicht und hält den Access Point offen. Nachdem ich ihn gestern lange vom Strom hatte, ging es dann plötzlich. Allerdings war der OpenDTU nach einer gewissen Zeit dann doch wieder nicht erreichbar und ich musste ihn erneut für länger vom Strom nehmen (einfacher Restart funktioniert nicht). Heute morgen dann bekomme ich ihn wieder nicht mehr zum laufen. Hab dann via dem Online ESP-Tool "Factory_Reset_And_Bootloader_Repair.bin" ausgeführt und eine 2Wo. alter opendtu-generic.factory.bin aufgespielt. Es bleibt aber dabei: Ich bekomme das Gerät nicht mehr (stabil) in mein WLAN?! BITTE UM HILFE! Ich verzweifle langsam...

### To Reproduce Bug

see above

### Expected Behavior

see above

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

Build #1176

### Relevant log/trace output

```Shell
Ask me what I need to provide (I am an absolute beginner).
```

### Anything else?

Ask me what I need to provide (I am an absolute beginner).

stefan123t, 2024-01-18:
@fritzCARaldo bitte schau Dir die Anleitung hier https://tbnobody.github.io/OpenDTU-docs/firmware/howto/serial_console/ an und beschreibe hier mal Deinen Boot-Vorgang mit dem Logfile bei einem Serial USB Kabel angeschlossen an der OpenDTU.

Am Besten Deinen initialen Post editieren und dort unter "Relevant log/trace output" zwischen den drei ``` oberhalb und unterhalb eingeben.

fritzCARaldo, 2024-01-18:
> @fritzCARaldo bitte schau Dir die Anleitung hier https://tbnobody.github.io/OpenDTU-docs/firmware/howto/serial_console/ an und beschreibe hier mal Deinen Boot-Vorgang mit dem Logfile bei einem [...]

Sei mir nicht böse Stefan, aber ich verstehe nur Bahnhof. Ich hab einen Mac und habe bislang immer mit Tasmota und heute mit https://espressif.github.io/esptool-js/ geflasht. Zuletzt auf einen 2 Wochen alten Build. Erst ging nichts, aber nach einer Stunde ohne Strom hat die Kiste aktuell doch wieder Funktion aufgenommen. Tut mir leid, dass ich das Technische nicht wirklich verstehe. Aber Danke für die Hilfe bzw. den Versuch.

stefan123t, 2024-01-18:
Na wenn es jetzt wieder geht kannst Du das Issue ja ggf. schließen.

In der o.a. Seite ist leider keine Anleitung um das Serial Log auf einem Mac aufzuzeichen:
Eventuell klappt das ja mit den folgenden beiden Kommandos (von hier https://jasonmurray.org/posts/2020/serialconsolemacos/) in einem Terminal ?

* das Erste prüft ob das USB2Serial Kabel angeschlossen ist.

```bash
ls -al /dev/tty.usbserial
# crw-rw-rw-  1 root  wheel    9,   8 Sep  6 14:21 /dev/tty.usbserial
```

* und das Zweite speichert die Ausgabe von der OpenDTU in eine Datei `/tmp/capture.txt`

```bash
sudo cu -s 115200 -l /dev/tty.usbserial > /tmp/capture.txt
```

Kann man ggf. auch erstmal ohne `> /tmp/capture.txt` ausführen, dann sieht man direkt die Ausgabe im Terminal.

fritzCARaldo, 2024-01-18:
> Na wenn es jetzt wieder geht kannst Du das Issue ja ggf. schließen.
> 
> In der o.a. Seite ist leider keine Anleitung um das Serial Log auf einem Mac aufzuzeichen: Eventuell klappt das ja mit den folgenden beiden Kommandos (von hier https://jasonmurray.org/posts/2020/serialconsolemacos/) in einem Terminal ?
> 
> * das Erste prüft ob das USB2Serial Kabel angeschlossen ist.
> 
> ```shell
> ls -al /dev/tty.usbserial
> ```
> 
> -> das hier gibt das Terminal dann aus:
> 
> ```
> crw-rw-rw-  1 root  wheel    9,   8 Sep  6 14:21 /dev/tty.usbserial
> ```
> 
> * und das Zweite speichert die Ausgabe von der OpenDTU in eine Datei `/tmp/capture.txt`
> 
> ```shell
> sudo cu -s 115200 -l /dev/tty.usbserial > /tmp/capture.txt
> ```
> 
> Kann man ggf. auch erstmal ohne `> /tmp/capture.txt` ausführen, dann sieht man direkt die Ausgabe im Terminal.

Wieso sollte ich das schließen, wenn ich nach 1000 Versuchen mit einem 2 Wochen alten Build seit 2h eine Verbindung habe? Damit ist doch die Ursache nicht gefunden? Ich werde das beobachten und dann löschen, wenn der nächste Master-Build funktionieren sollte. Bis dahin steht das Problem für mich "unter Beobachtung".

stefan123t, 2024-01-18:
Na dann kannst es ja auch noch mit dem Serial2USB Kabel und Deinem Terminal auf dem Mac beobachten.
Und bei Bedarf kannst Du die Logfiles hier einstellen, damit wir ggf. auch den Fehlerfall sehen können :wink:

fritzCARaldo, 2024-01-21:
Es lief jetzt mehrere Tage mit der älteren Firmware einigermaßen (2 Mal musste ich neu starten und dazwischen länger vom Strom nehmen). Jetzt geht es wieder nicht: IP ist nicht erreichbar, obwohl das OpenDTU in der Fritzbox angemeldet und aktiv ist. 

Habe die aktuellste Master-Version () jetzt nochmal via https://espressif.github.io/esptool-js/ geflasht auf OpenDTU Build **#1183** geflasht. Hier das Protokoll:

_esptool.js
Serial port WebSerial VendorID 0x10c4 ProductID 0xea60
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:f8:df:78
Uploading stub...
Running stub...
Stub running...
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Compressed 1702112 bytes to 1087571...
Writing at 0x1000... (1%)
Writing at 0x11e72... (2%)
Writing at 0x16aec... (4%)
Writing at 0x1aaba... (5%)
Writing at 0x1ed59... (7%)
Writing at 0x22d54... (8%)
Writing at 0x26d4f... (10%)
Writing at 0x2ad4a... (11%)
Writing at 0x2ed45... (13%)
Writing at 0x32d40... (14%)
Writing at 0x36d3b... (16%)
Writing at 0x3ad36... (17%)
Writing at 0x3ed31... (19%)
Writing at 0x45ceb... (20%)
Writing at 0x5035c... (22%)
Writing at 0x5c80f... (23%)
Writing at 0x67a2e... (25%)
Writing at 0x72763... (26%)
Writing at 0x7f7f8... (28%)
Writing at 0x8f176... (29%)
Writing at 0x94e13... (31%)
Writing at 0x9aeea... (32%)
Writing at 0xa1a63... (34%)
Writing at 0xa8368... (35%)
Writing at 0xaf729... (37%)
Writing at 0xb5cda... (38%)
Writing at 0xbbc9a... (40%)
Writing at 0xc1ac1... (41%)
Writing at 0xc810a... (43%)
Writing at 0xce088... (44%)
Writing at 0xd3d59... (46%)
Writing at 0xd9962... (47%)
Writing at 0xdf298... (49%)
Writing at 0xe5cc6... (50%)
Writing at 0xeb0fa... (52%)
Writing at 0xf02b4... (53%)
Writing at 0xf564b... (55%)
Writing at 0xfa98f... (56%)
Writing at 0xffcd3... (58%)
Writing at 0x104ed4... (59%)
Writing at 0x10a529... (61%)
Writing at 0x10f45f... (62%)
Writing at 0x114d68... (64%)
Writing at 0x119f78... (65%)
Writing at 0x11efa4... (67%)
Writing at 0x1242ac... (68%)
Writing at 0x12a01d... (70%)
Writing at 0x12f5de... (71%)
Writing at 0x13487b... (73%)
Writing at 0x139d0a... (74%)
Writing at 0x13f052... (76%)
Writing at 0x144472... (77%)
Writing at 0x1498ab... (79%)
Writing at 0x14ee83... (80%)
Writing at 0x154744... (82%)
Writing at 0x15a69f... (83%)
Writing at 0x1600b8... (85%)
Writing at 0x165bf2... (86%)
Writing at 0x16b4cc... (88%)
Writing at 0x1710cf... (89%)
Writing at 0x179be2... (91%)
Writing at 0x182f90... (92%)
Writing at 0x188691... (94%)
Writing at 0x18dfe0... (95%)
Writing at 0x19381f... (97%)
Writing at 0x198986... (98%)
Writing at 0x19e5c3... (100%)
Wrote 1702112 bytes (1087571 compressed) at 0x1000 in 99.766 seconds.
Hash of data verified.
Leaving..._

Bislang nicht erreichbar...

Kommt hier zufällig jemand aus der Region Osnabrück (und kann/mag helfen)?

Omega13x, 2024-01-21:
Läuft super hier.
[image]
[image]

fritzCARaldo, 2024-01-23:
Ich verstehe es einfach nicht... Mal erreichbar, dann wieder nicht. Das heißt doch, dass alles korrekt aufgespielt wurde. Ich habe keinen Lösungsansatz, den ich als Laie noch verfolgen könnte. Das macht für mich einfach keinen Sinn. Sollte das an der Fritzbox liegen? Aber auch die läuft seit Monaten ohne Veränderung oder Probleme...

huckes, 2024-01-23:
Evtl. ein Einzelschicksal bzw. ein defektes ESP32-Modul?
Hast Du die Möglichkeit, das ESP32-Modul zu ersetzen?
Hardware ist ja auch fehlbar ;-)
Fritzbox neu starten tut allerdings auch nicht weh. Ist ja kein großes Ding.

PS: Ich hatte bei einer anderen ESP32-Verwendung längere Zeit den Effekt, das die WLAN-Verbindung nach wenigen Stunden nicht mehr möglich war, sonst funktionierte das Modul. Irgendwann ging gar nix mehr. Das Modul hat dann am USB-Port beim Booten nur noch "Brownout" gemeldet. Also ab in den Verwurf. Neues Modul geflasht eingebaut --> fertig.

fritzCARaldo, 2024-01-23:
> Evtl. ein Einzelschicksal bzw. ein defektes ESP32-Modul? Hast Du die Möglichkeit, das ESP32-Modul zu ersetzen? Hardware ist ja auch fehlbar ;-) Fritzbox neu starten tut allerdings auch nicht weh. [...]

Danke für deine Rückmeldung! Ich habe das Modul bei Kleinanzeigen inkl. Antenne und Gehäuse in einer verlöteten Variante bestellt, weil das echt nicht mein technisches Kompetenzgebiet ist. Lief jetzt auch ein Jahr problemlos. Müsste mir ein neues Setup zum ausprobieren bestellen... :/

Zu deiner Frage: Ja, ausschließen kann man das nicht. Ich stoße hier an meine Logikgrenze. Aktuell funktioniert alles wie gewünscht, aber in ein paar Std. wird das (ohne mein Zutun) vermutlich wieder anders aussehen... Ist für mich Strange, warum es mal komplett funktioniert (heißt: Ich hab richtig geflasht / konfiguriert) und dann wieder nicht mehr... 

FritzBox ist zwischenzeitlich mehrfach neu gestartet. Die alte AEG-Fehlerbehebung kenne ich;: Ausschalten, Einschalten, Geht. :D

huckes, 2024-01-23:
Hast Du schon mal den Verkäufer kontaktiert?
Einige openDTU-Versionen aus Kleinanzeigen oder ebay haben ein ESP-Modul (Nodemcu, D1Mini etc.) nur aufgesteckt. Da kannst Du das ganze Modul herausnehmen und gegen ein baugleiches ausstauschen. Gibt's bei ebay, Aliexpress und weiteren Shops für kleines Geld.

fritzCARaldo, 2024-01-23:
[](url)

> Hast Du schon mal den Verkäufer kontaktiert? Einige openDTU-Versionen aus Kleinanzeigen oder ebay haben ein ESP-Modul (Nodemcu, D1Mini etc.) nur aufgesteckt. Da kannst Du das ganze Modul [...]

[image]
[image]

fritzCARaldo, 2024-01-23:
> Hast Du schon mal den Verkäufer kontaktiert? Einige openDTU-Versionen aus Kleinanzeigen oder ebay haben ein ESP-Modul (Nodemcu, D1Mini etc.) nur aufgesteckt. Da kannst Du das ganze Modul [...]

Kann man da was erkennen?

[image]
[image]

huckes, 2024-01-23:
Sieht aus wie dieses: https://www.azdelivery.de/products/esp32-developmentboard
Ob das Modul gesteckt ist oder verlötet, kann ich nicht erkennen. Müsstes Du ausprobieren bzw. mal aus dem Gehäuse heraus nehmen.

fritzCARaldo, 2024-01-23:
> Sieht aus wie dieses: https://www.azdelivery.de/products/esp32-developmentboard Ob das Modul gesteckt ist oder verlötet, kann ich nicht erkennen. Müsstes Du ausprobieren bzw. mal aus dem Gehäuse heraus nehmen.

Es scheint mir eine extra Platine dazwischen zu sein. Aber ehrlich gesagt weiß ich nicht, wie das montiert ist. Hatte gerade die Antenne ab und dachte, dann kommt mir das Modul entgegen. Tut es aber nicht. Ich schau mal, ob ich den Verkäufer noch erreichen kann. Im schlimmsten Fall muss ich mir wohl einen neuen Bausatz kaufen... Wäre schade...

Aber immerhin: Nachdem ich es gerade abgestopfelt und wieder angeschlossen habe, ist es direkt erreichbar.

stefan123t, 2024-01-23:
@fritzCARaldo kannst Du mal prüfen ob Du eines dieser schlauen WLAN Geräte in Deiner FritzBox registriert hast, das alle X  Minuten seine MAC Adresse wechselt und damit nach einer Weile Deinen kompletten WLAN Adressbereich belegt ?
Hatte bei mir auch so einen Kunden, der das "Feature" in einem privaten WLAN aktiviert hatte und mich gewundert, dass ich mit meinen Geräten keine IP Adresse mehr vom DHCP Server bekommen konnte.

fritzCARaldo, 2024-01-24:
> @fritzCARaldo kannst Du mal prüfen ob Du eines dieser schlauen WLAN Geräte in Deiner FritzBox registriert hast, das alle X Minuten seine MAC Adresse wechselt und damit nach einer Weile Deinen kompletten WLAN Adressbereich belegt ? Hatte bei mir auch so einen Kunden, der das "Feature" in einem privaten WLAN aktiviert hatte und mich gewundert, dass ich mit meinen Geräten keine IP Adresse mehr vom DHCP Server bekommen konnte.
> 
> **So deaktivieren Sie zufällige MAC-Adressen auf Android** https://de.moyens.net/wie-man/so-deaktivieren-sie-zufaellige-mac-adressen-auf-android/ ![image](https://private-user-images.githubusercontent.com/11159753/299120707-e898f1d9-5e3c-4eef-8fc3-b7ae25a2ce60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwOTQ2MzksIm5iZiI6MTcwNjA5NDMzOSwicGF0aCI6Ii8xMTE1OTc1My8yOTkxMjA3MDctZTg5OGYxZDktNWUzYy00ZWVmLThmYzMtYjdhZTI1YTJjZTYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI0VDExMDUzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFjZDFmYjdkZjFjYjRkZWEwNzI0NTBhOGJlNWQwN2JjM2RkMzI1ZmVjMGYxMDRhYTA1NmQ0NjNhZTkwMmY5MTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Qmspj5nCFc69QkMk6xk87gpMRjMg-LYm400UM628LlI)
> 
> Die Geräte melden sich dann meist mit Namen wie `PC-12-34-56-78-9A-BC` oder so ähnlich.
> 
> Wie viele dieser Einträge hast Du denn in Deiner WLAN/Netzwerkübersicht der FritzBox aktuell ?

Danke für den Versuch, aber wenn ich das umstelle (hab nur ein Google Pixel als in Frage kommendes Gerät) meldet dieses ""Vorrübergehend schlechte Netzwerkverbindung". Hab in der FritzBox mal eine feste IP vergeben. Aber ansonsten ist meine Netzwerkübersicht aufgeräumt. Da achte ich drauf...

stefan123t, 2024-01-24:
@fritzCARaldo habe ich das richtig verstanden, Du hast "Use randomized MAC (default)" an Deinem Google Pixel aktiviert, weil es ohne die random MAC eine "vorrübergehend schlechte Netzwerkverbindung" angibt ?

fritzCARaldo, 2024-01-24:
> @fritzCARaldo habe ich das richtig verstanden, Du hast "Use randomized MAC (default)" an Deinem Google Pixel aktiviert, weil es ohne die random MAC eine "vorrübergehend schlechte Netzwerkverbindung" angibt ?

Nach dem Umstellen von "Zufällige MAC-Adresse verwenden (Standard)" auf "MAC-Adresse des Geräts verwenden" kam keine Verbindung mehr zustande. Hinweis "vorrübergehend schlechte Netzwerkverbindung".

stefan123t, 2024-01-24:
Also ich habe bei mir gerade mal wieder die vielen alten Random MAC Einträge einzeln und von Hand aus der FritzBox geschmissen. Das Gerät, das diese Einträge verursacht hat war übrigens auch ein Google Pixel 6a und ist hier seit über einem Monat nicht mehr im Netz gewesen.

Eventuell ist das auch bei Dir die Ursache für beide Probleme: das der OpenDTU und das Deines Pixels selbst.

Da eine random MAC ein **Privacy Feature** für ein *öffentliches Netz* ist, hat diese Funktion (bei Google anscheinend default aktiv) m.E. in einem *privaten WLAN* nichts zu suchen, wozu auch ?

Noch dazu wenn der für WLAN verfügbare DHCP Adress-Pool mit knapp 180 IPs (bei der Fritz Box im Default) viel zu klein ist und von der FritzBox während der für dieses "Feature" viel zu langen DHCP LeaseTime m.W. nicht automatisch ordentlich aufgeräumt wird.

Was steht denn bei Dir unter **Heimnetz > Netzwerk** als Ungenutzte Verbindungen ?

Was die Nachvollziehbarkeit Deines Problems auf unserer Seite angeht, ist es so, daß Du uns ein Serielles Log schicken solltest aus dem hervorgeht, warum der ESP32 sich nicht mit Deinem Netzwerk verbinden will / kann. 
Mit der Info "geht nicht" kommen wir an dieser Stelle leider nicht weiter, sondern tappen genauso wie Du im Dunkeln. 
Also bitte ein Serielles USB Kabel an den ESP32 anschließen und ein Logfile wie oben bzw. in der OpenDTU Dokumentation beschrieben mit loggen.

fritzCARaldo, 2024-01-24:
> Also ich habe bei mir gerade mal wieder die vielen alten Random MAC Einträge einzeln und von Hand aus der FritzBox geschmissen. Das Gerät, das diese Einträge verursacht hat war übrigens auch ein [...]

Ich hab das WLAN eigentlich immer zu und die Geräte alle benannt (für die Zuordnung). Tatsächlich konnte ich gerade zwei Geräte noch entfernen. EInmal irgendein Android, welches sich die Tage angemeldet hatte (k.A. was das war) und eine Kamera, die ich aber verkauft habe. Ansonsten hat alles seine Berechtigung.

Zur Info: Aktuell läuft es wohl stabil. Letztes Update ist auch drauf. Ich werde mal weiter beobachten und mich dann ggf. mit der Protokollierung auseinandersetzen. Ist echt ein Buch mit 7 Siegeln für mich und da ich mich aktuell schon in einer Weiterbildung befinde, raucht mir auch so schon der Kopf... :/

Merkwürdig ist noch: Auch mit Nanoleaf Shapes habe ich plötzlich ein Problem. Gestern nach vielen vielen Versuchen ans Laufen gebracht. Heute morgen bekomme ich vom Pixel 7 (mein einziges Handy) wieder keine Verbindung, obwohl die Nanoleafs in der FritzBox verbunden sind... Aber da gab es ja auch kein aktuelles Update der FritzBox, die sie als Ursache identifizieren würde. Hier ist nix neu... DANKE für deine / eure Hilfe!

rolsch, 2024-01-27:
Hmm,
es hatte noch keiner nach der Stromversorgung gefragt.
Hast du die mal gewechselt?
Wie schaut es mit der Sendeleistung des NRF24 aus, steht die auf low (was ok wäre).

fritzCARaldo, 2024-01-27:
Die Lösung ist vermutlich, dass sich das WLAN Modul der Fritzbox 7590 schleichend verabschiedet. Jeden Tag dtiegen mehr und mehr Geräte aus. Teilweise ist eine Kamera Online, 2 weitere des gleichen Typs sind Offline. Habe heute eine 7590 AX V2 gekauft, die ich nach dem Ausschlafen installieren werdenm Hoffe der Spuk hat dann ein Ende. Ich werde mich hier melden! Danke euch!

closed 2024-01-28 by fritzCARaldo