tbnobody/OpenDTU #1668: Settings- and Info-Menue not scrollable

dragricola, 2024-01-19:
### What happened?

At small displays like smartphone displays some menue items at the bottom side of the menue lists "Settings" and "Info" are not accessible because they are outside of the display area and the menue window is not scrollable.  

### To Reproduce Bug

This problem only appears on small displays like smartphones under the menu items Settings (Einstellungen) and Info. 

### Expected Behavior

It would be fine, when configuration and informations of openDTU can be also done with a smartphone.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.1.18 and many versions before

### Anything else?

[image]

This screenshot shows an example, where the menu points below DTU are not accessible.

nkappler, 2024-01-27:
I've just investigated this a little bit, the fix is not that trivial, since there's another case where scrolling could appear:
[image]

I've tried a css only solution but since the same DOM elements are used for both layouts, it's tricky...

nkappler, 2024-01-27:
ok I've got a solution now...
That was quite an interesting challenge, I've learned a bit more about vue.js, thanks 🥳😂

dragricola, 2024-01-27:
`ok I've got a solution now...`

Thank you! Your solution works very fine!

commit 86b96252 (2024-02-12): fix #1668

use builtin bootstrap css for scrolling
closed 2024-02-12 by tbnobody