tbnobody/OpenDTU #1670: Leistung und Tagesertrag nicht aktuell/zu niedrig

NintendoTito, 2024-01-20:
### What happened?

Soeben um 10:30 Uhr, bei bedecktem Himmel wurde ein Tagesertrag von 0 kWh und eine aktuelle Leistung von 4 Wh angezeigt. Der grüne Punkt blinkte und auch die Anzeige auf meiner Opendtu zeigt 4 Wh an. Der Shelly zeigte eine Leistung von 67 Wh an.

Ich habe die Opendtu Seite im Browser mehrmals aktualisiert, die Anzeige blieb falsch. Ein Neustart der Opendtu behob das Problem. 

### To Reproduce Bug

Leider keine Idee wie das möglich ist 

### Expected Behavior

Korrekte Anzeige

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.1.18

tbnobody, 2024-01-20:
Was hast du in der Console gesehen? Nur so lässt sich nachvollziehen ob die Werte so vom Wechselrichter kamen oder ggf. falsch interpretiert wurden.

NintendoTito, 2024-01-20:
Ich habe in der Console nachgesehen, aber das stand auf automatisch laden und da standen keine alten Einträge.

tbnobody, 2024-01-20:
> Ich habe in der Console nachgesehen, aber das stand auf automatisch laden und da standen keine alten Einträge.

Was meinst du damit? Es sollten ja eine Textausgabe erfolgt sein. "automatisch laden" kommt so als Text nirgends vor.
Nachdem vmtl. auch keine Screenshots von "Info" --> "System"  o.ä. vorliegen kann man den Fehler gar nicht nachvollziehen.

NintendoTito, 2024-01-20:
Ok, falls der Fehler nochmal auftritt versuche ich es beim nächsten mal besser zu machen.

Was ich eben meinte war "automatisch scrollen". Und das war eben noch aktiviert. Und wenn ich die Konsole öffne, dann ist der zu Beginn leer und dann beginnen halt Daten einzulaufen. Das heißt, dass ich gar nicht auf ältere Daten von 10:20 Uhr zugreifen kann. Ich hätte da ein Logfile erwartet wo ggf auch ältere Daten stehen und nicht eben nur aktuelle Daten.

tbnobody, 2024-01-20:
Ein Logfile würde voraussetzen das Speicher im Überfluss vorhanden ist. Aber der ESP ist immer noch ein Mikrocontroller mit max. 4MB Flash. Und ja, auch nur neuere Daten würden in dem Fall helfen weil die Kommunikation zwischen ESP und Wechselrichter ersichtlich ist. Wenn dort dann andere Werte als 4Wh in den Telegrammen steht weiß man wo man ungefähr zum suchen anfangen muss. 

Würde den Issue erstmal schließen. Mach ihn einfach wieder auf oder lege einen Neuen an wenn das Problem wieder auftritt.

closed 2024-01-20 by tbnobody