tbnobody/OpenDTU #1671: Nach Browser Reset keine Anzeige der Live Ansicht ohne DTU Neustart möglich

T2023om, 2024-01-20:
### What happened?

Nach Öffnen der Live Ansicht und anschließender Aktualisierung verschwinden die Daten und die Seite bleibt leer. 
Siehe unten: Daten der Konsole zu diesem Zeitpunkt
Alle anderen Funktionen wie Einstellungen bleiben erhalten. Beim erneuten Aufruf der Live Ansicht von einer anderen Seite sieht man kurz  "Live Ansicht" und den roten Punkt und danach nur die leere Seite.
Ein DTU Reset bringt die normale Anzeige zurück.
PS:
Während ich geschrieben habe kehrte die Live Seite von allein zurück aber nur mit der Kopfzeile und dem grünen Punkt.
Ein weiterer  Reset brachte alle Informationen zurück.

### To Reproduce Bug

Browser aktualisieren mit Live Ansicht mit rotem Punkt,
danach Live Ansicht mit leerer Seite
Nach DTU Reset normale Anzeige

### Expected Behavior

Nach Browser Reset aktuelle Live Anzeige mit grünem Punkt

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

opendtu-generic_esp32.bin v24.1.14 und v24.1.18

### Relevant log/trace output

```Shell
Websocket: [/livedata][30] disconnect
13:15:47.054 > Alloc failed: loop, 68
13:15:48.058 > Alloc failed: loop, 68
13:15:49.135 > Alloc failed: loop, 68
13:15:49.442 > Fetch inverter: 116190746898
13:15:49.489 > TX RealTimeRunData Channel: 61 --> 15 90 74 68 98 80 12 22 28 80 0B 00 65 AB B9 75 00 00 00 00 00 00 00 00 CA 87 5D 
13:15:49.499 > Interrupt received
13:15:49.508 > RX Channel: 3 --> 95 90 74 68 98 90 74 68 98 01 00 01 01 A9 00 66 00 59 01 B1 01 79 00 01 D1 F5 EF | -80 dBm
13:15:49.513 > Interrupt received
13:15:49.522 > RX Channel: 3 --> 95 90 74 68 98 90 74 68 98 02 00 01 86 53 01 6A 00 F7 01 4F 01 8E 00 63 05 36 4E | -80 dBm
13:15:49.530 > Interrupt received
13:15:49.537 > RX Channel: 40 --> 95 90 74 68 98 90 74 68 98 03 01 4C 00 01 BF 39 00 01 BA A1 01 50 01 3E 08 F7 D7 | -80 dBm
13:15:49.570 > Interrupt received
13:15:49.580 > RX Channel: 23 --> 95 90 74 68 98 90 74 68 98 84 13 88 09 30 00 CD 00 67 03 E4 00 74 00 01 7D 32 C4 | -80 dBm
13:15:49.956 > RX Period End
13:15:49.956 > Success
13:15:50.056 > Alloc failed: loop, 68
13:15:51.083 > Alloc failed: loop, 68
13:15:52.057 > Alloc failed: loop, 68
13:15:53.065 > Alloc failed: loop, 68
13:15:54.152 > Alloc failed: loop, 68
13:15:54.356 > Fetch inverter: 116190746898
13:15:54.379 > TX RealTimeRunData Channel: 75 --> 15 90 74 68 98 80 12 22 28 80 0B 00 65 AB B9 7A 00 00 00 00 00 00 00 00 3A C6 E3 
O13:15:54.433 > Interrupt received
13:15:54.449 > RX Channel: 61 --> 95 90 74 68 98 90 74 68 98 01 00 01 01 A8 00 63 00 55 01 A2 01 6A 00 01 D1 F5 E7 | -80 dBm
13:15:54.480 > Interrupt received
13:15:54.481 > RX Channel: 61 --> 95 90 74 68 98 90 74 68 98 02 00 01 86 53 01 6A 00 F7 01 50 01 80 00 5F 05 0B 5E | -80 dBm
13:15:54.495 > Interrupt received
13:15:54.502 > RX Channel: 61 --> 95 90 74 68 98 90 74 68 98 03 01 40 00 01 BF 39 00 01 BA A1 01 50 01 3E 09 01 2C | -80 dBm
13:15:54.570 > Interrupt received
13:15:54.582 > RX Channel: 61 --> 95 90 74 68 98 90 74 68 98 84 13 88 08 DF 00 CF 00 63 03 E4 00 74 00 01 F3 82 12 | -80 dBm
13:15:54.884 > RX Period End
13:15:54.885 > Success
```

### Anything else?

[image]

tbnobody, 2024-01-20:
> Alloc failed: loop, 68

Das bedeutet, dass nicht genug Speicher zur Verfügung stand um die Statistikdaten auszugeben
https://github.com/tbnobody/OpenDTU/blob/bdfdc3913b1641d2442ea56b2ce2680a22ab4862/src/WebApi_ws_live.cpp#L68
Wie viele Inverter hast du aktuell eingestellt? 

Es kann teilweise sein das ein Browser "Amok" läuft und ggf. zu viele Verbindungen zum ESP aufbaut. Das führt dann genau zu solchem Verhalten. Generell hab ich diese Codestelle schon am Schirm zum überarbeiten.

T2023om, 2024-01-21:
Ich habe nur einen Inverter eingestellt.
Anbei 2 Bilder mit den Heap Details
1. Bild  9:33 Es liefen gleichzeitig ein Ipad und ein WIN10 Laptop mit ihren Browsern und DTU
2. Bild 12:40 nur WIN10 seit 3 Stunden allein

[image]
[image]

tbnobody, 2024-01-21:
Bei den Bildern funktioniert aber alles oder? 

Du schreibst oben bei "To Reproduce Bug" 

> Browser aktualisieren mit Live Ansicht mit rotem Punkt,
> danach Live Ansicht mit leerer Seite
> Nach DTU Reset normale Anzeige

Kannst du das mal machen und dann ein Bild vom Heap Status machen? Ich habe das hier ein paar mal versucht, aber kann es nicht nachvollziehen.

T2023om, 2024-01-21:
Hi, 
ja bei den Bildern hat alles funktioniert.

Und nun noch der Vorzeige-Effekt:
Ich habe nun sogar noch als 3. ein Android Handy eingeschaltet und kann den Fehler nicht reproduzieren.
An den Zahlen des Heap Details hat sich trotz der 3 Browser nichts geändert.

Schönen Sonntag noch und danke

schneeer, 2024-01-29:
Hi,
evtl, hatte ich ein ähnliches Problem.
Ich bastle nebenher an der Web-App herum, da ich zusätzliche Displayeinstellungen wünsche.
Beim Testen ist mir aufgefallen, das im Google Chrome vorgenommene Änderungen an der Web-App nicht angezeigt werden, oder schlimmstenfalls die Hardwarekonfigurationsseite komplett leer blieb.
Chrome- oder PC-Neustart half nicht wirklich. Was aber funktioniert hat, war, den Browserverlauf jedesmal zu löschen. Damit wurde die Webseite auf Anhieb korrekt aufgebaut.

cskowronnek, 2024-01-29:
Ich habe das Problem auch. Alloc failed, Heap bei 50%
Tritt erst nach einigen Stunden auf und dann auf allen Browsern und Endgeräten. Kam mit der 24.1.18 oder 24.1.14

tbnobody, 2024-01-29:
@schneeer dein Problem ist ein komplett Anderes. Hängt mit dem Cache zusammen. Der Cache wird erst mit einem neuen Git-Hash invalidiert. Wenn du am testen bist kannst du das auch einfacher mit dem Development Server (siehe [hier](https://tbnobody.github.io/OpenDTU-docs/firmware/compile_webapp/#starting-the-development-server)) machen.

stefan-lange, 2024-02-12:
Habe ein ähnliches Problem derzeit mit `v24.1.26`, aber war in der Vergangenheit auch schon öfters mit anderen Versionen der Fall. Ein Neustart fixt es immer für eine Zeit.

## Live
[image]

## System
[image]

## Konsole
[image]

Vielleicht helfen die Screens.

MetaChuh, 2024-02-12:
> Habe ein ähnliches Problem derzeit mit `v24.1.26`, aber war in der Vergangenheit auch schon öfters mit anderen Versionen der Fall. Ein Neustart fixt es immer für eine Zeit.

sieht aus als ob der browser den web socket data stream nicht erreichen kann.
1) siehst du in `info > konsole` den log ticker, oder bleibt dein log schwarz ?

2) ist deine opendtu hinter einem reverse proxy oder gateway, oder passiert das auch, wenn du lokal auf die interne ip (zb. 192.168.0.x) zugreifst ?

MetaChuh, 2024-02-13:
@stefan-lange 

sehr cool, danke für die ausführliche info.
deinen screenshot vom log hatte ich übersehen, sorry.

bin schon gespannt was du bei dir herausfindest.

ich konnte zb. sowohl die live view seite, als auch die konsole nur sehen, wenn ich physisch im gleichen vlan war.
(komischerweise bei mir im vergleich zu bei dir, konnte ich die leistungs summary sehen, und "nur" die inverter auswahl und inverter details waren weg)

via nginx reverse proxy, 2. nic verbunden zum user vlan, gings dann mit den zusatzeinstellungen `proxy_set_header Upgrade $http_upgrade;`
und `proxy_set_header Connection $connection_upgrade;`

(mit der einschränkung bei mir, dass ich explizit http1.1 requests machen musste, da http2 bei mir die $http_upgrade variable irgendwie "verschluckte")

greetings

stefan-lange, 2024-02-13:
Zufälligerweise ist es vorhin während ich hier geschrieben habe wieder zu dem Zustand gekommen.

## Reading the blackbox
[image]

Nach `ping` vom Client alle 60s wurde Verbindung geschlossen mit Code `1005`
```
1005 CLOSED_NO_STATUS

1005 is a reserved value and MUST NOT be set as a status code in a Close control frame by an endpoint. It is designated for use in applications expecting a status code to indicate that no status code was actually present.
```

## Wiederbelebungsversuche

1. Ein Browserfenster mit Live-View und dort `F5`
2. Zweites Fenster mit Konsole zum Checken

[image]

Auf Anfrage des Browsers sieht man, dass die OpenDTU eine neue Verbindung aufbaut `connect` und die vorherige schließt, aber mit Fehler `Websocket: [/livedata][xxx] disconnect Alloc failed: sendDataTaskCb, 74`

## Ergebnis
Die Live-View leer bleibt wie in meinem initialen Post zu sehen (https://github.com/tbnobody/OpenDTU/issues/1671#issuecomment-1938550801)

Er springt im Code hier raus: https://github.com/tbnobody/OpenDTU/blob/master/src/WebApi_ws_live.cpp#L77
Ich bin kein CPPler. Das Doc `DynamicJsonDocument root(4096)` scheint leer zu sein, warum und weshalb kann vielleicht jemand anders beantworten.

stefan-lange, 2024-02-13:
## Interessantes

Vor paar Minuten ist der Debugging Tab der OpenDTU-Konsole im Firefox abgestürzt. Nach Reload des Tabs geht auch die Live-View wieder mit sauberem `connect/disconnect` der Websockets bei Page-Refresh.

[image]
Die `Middle missing` Nachricht im Screen während connect ist sicherlich ein anderes Thema.

Reine Spekulation: ich vermute fast, dass durch den `sendDataTaskCb` Fehler beim disconnect die Verbindungen vielleicht nicht ganz sauber geschlossen werden.

MetaChuh, 2024-02-13:
spannend.

falls du die dtu noch nicht rebootet hast:

was siehst du, wenn du dich ausschließlich mit dem gleichen vlan der dtu verbindest, und auf die native ip der dtu direkt zugreifst ?
(zb einen beliebigen port am switch nur diesen wlan tag geben und ein notebook anstöpseln)

same same oder anders ?

falls du einen extra esp32 hast, kannst du auch mal die v24.2.12 zusätzlich probieren, weil wegen den api änderungen ist auch die webapp refactored.
(disclaimer: bitte nur zusätzlich, falls du die api brauchst, und änderungen noch nicht kennst)

> `DynamicJsonDocument root(4096)` scheint leer zu sein, warum und weshalb kann vielleicht jemand anders beantworten.

ja

stefan-lange, 2024-02-13:
Gerade ist der Zustand nicht (mehr) vorhanden. Neugestartet ist auch noch nichts, vielleicht kommt es morgen noch mal zu dem Problem.

Vorhin hatte ich während des Fehlers auch mal `http://10.30.50.70/api/livedata/status` aufgerufen. Es lieferte `null`.

In das andere VLAN kann ich auch wechseln. Das ist aber transparent. Es ist kein Proxy dazwischen, nur eine PfSense mit Routing und paar Regeln. Die Verbindungen waren auch `established` vorhin vom Client zur OpenDTU, also ganz normal. Es kam aber nichts von der DTU zurück.

MetaChuh, 2024-02-13:
danke dir für die infos.

> Es lieferte `null`.

null ist nicht gut.
ich hab' zwar in den steuerungen ein failsafe d'rin, der die dtu via api rebootet, wenn 5x hintereinander mit je 1s query abstand null kommt, jedoch wurde dieser failsafe schon lange bei mir nicht mehr getriggert (oder ich hab's übersehen).

> Es kam aber nichts von der DTU zurück.

soweit ich dies bis jetzt verstehe, bezieht sich dies auf api, bzw. websocket daten, während der async web server nach wie vor die html daten liefert, ist dies korrekt ?

> nur eine PfSense mit Routing und paar Regeln.

bei mir opnsense, also im core wahrscheinlich nicht viel unterschied.

> Einen zweiten ESP32 habe ich leider nicht.

ich mache nichts mehr ohne extra testgeräte (gilt für it, solar und alles).
sonst sind die produktivumgebungen das einzige testfeld, und wenn etwas nach updates oder änderungen nicht passt, ist's stress pur ... je nach kunde interner stress, externer stress, oder beides.
(ist gottseidank günstiger, als ob man einem kunden sagt: sie brauchen ihre gesamte infrastruktur für die tests vor einem upgrade 1:1 ein zweites mal 😉)

stefan-lange, 2024-02-13:
> soweit ich dies bis jetzt verstehe, bezieht sich dies auf api, bzw. websocket daten, während der async web server nach wie vor die html daten liefert, ist dies korrekt ?
Ja, ist korrekt. UI funktionierte wie immer, bis auf den Livedaten-Websocket. Der Konsolen-Websocket hat funktioniert. Auch MQTT im Hintergrund hat die ganze Zeit funktioniert. Also die Verbindung intern (DTU -> Wechselrichter) schien ok zu sein. Das grenzt es mMn schon etwas ein.

Habe auf die v24.2.12 aktualisiert. Sonst bin ich auch "pendatisch" was prod angeht. OpenDTU ist bei mir aber ein Spielprojekt und es ist selbst ja auch nicht die Quelle der Daten, sondern die Bridge. Es kann nicht viel verloren gehen, von daher sehe ich es etwas lockerer ;)

stefan-lange, 2024-02-15:
Läuft seit über 2 Tagen erste Sahne. Ich beobachte passiv weiter.

Danke euch, auch grundlegend für eure Arbeit am Projekt.

MetaChuh, 2024-02-15:
> Läuft seit über 2 Tagen erste Sahne.

perfekt.

wenn du zufrieden bist, kannst du dieses issue schließen.
bei bedarf kannst du es manuell wieder öffnen.
wenn du zusätzliche informationen hast, kannst du die gerne nachträglich posten, ohne dass das issue automatisch neu geöffnet wird.

viel freude und wartungsfreie stabilität.

greetings

stefan-lange, 2024-02-16:
> wenn du zufrieden bist, kannst du dieses issue schließen

Würde ich machen, wenn es meines wäre ;)

T2023om, 2024-02-16:
Danke für die Erfolgsnachricht, habe aber z.Z keinen Zugriff auf meine DTU und konnte das Update nicht einspielen. In 3 Wochen kann ich mich melden und berichten und diese issue schließen.Nochmals danke für die tolle Arbeit

barnosch, 2024-02-29:
Ich hab das selbe Problem.
Muss jeden Tag neustarten, damit mir wieder was angezeigt wird.
Habe seit kurzem einen 2ten Wechselrichter dran, aber ich meine, das Problem tritt erst seit der aktuellen Firmware auf.

stefan-lange, 2024-02-29:
Welche Version der DTU hast du denn installiert? Seit meinem Update auf `v24.2.12` läuft die kleine ohne Probleme durchweg. Mittlerweile knapp 16 Tage.

barnosch, 2024-03-01:
> Ich hab die aktuellste 24.2.16 drauf. Dann geh ich auch mal zurück auf die 2.12. und teste das

Kein Unterschied, nach ca. 24h ist die "Live" Ansicht wieder hängengeblieben. Alles andere funktioniert und ich kann mich durch die Website klicken.

stefan-lange, 2024-03-01:
Kannst nur versuchen erst mal paar Dinge auszuschließen.

1. Ist irgendwas zwischen dir (Browser) und der DTU, z.B. Firewalls, Proxies etc.
2. Gibt es Fehler im Log `http://.../info/console` ?
3. Selber die Websocket-Kommunikation angucken, ob es zu Fehlern kommt (mit denen die Devs hier dann was anfangen können)
z.B. im Firefox die Entwickler-Konsole öffnen (`Strg + Shift + E`) > Tab `Netwerkanalyse`. Rechts den Filter auf `WebSockets` stellen. Auf der linken Seite siehst du die Verbindungen, z.B. `livedata`. Das auswählen und wieder rechts auf Tab `Antwort` gehen. Dort müssen Daten alle paar Sekunden kommen oder in deinem Fall sieht man evtl. Fehler à la `null` etc.

T2023om, 2024-03-22:
Nach 3 Wochen ohne weitere Kommentare und ungeklärte Probleme bin ich der Meinung diese issue als geklärt zu schliessen

closed 2024-03-22 by T2023om