tbnobody/OpenDTU #1684: Hardware Partnumber is given as int but Hoymiles interprets it in hex / custom numbering scheme

stefan123t, 2024-01-25:
### What happened?

My Hardware Partnumber is given in OpenDTU as 269553683.

Converting this from decimal to hex yields 0x10111013,
which is referred to as the Hardware Partnumber in Hoymiles as:

Hardware Ver.: 01.01.01
Software Ver.: 01.01.03

Maybe someone with a Hoymiles DTU can double check this ?

### To Reproduce Bug

always, just see Inverter > Inverter Info Dialog:

### Expected Behavior

Use the Hoymiles Syntac for HW Partnumber and Hardware / Software Versions numbering scheme

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

a5cc0a4

### Relevant log/trace output

```Shell
See Inverter > Inverter Info Dialog:

Hardware Part Number	269553683
```

tbnobody, 2024-01-26:
Currently it looks like this:

Software Ver.: `(0x10111013 >> 8 & 0b1111) . (0x10111013 >> 4 & 0b1111) . (0x10111013 >> 0 & 0b1111)`
Hardware Ver.:  `(0x10111013 >> 28 & 0b1111) . (0x10111013 >> 20 & 0b1111) . (0x10111013 >> 16 & 0b1111)`

But it would be nice to test it with other inputs.

stefan123t, 2024-02-09:
@broth-itk would you kindly check / compare this between your Hoymiles DTU Pro and the OpenDTU outputs ?

stefan123t, 2024-07-31:
@broth-itk friendly reminder to double check this with an Hoymiles DTU :stuck_out_tongue_winking_eye:

broth-itk, 2024-07-31:
@stefan123t 

Hardware Part Number	269635841, shows as Hardware version 01.00 in OpenDTU = 	Hardware version S-Miles Cloud: H00.04.00
Hardware Part Number	269644033, shows as Hardware version 01.00 in OpenDTU = 	Hardware version S-Miles Cloud: H00.04.00
Hardware Part Number	270541056, shows as Hardware version 02.00 in OpenDTU = 	Hardware version S-Miles Cloud: H00.08.00

Does this help?

stefan123t, 2024-07-31:
Maybe we should double check the code in the S-Miles Installer App if we can find some routines for converting this binary data.

But then the HMS-x00**W** WiFi Series Models have a built-in DTUBI and therefor the protobuf NetworkCommands may contain already pre-decoded HW Version numbers 🙂

closed 2025-04-19 by github-actions[bot]